genodelabs/genode #3792: sel4: BOARD not overriden in platform-specific library makefiles

chelmuth, 2020-06-30:
As seen repeatedly in the last nights, the builder fails to build `make -C build/arm_v7a KERNEL=sel4 BOARD=imx6q_sabrelite` with

```
[2020-06-29 12:20:09]  [STAGE] autoconf.h
[2020-06-29 12:20:09] cp: cannot stat './include/generated/autoconf.h': No such file or directory
[2020-06-29 12:20:09] common.mk:227: recipe for target 'install-headers' failed
[2020-06-29 12:20:09] make[4]: *** [install-headers] Error 1
[2020-06-29 12:20:09] make[4]: Target 'default' not remade because of errors.
[2020-06-29 12:20:09] /data/genode/repos/base-sel4/lib/mk/spec/arm/kernel-sel4.inc:16: recipe for target 'elfloader/elfloader.o' failed
[2020-06-29 12:20:09] make[3]: *** [elfloader/elfloader.o] Error 2
[2020-06-29 12:20:09] make[3]: Target 'all' not remade because of errors.
[2020-06-29 12:20:09] var/libdeps:4389: recipe for target 'kernel-sel4-imx7d_sabre.lib' failed
```

This is strange as I had expected kernel-sel4-imx7d_sabre.lib would not be built in this step, which is a false assumption. Any arm_v7a sel4 kernel library will be built in this step because the `BOARD` is not used as `SPEC` selector since #3316 was completed.

The reason for the build error above is a non-existing link target  *.../contrib/sel4-7935487f91a31c0cd8aaf09278f6312af56bb935/src/kernel/sel4/configs/imx7/imx6q_sabrelite/autoconf.h* that is created by https://github.com/genodelabs/genode/blob/4450b37ff5b22e99133863a6f56d5b715c7f3aaa/repos/base-sel4/lib/mk/spec/arm/kernel-sel4.inc#L22 included by https://github.com/genodelabs/genode/blob/4450b37ff5b22e99133863a6f56d5b715c7f3aaa/repos/base-sel4/lib/mk/spec/arm/kernel-sel4-imx7d_sabre.mk#L5

The cause of this that when building `make -C build/arm_v7a KERNEL=sel4 BOARD=imx6q_sabrelite LIB=kernel-sel4-imx7d_sabre` (which is comparable to nightly), the `BOARD` variable is set on the command line and the assignment at https://github.com/genodelabs/genode/blob/4450b37ff5b22e99133863a6f56d5b715c7f3aaa/repos/base-sel4/lib/mk/spec/arm/kernel-sel4-imx7d_sabre.mk#L3 is not effective. Prepending the statement with `override` fixes the issue quickly but the use of `BOARD` in makefiles is still questionable according to an offline discussion with @skalk.

nfeske, 2020-07-03:
Since https://github.com/genodelabs/genode/commit/8e205e0324ca5d0c7ac1bb4385836fdc6d14a996 entered the master branch, can this issue be closed?

chelmuth, 2020-07-03:
For me the concrete issue is fixed. What kept me from adding `Fixes #3792` was the questionable use of `BOARD` in the first place. That said, I'd leave the decision to @skalk.

skalk, 2020-07-07:
I'm fine with closing this. The questionable use of `BOARD` indeed would be good to replace by another non-environment variable like `SEL4_BOARD` or similar, but it is high-level build-system sanitation and I see no urgent pressure for us to do it right away.

closed 2020-07-07 by skalk