genodelabs/genode #3795: qt5: build Qt libraries and applications with qmake

cproc, 2020-06-30:
This is a prerequisite for building Qt applications with the goa tool and for building the qtwebengine module, as planned for the 20.08 release.

The Qt libraries are built with module granularity (qtbase, qtdeclarative, ...) now to avoid redundant configuration and build overhead. Also, there's a single 'qt5' api package now, which is easier to generate and prepare for qmake than multiple api packages.

The QPA plugin (GUI backend) is built as actual Qt plugin now and needs to be initialized with the Genode environment by preloading the plugin with `dlopen()`. This is done by the new `qpa_init()` function, which is implemented in the `qt5_component/qpa_init.h` header file, which can be included in the cases where `Libc::Component::construct()` is implemented manually instead of linking with the `qt5_component` library.

A target.mk file for a simple Qt application now looks like this:

```
QMAKE_PROJECT_FILE = $(PRG_DIR)/qt_quick.pro

QMAKE_TARGET_BINARIES = test-qt_quick

QT5_PORT_LIBS += libQt5Core libQt5Gui libQt5Network
QT5_PORT_LIBS += libQt5Qml libQt5Quick

LIBS = libc libm mesa qt5_component stdcxx $(QT5_PORT_LIBS)

include $(call select_from_repositories,lib/import/import-qt5_qmake.mk)
```

where `QMAKE_PROJECT_FILE` is the path to the project file to be processed by qmake, `QMAKE_TARGET_BINARIES` lists the names of the generated binaries for the creation of symlinks in the `bin` and `debug` directories and `QT5_PORT_LIBS` lists the names of the used Qt 5 port libraries. These are listed in this extra make variable in addition to the `LIBS` variable, because all libraries need to be symlinked into a dedicated directory tree where qmake can find them and the Qt 5 port libraries don't always have ABI libraries because of external symbol files, so the symlinks for Qt libraries need to be created differently. The actual preparation steps for qmake and the execution of qmake itself are finally handled by the `import-qt5_qmake.mk` file.

The `tool_chain_qt5` script has been updated to install the `qmake` program as well, so it needs to be run again before qmake can be used. The updated tool chain gets installed to `/usr/local/genode/qt5/20.08` by default, so there should be no version conflict with an existing Qt tool chain.

chelmuth, 2020-07-01:
> Showing 516 changed files with 1,822 additions and 20,987 deletions.

That is a huge pile of changes. Thanks for your efforts! It looks much leaner now and the prospect to use Goa for Qt applications exclusively soon is quite encouraging.

nfeske, 2020-07-03:
I have to echo @chelmuth comment. This big rework is hugely impressive. Thank you @cproc for the great care.

I merged your commit to staging.

For the record (e.g., the release notes), it would be worthwhile to collect a few implications of this rework.

- The removal of Arora

- The switch of Qt5-based examples to the drivers_interactive pkg, which excludes the use of networking and storage. When crafting applications that depend of such I/O, a Goa project would normally be the tool of choice. Of course, special scenarios - crafted for a specific hardware config instead of using the drivers_interactive pkg - can still be created manually. But the run scripts found in the Genode tree don't do that.

commit 5af45525 (2020-07-03): qt5: build Qt libraries and applications with qmake

Fixes #3795
closed 2020-07-13 by nfeske