genodelabs/genode #3797: base_linux: segmention fault in ld.lib.so

alex-ab, 2020-07-01:
Sporadically, I get segfaults in ld.lib.so (genode master 4450b37ff5b22e99133863a6f56d5b715c7f3aaa, scrcp_rgb888)
```
[init -> scrcpy] lwIP Nic interface up address=10.0.3.2 netmask=255.255.255.0 gateway=10.0.3.1
[init -> scrcpy] create thread stream
[init -> scrcpy] create pthread.1 -> cpu 0
[init -> scrcpy] create thread controller
[init -> scrcpy] create pthread.2 -> cpu 0
[init -> scrcpy] create pthread.3 -> cpu 0
[init -> scrcpy] create thread receiver
[init -> scrcpy] create pthread.4 -> cpu 2
Kernel: Segmentation fault (signum=11), see Linux kernel log for detail
```

```
[ 7593.836907] audit: type=1326 audit(1593597066.198:50): auid=1000 uid=1000 gid=1000 ses=3 pid=601 comm="ld.lib.so" exe="build/linux/var/run/scrcpy/genode/ld.lib.so" sig=31 arch=c000003e syscall=15 compat=0 ip=0x500aba67 code=0x80000000
```

chelmuth, 2020-07-01:
Unfortunately, the kernel-log line tells us that `syscall=15` which is  `rt_sigreturn` failed. Is there more in the log before this message?

alex-ab, 2020-07-01:
Not really:

```
[ 5432.784797] CPU2: Package temperature/speed normal
[ 5986.446253] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
[ 6601.833842] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
[ 6840.116627] IPv6: ADDRCONF(NETDEV_CHANGE): tap0: link becomes ready
[ 7593.836907] audit: type=1326 audit(1593597066.198:50): auid=1000 uid=1000 gid=1000 ses=3 pid=601 comm="ld.lib.so" exe="/home/alex/genode/genode.devel/build/linux/var/run/scrcpy/genode/ld.lib.so" sig=31 arch=c
000003e syscall=15 compat=0 ip=0x500aba67 code=0x80000000
```

alex-ab, 2020-07-01:
> Unfortunately, the kernel-log line tells us that `syscall=15` which is `rt_sigreturn` failed. Is there more in the log before this message?

I updated the objdump snippet. As @ssumpf pointed out, the rt_sigreturn code is before the fault ip. Is after the syscall something like ret* missing ?

chelmuth, 2020-07-02:
As I understand it the `rt_sigreturn` is a return from signal handler to the normal execution and should never return to the calling IP, ie. `lx_restore_rt` is `noreturn`.

chelmuth, 2020-07-02:
I found https://bugs.launchpad.net/ubuntu-z-systems/+bug/1686618 which shares the cryptic log message with this issue (`sig=31...`). @throwException Is this an indicator for a missing syscall permission in seccomp?

chelmuth, 2020-07-02:
@alex-ab please try to copy https://github.com/genodelabs/genode/blob/4450b37ff5b22e99133863a6f56d5b715c7f3aaa/tool/seccomp/seccomp_bpf_compiler.h#L183-L184 to the x86 path and retry.

throwException, 2020-07-02:
> I found https://bugs.launchpad.net/ubuntu-z-systems/+bug/1686618 which shares the cryptic log message with this issue (`sig=31...`). @throwException Is this an indicator for a missing syscall permission in seccomp?
There should be no segfault when seccomp terminates a process as SIGSYS is sent to all threads at once.

> @alex-ab please try to copy
> https://github.com/genodelabs/genode/blob/4450b37ff5b22e99133863a6f56d5b715c7f3aaa/tool/seccomp/seccomp_bpf_compiler.h#L183-L184
Beware to update the binary BPF filter using `repos/base-linux/src/lib/seccomp/update.sh`.

> [ 7593.836907] audit: type=1326 audit(1593597066.198:50): auid=1000 uid=1000 gid=1000 ses=3 pid=601 comm="ld.lib.so" exe="build/linux/var/run/scrcpy/genode/ld.lib.so" sig=31 arch=c000003e syscall=15 compat=0 ip=0x500aba67 code=0x80000000
Have you enabled audit in `seccomp_bpf_compiler.h`? Otherwise I wouldn't expect a message in `dmesg` as this feature is not enable by default as it's not available on all targets.

As a side note: To debug seccomp, `strace` is very helpful, especially if all known good syscalls are filtered out.

chelmuth, 2020-07-02:
> Have you enabled audit in `seccomp_bpf_compiler.h`? Otherwise I wouldn't expect a message in `dmesg` as this feature is not enable by default as it's not available on all targets.

It seems my Ubuntu 18.04 system is that verbose by default.
```
> cat /proc/sys/kernel/seccomp/actions_logged
kill_process kill_thread trap errno trace log
```

> As a side note: To debug seccomp, `strace` is very helpful, especially if all known good syscalls are filtered out.

Are you referring to `strace -p <pid>`? Do you have an example command line with all filters specified?

throwException, 2020-07-02:
> Are you referring to `strace -p <pid>`? Do you have an example command line with all filters specified?

Yes, I change `tool/run/power_on/linux` with the following line to debug:
`eval spawn strace -e !sendmsg,recvmsg,close,epoll_wait,epoll_ctl,epoll_create,socketpair,mmap,dup,dup2,futex,nanosleep,gettimeofday,fstat,getpid,munmap,rt_sigaction -ff -o /tmp/genode ./core`

chelmuth, 2020-07-02:
Thanks, this looks quite useful.

alex-ab, 2020-07-06:
Addressed by #3799

closed 2020-07-06 by alex-ab