genodelabs/genode #3799: tool/seccomp: allow sigreturn on x86

chelmuth, 2020-07-02:
Contrary to the code comment signals are also used on x86, which requires to return from the signal handler.

During my tests I was wondering how the correct syscall numbers are picked up when generating BPF rules cross-platform. @throwException do you have more information on this? Does it work via the pseudo syscall numbers?

throwException, 2020-07-02:
> During my tests I was wondering how the correct syscall numbers are picked up when generating BPF rules cross-platform. @throwException do you have more information on this? Does it work via the pseudo syscall numbers?

I was wondering the same thing, but came to the conclusion that the pseudo syscall numbers work well.

nfeske, 2020-07-03:
Merged to staging.

commit 93ab972d (2020-07-03): tool/seccomp: allow sigreturn on x86

Fixes #3799
closed 2020-07-13 by nfeske