genodelabs/genode #3800: vfs read-write deadlock

throwException, 2020-07-06:
**Scenario**

One test process continuously creates and deletes a file on a vfs. Another process contentiously tries to read that file.

**Background**

This came up, when testing a component that computes a hash over a file system, then watches and recomputes the that hash.

**Observed behaviour**

The vfs component deadlocks.

> [07:09:40.548] [init -> test_fs] Error: deadlock ahead, mutex=0x7ffff7fe49c8, return ip=0xafa7784

**Expected behaviour**

The operation should continue forever, reading the file over and over. Some exceptions are expected, but ignored.

**To reproduce**

1. Checkout https://github.com/throwException/genode/tree/vfs_read_write
2. run `test-read_write_file`

chelmuth, 2020-07-06:
This seems to be a race between `unlink()` and `complete_read()` from an open VFS handle in the RAM VFS plugin. The implementation of `unlink()` implicitly unlinks the file and removes the RAM node. Fix is pending.

chelmuth, 2020-07-06:
Sorry, please ignore my confusing remark about unlink and node removal. The real issue was a never released node mutex on `remove()`.

```diff
diff --git a/repos/os/src/lib/vfs/ram_file_system.h b/repos/os/src/lib/vfs/ram_file_system.h
index dab6a7922a..5f903885cb 100644
--- a/repos/os/src/lib/vfs/ram_file_system.h
+++ b/repos/os/src/lib/vfs/ram_file_system.h
@@ -887,8 +887,8 @@ class Vfs::Ram_file_system : public Vfs::File_system
 
 			Node *node = parent->child(basename(path));
 			if (!node) return UNLINK_ERR_NO_ENTRY;
+			Node::Guard node_guard(node);
 
-			node->acquire();
 			parent->release(node);
 			node->notify();
 			parent->notify();
```

I also adapted your test program to catch the `Truncated_during_read` exception.

```diff
diff --git a/repos/os/src/test/test-read_file/main.cc b/repos/os/src/test/test-read_file/main.cc
index e3538fab5d..6187186829 100644
--- a/repos/os/src/test/test-read_file/main.cc
+++ b/repos/os/src/test/test-read_file/main.cc
@@ -38,6 +38,9 @@ struct Test::Main
 		catch (Genode::File::Open_failed)
 		{
 		}
+		catch (Genode::File::Truncated_during_read)
+		{
+		}
 	}
 
 	Main(Env &env) : _env(env)
```

chelmuth, 2020-07-06:
@nfeske @ehmry any objections to 53103ddf055f963fc61aaba8f89aa0f13a65bd8c ?

nfeske, 2020-07-06:
No objection from my side. Soon the VFS-internal synchronization will be a thing of the past anyway. ;-)

Thanks @throwException for the excellent bug report and @chelmuth for solving it so quickly.

ehmry, 2020-07-06:
Looks good to me.

commit 53103ddf (2020-07-06): ram fs: Release node mutex on unlink

This is needed if an open file is unlinked.

Fixes #3800
commit e163055f (2020-07-13): ram fs: Release node mutex on unlink

This is needed if an open file is unlinked.

Fixes #3800
closed 2020-07-13 by nfeske