genodelabs/genode #3806: base: remove deprecated cancel_blocking() from Genode::Cpu_thread and Genode::Thread interface

alex-ab, 2020-07-06:
The functionality is not used by any component anymore. Additionally, the removal is a prerequisite to privatize and/or remove/modify the Genode::Cancelable_lock implementation as next step.

nfeske, 2020-07-07:
> Additionally, the removal is a prerequisite to privatize and/or remove/modify the Genode::Cancelable_lock implementation as next step.

Just as a minor nitpick, there is no such dependency. The removal/privatization of the `Cancelable_lock` could be done without this patch. Regardless of the order, however, our cleanup work is of course more than welcome! I merged your commit to staging.

alex-ab, 2020-07-07:
> Just as a minor nitpick, there is no such dependency. The removal/privatization of the `Cancelable_lock` could be done without this patch. Regardless of the order, however, our cleanup work is of course more than welcome! I merged your commit to staging.

I don't understand or we misunderstand. If I remove the handling of the cancellation of thread blocking within the Cancelable Lock implementation first, I still would be able to forecefully cancel the by a Genode RPC interface. That would lead to an inconsistent state within our internal Lock implementation (like assertion), or ? To avoid that, one should first remove the ability to forceful cancel before removing the overall handling in the current Cancelable_lock implementation.

nfeske, 2020-07-07:
> I don't understand or we misunderstand.

The misunderstanding may be that I saw the removal ("privatization") from the public API as a separate step that does not change the current meaning of regular (non-cancelable) lock, whereby the rework of the lock can come at a later time as a subsequent optimization step (removing the loop). In contrast, I think you presumed that the merging of both the `Cancelable_lock` and `Lock` (in particular removing the while loop) as part of this "privatization" step.

Either way is fine. ;-)

commit eb97952d (2020-07-07): base: remove deprecated cancel_blocking() support

for threads.

Fixes #3806
commit f3efbe50 (2020-07-13): base: remove deprecated cancel_blocking() support

for threads.

Fixes #3806
closed 2020-07-13 by nfeske