genodelabs/genode #3808: UEFI/boot_fb: constrain Qemu resolution to reasonable value

nfeske, 2020-07-07:
When booting graphical scenarios via UEFI on Qemu, the platform info reports a frame buffer of 2048x2048. These values are picked up by the boot-fb driver, which tries to allocate the backing store accordingly. This allocation of 16 MiB is not anticipated by most Qemu-targeting scenarios. It also goes without saying that a default size of the Qemu window of 2048x2048 is not very useful in practice.

Fortunately, this behavior can be tweaked via the Qemu command-line argument `-device VGA,vgamem_mb=8`, which deliberately limits the boot framebuffer to a max of 1920x1080.

I'd like to add this option by default in tool/run/power_on/qemu. @alex-ab Do you agree?

alex-ab, 2020-07-07:
> I'd like to add this option by default in tool/run/power_on/qemu. @alex-ab Do you agree?

Yes, I'm fine.

nfeske, 2020-07-07:
I added 00f3d5663b8588e70f48cd34e314e171d005e72d to staging.

commit 00f3d566 (2020-07-07): UEFI/boot_fb: constrain Qemu to 1920x1080

Fixes #3808
commit a0b0892d (2020-07-13): UEFI/boot_fb: constrain Qemu to 1920x1080

Fixes #3808
closed 2020-07-13 by nfeske