genodelabs/genode #3810: vbox5-nova: avoid triggering Blocking_canceled exception

alex-ab, 2020-07-07:
The transparent handling of this cancellation state is scheduled to be removed from the Genode Lock implementation and should not be triggered accidentally by the nova specific adaptations in vbox5-nova.

alex-ab, 2020-07-08:
After a offline discussion with @chelmuth we concluded to try out to come up with a pthread condwait variant.

alex-ab, 2020-07-08:
> After a offline discussion with @chelmuth we concluded to try out to come up with a pthread condwait variant.

f0def8b2a7dd29c87af0094c7943c9e0083b8ba1 uses pthread primitives to implement the blocking. Tested with vbox5_ubuntu_16_04 run script.

chelmuth, 2020-07-08:
I merged the pretty slim commit to staging.

commit 0938acf7 (2020-07-08): vbox5-nova: avoid Blocking_canceled exception

using pthread primitives to implement block/wakeup of EMT thread

Fixes #3810
commit 641679f7 (2020-07-13): vbox5-nova: avoid Blocking_canceled exception

using pthread primitives to implement block/wakeup of EMT thread

Fixes #3810
closed 2020-07-13 by nfeske