genodelabs/genode #3811: trace: mechanism to detect to small argument-buffer

alex-ab, 2020-07-07:
When a Trace connection is created, one has to provide the amount of memory for the session argument-buffer.

If more trace subjects are available than space is available in the argument buffer, the caller will miss to see some of the trace subjects.

For ```size_t subjects(Subject_id *dst, size_t dst_len)``` exists the vague hint, that one misses potentially some trace subjects, if the return value is equal to dst_len.

For ```template <typename FN> unsigned for_each_subject_info(FN const &fn)``` this hint can't be derived anymore.

alex-ab, 2020-07-07:
6b1ef54276c5770f8a6e43cb35dcf4a6ba0426cd would help for my current case, but I'm open for other suggestions. With this change I could detect the too small argument buffer and re-create the trace connection with increase argument buffer size.

nfeske, 2020-07-08:
I would prefer to avoid boolean return values unless the meaning is intuitively clear from the name of the called method (as is the case for "event_pending", "valid", or similar accessors).

How do you think about introducing a new type as follows?

```
struct For_each_subject_info_result { unsigned count, limit; };
```

The end of the `for_each_subject_info` method would become:

```
return { .count = num_subjects,
         .limit = max_subjects };
```

With this return value in place, the caller can draw the conclusion by self-explanatory code:

```
auto result = for_each_subject_info(...
...
if (result.count > result.limit)
   ...
```

alex-ab, 2020-07-09:
@nfeske: Thanks for the hints. ef98aefb87bc1b719d083a1ecc0b60b58b67a7cd and 2586b1614ca04256875eafd418b50f240b9d4975 contains the adjustments.

nfeske, 2020-07-09:
Very very nice! Merged to staging. :-)

commit 3274c94f (2020-07-09): trace: extend for_each_subject_infos return value

Fixes #3811
commit 21e48a8e (2020-07-13): trace: extend for_each_subject_infos return value

Fixes #3811
closed 2020-07-13 by nfeske