genodelabs/genode #3813: Turn framebuffer drivers from servers into capture clients

nfeske, 2020-07-08:
With the capture session (issue #3812) provided by the nitpicker GUI server, all framebuffer drivers must be reworked to act as capture clients instead of framebuffer servers. Those drivers are fb_sdl, vesa_fb_drv, boot_fb_drv, imx8_fb_dv, intel_fb_drv, imx53_fb_drv, omap4_fb_drv, exynos5_fb_drv, pl11x_fb_drv, and rpi_fb_drv. Furthermore, all run scripts for graphical scenarios must be adjusted in the following ways.

* The drivers_interactive subsystem no longer provides a "Framebuffer" service, but it needs a "Capture" route instead. This one must be routed to a nitpicker instance.
* The nitpicker GUI server must be prevented from requesting a framebuffer session. This can be achieved by adding a `request_framebuffer="no"` config attribute.
* The nitpicker GUI server must be instructed to offer a capture service (to be used by the drivers subsystem). This can be accomplished by adding a `<capture/>` sub node its config. In the future, this node will be the natural place for expressing capture-client policies.

nfeske, 2020-07-08:
Depends on #3812.

nfeske, 2020-07-09:
The issue is fixed with commits on the branches https://github.com/nfeske/genode/commits/capture_session (genode) and https://github.com/nfeske/genode-world/commits/capture_session (genode-world) respectively.

There are few open tests that are missing. I'd be very happy about helping hands.

* The new version of the imx8_fb_drv has not been tested on the EVK yet. @skalk or @cproc or @ssumpf Could one of you please give the framebuffer.run script a try on this board?
* Although I tried to adjust the various vbox run scripts, I have executed only the bare-bones virtualbox.run scripts on Qemu. There are many more scenarios that rely on more complicated infrastructure (and VDIs). @alex-ab Shall we rely solely on the autopilot do the testing or would you try a few selected manual test runs? The latter may be worthwhile especially for the vm_stress scenarios.
* I tried to adopt and test most scenarios of genode-world. But I have not executed the scrcpy scenario. Could you give it a try with your existing Android setup?

alex-ab, 2020-07-09:
> * Although I tried to adjust the various vbox run scripts, I have executed only the bare-bones virtualbox.run scripts on Qemu. There are many more scenarios that rely on more complicated [...]

I will run some tests and report back.

alex-ab, 2020-07-09:
I do not see any output for the vbox5 run scripts. It seems that due to line https://github.com/nfeske/genode/blob/capture_session/repos/libports/src/drivers/framebuffer/vesa/main.cc#L115 the best graphic mode is not tried to detect. With the fixup commit I have output, but maybe you want to fix it differently.

nfeske, 2020-07-09:
The fixup is cool. Very pragmatic, in a good way. :-)

nfeske, 2020-07-09:
Merged to staging.

cproc, 2020-07-09:
framebuffer.run works on i.MX8 EVK board with fixup commit e4bf557

nfeske, 2020-07-09:
Thanks @cproc for testing and the fixup!

alex-ab, 2020-07-10:
> I will run some tests and report back.

The vbox5 and vm_stress scripts work now, tested on our nightly test machine.

The scrcpy scenario is not working anymore, will try to find the reason.

alex-ab, 2020-07-10:
@nfeske: if you run/sdl or run/sdl2 one gets a very tiny window. Do you have an idea ?

nfeske, 2020-07-10:
Now, the application may start before the framebuffer driver is up. If the application requests the mode once and never responds to mode changes, it will stay with a 1x1 pixel sized mode. I observed the same effect in the flif.run scenario.

The simplest solution would be the pinning the nitpicker domain to a predefined size using the `width` and `height` attributes.

In the longer term, it might be nice to enable SDL2 to gracefully deal with such an initial situation.

alex-ab, 2020-07-10:
With your suggestion the scenario boots up and is working.

nfeske, 2020-07-10:
May you share a patch for the run script that works for you (including the specific values)?

commit bcf0c73c (2020-07-09): Turn framebuffer drivers into capture clients

This patch replaces the use of the "Framebuffer" session interface by
the new "Capture" session interface in all framebuffer drivers. Thanks
to this change, those drivers have become mere clients of the nitpicker
GUI server now, and are no longer critical for the liveliness of the GUI
server.

The patch touches the following areas:

- The actual driver components. The new versions of all drivers have
  been tested on the respective hardware. Generally, the drivers
  have become simpler.

- The drivers_interactive packages for various boards. The d [... cut]
commit 0c85e0a8 (2020-07-17): pbxa9_drivers: add blit api

This is a follow-up fix of "Turn framebuffer drivers into capture
clients".

Fixes #3813
commit d7d88edf (2020-07-20): pbxa9_drivers: add blit api

This is a follow-up fix of "Turn framebuffer drivers into capture
clients".

Fixes #3813
commit bfd94d64 (2020-07-30): pbxa9_drivers: add blit api

This is a follow-up fix of "Turn framebuffer drivers into capture
clients".

Fixes #3813
closed 2020-07-13 by nfeske