genodelabs/genode #3815: Support LTE USB modems that speak the MBIM protocol

m-stein, 2020-07-09:
In the context of this issue we aim to enable using a modern LTE modem connected via USB that speaks the MBIM binary protocol. As test hardware we will use the Huawei ME 906s and a WWAN Card To USB Adapter with SIM Slot.

ssumpf, 2020-07-13:
The progress of Genode's `libmbim` port is discussed in genodelabs/genode-world#217

ssumpf, 2020-07-15:
The progress of the USB driver part can be found under: #3822

ssumpf, 2020-12-02:
@nfeske, @chelmuth: Here the initial LTE support for Genode. Since it gathers `world` and `staging` this is the place where to find everything. Please consider for staging:

From world:
https://github.com/ssumpf/genode-world/commit/c40d21c48dc76c891db3fbcdf146c77f630f7d16 (glib adaptions by @m-stein and @cnuke )
https://github.com/ssumpf/genode-world/commit/ea92209901c02bacecdba90e30880a91cae6dc9c (libmbim support by @m-stein, @cnuke, and @ssumpf)
https://github.com/ssumpf/genode-world/commit/4b12745f8a86baa31c28aaad195a7e0557669010 (mbimcli support by @ssumpf)

Prerequisites for Genode's staging:
https://github.com/ssumpf/genode/commit/6d8b3db514c54f15e0a93234e6a48eef9fb6eb93 (port of ffi for ARMv8)
https://github.com/ssumpf/genode/commit/840287c9ea2ba0b8422bdd8c4e017cd9e1f40c7c (experimental use of  no ARP support within the NIC router)
https://github.com/ssumpf/genode/commit/44cbb12e4ef25a5725b6c7ec48252b78efdc5aee (extra data + zero config support for usb)
https://github.com/ssumpf/genode/commit/05040f7c25e7dbd1f450fd70a6cc99af66b48e23 (usb modem driver for usb_host)

Related issues that might be closed:
https://github.com/genodelabs/genode/issues/3815
https://github.com/genodelabs/genode/issues/3822
https://github.com/genodelabs/genode-world/issues/168
https://github.com/genodelabs/genode-world/issues/223

My branches for this issue:
https://github.com/ssumpf/genode-world/tree/issue127_mbimcli (world)
https://github.com/ssumpf/genode/tree/issue3822_mbim (staging)

chelmuth, 2020-12-02:
After a complete review, I merged all commits for genode-world as well as

10f6bd46bbb libports: ffi for ARM_64
f14d48dbe9d dde_linux: usb support extra data and zero config

This leaves the nic_router adaptions and the commented dde_linux commit unmerged until you notify me. Sorry @m-stein for my premature note about merging the nic_router changes as is - I'm looking forward to a clean and approved commit from you.

ssumpf, 2020-12-03:
@chelmuth: I have tested everything with current staging in both `world` and `Genode` and have added two more commits:
https://github.com/ssumpf/genode/commit/3dd34fe2935dcaf08516c43be85b07e428dfc97a (adds a README to usb_modem)
https://github.com/ssumpf/genode-world/commit/7289437353b4173e74b10ddef1c8039a947efb35 (fixes bogus autofree pointers in mbimcli)

As a next step I gonna write a test that uses `curl` and DNS in order to download something via the LTE modem. After that the following issues can be closed:
https://github.com/genodelabs/genode/issues/3815
https://github.com/genodelabs/genode/issues/3822
genodelabs/genode-world#168
genodelabs/genode-world#223
genodelabs/genode-world#168
genodelabs/genode-world#217

ssumpf, 2020-12-04:
@chelmuth: https://github.com/ssumpf/genode-world/commit/864355f2f5831ca25815f7cf9dba4f7051dc70ef showcases our LTE support using `fetchurl`.

nfeske, 2020-12-07:
@ssumpf there seems to be a build problem related to the USB modem driver. May you have a look?

```
[2020-12-05 15:38:58] /usr/local/genode/tool/19.05/bin/../lib/gcc/x86_64-pc-elf/8.3.0/../../../../x86_64-pc-elf/bin/ld: drivers/net/usb/cdc_ncm.o: in function `cdc_ncm_select_altsetting':
[2020-12-05 15:38:58] /data/genode/contrib/dde_linux-17ca56af6e9fde7df9b2ab6bdaa8f9b65c3bc533/src/drivers/usb_modem/drivers/net/usb/cdc_ncm.c:1022: undefined reference to `usb_altnum_to_altsetting'
[2020-12-05 15:38:58] collect2: error: ld returned 1 exit status
[2020-12-05 15:38:58] make[3]: *** [/data/genode/repos/base/mk/prg.mk:213: usb_modem_drv] Error 1
[2020-12-05 15:38:58] make[3]: Target 'all' not remade because of errors.
[2020-12-05 15:38:58] make[2]: *** [var/libdeps:3795: usb_modem_drv.prg] Error 2

```

ssumpf, 2020-12-07:
@nfeske: cf0db0f compiles the modem driver for `-O0` runs

m-stein, 2021-01-19:
@ssumpf May we close this issue?

closed 2021-01-19 by m-stein