genodelabs/genode #3816: os: access to privileged system management API needed

skalk, 2020-07-09:
To implement certain platform driver functionality like powering of devices on certain ARM platforms access to either the ARM Trusted Firmware (ATF) or privileged-only I/O peripherals is needed. In general the shifting of powering, clocking, and pin-muxing into the ATF seems to be the future.

However, this kind of system management firmware is only accessible via secure-monitor-calls, which is a privileged instruction not accessible by a normal component like the platform driver. Therefore, an API is needed to tunnel those calls via the kernel, but it should be restricted to system managing components like the platform driver.

There is already a special privilege used by the platform driver to constrain the physical memory to be used. We can extend the `constrain_phys` attribute used in the RAM resource configuration of the platform driver by renaming it into `managing_system` and moving it into the `start` node of the driver. The sandbox library of init can use that to enforce the managing_system attribute to be used in the PD session creation as argument. Thanks to @nfeske for hinting me to that point.

The last open question is how the API should be exposed to the platform driver. I see three different possibilities here:

# Simply add a function like `manage_system_call(arg0, arg1, arg2, arg3) to the base-library
# Add that kind of function to the PD session interface
# Just execute the forbidden secure-monitor-call and emulate it in the kernel

The first option is straight forward to implement, but of course further pollutes the base-library API. The second one also means an API change, but has the advantage to be most flexible with regard to potentially other kernel-APIs that might expose such privileges to the root-task (core). Thanks to @chelmuth for your thoughts regarding this solution. The last one is somehow less explicit and more a hack, but it means no changes to the base-library's API. On the other hand it is more special and therefore slightly more complex to implement within the kernel, and completely tied to the kernel. AFAIK no other kernel (sel4, Fiasco.OC) exposes the functionality we need until now.

I'm open to implement any of the three points, and would be happy about feedback.

nfeske, 2020-07-10:
Of the three options, (2) looks most appealing to me. It keeps the management of the access control solely at the user level, which is nice. The function interface is of course pretty opaque - but so are the underying hardware/firmware mechanisms. I'd propose to wrap the args into a struct that is used as both argument and return value. So values can go in and out.

Regarding your patch, I looks nice except for the use of the `bool` constructor argument. Here the same style consideration applies as for https://github.com/genodelabs/genode/issues/3811#issuecomment-655424945. As a rule of thumb, primitive types - and `bool` in particular - should not be used as arguments of public methods, constructors / functions. Please consider introducing an enum or a compound struct to make the argument self-explanatory at the caller site.

To illustrate what I mean, use either:

```
enum class Managing_system { DENIED, PERMITTED };
...
/* caller */
Managing_system const managing_system { Managing_system::DENIED };
...
Platform_pd(label, managing_system) ...
```

or:

```
struct Attr { bool managing_system; };
...
/* caller */
bool const managing_system = false;
...
Platform_pd(label, { .managing_system = managing_system });
```

Both approaches have the effect that the intent behind the argument must be clearly spelled out at the caller site. Nowadays, I prefer the latter because it is less clunky and can in principle be extended more easily.

chelmuth, 2020-07-10:
It seems Fiasco.OC provides a special kernel object to enable components to execute `smc` on ARM https://github.com/kernkonzept/fiasco/commit/d52cf4a33742faea752e6b6e493b21d99de3a1e4#diff-c92ec6b15ac25b493c76a28007d9003f

skalk, 2020-07-10:
Thank you @nfeske and @chelmuth for your feedback! I'll follow your suggestions and rework the existing commit, as well as implement variant 2.

skalk, 2020-07-10:
I've re-implemented the original approach by adding a new PD session call. Thereby, the propagation into Platform_pd is no longer necessary. @nfeske I hope commit b586a5b fits your expectations. Commit 96fc0c3 adds the functionality needed to the base-hw kernel using ARMv8.

closed 2020-08-25 by nfeske