genodelabs/genode #3819: base: remove Cancelable_lock

alex-ab, 2020-07-14:


alex-ab, 2020-07-14:
4c5c5e82de298f058f21c3a11c59107faf207aa4 removes the Cancelable_lock support and makes the Lock implementation just usable by Mutex/Blockade. @nfeske: can you please check and propose changes. I'm not sure what to do with the ```throw Blocking_canceled``` case in lock.cc. In theory this case now should not happen, and the code can be removed. If this happens we must consider this as bug (e.g. due to kernel unblocks the thread for whatever reason), we should consider not to unwind the Applicant list. Any opinions ?

nfeske, 2020-07-14:
I like everything about your patch, not just the consolidation of `Lock` and `Cancelable_lock`, but also the removal of superfluous includes and the change of making the lock constructor explicit. Cool!

I merged your commit to staging.

nfeske, 2020-07-14:
>  If this happens we must consider this as bug [...]

I'm with you.

chelmuth, 2020-07-15:
@alex-ab please consider the following patch for *base-linux* to prevent EINTR when removing the `throw Blocking_canceled()` in *ipc.cc*.

```diff
--- a/repos/base-linux/src/lib/syscall/linux_syscalls.h
+++ b/repos/base-linux/src/lib/syscall/linux_syscalls.h
@@ -352,6 +352,11 @@ inline int lx_sigaction(int signum, void (*handler)(int), bool altstack)
 	struct kernel_sigaction act;
 	act.handler = handler;
 
+	/*
+	 * System calls should be restarted on signal occurrence and not return
+	 * with EINTR. We therefore set the SA_RESTART flag in signal handlers.
+	 */
+
 #ifdef _LP64
 	/*
 	 * The SA_RESTORER flag is not officially documented, but used internally
@@ -360,10 +365,10 @@ inline int lx_sigaction(int signum, void (*handler)(int), bool altstack)
 	 * when leaving the signal handler and it should call the rt_sigreturn syscall.
 	 */
 	enum { SA_RESTORER = 0x04000000 };
-	act.flags    = SA_RESTORER;
+	act.flags    = SA_RESTORER | SA_RESTART;
 	act.restorer = lx_restore_rt;
 #else
-	act.flags    = 0;
+	act.flags    = SA_RESTART;
 	act.restorer = 0;
 #endif
 ```

alex-ab, 2020-07-15:
> @alex-ab please consider the following patch for base-linux to prevent EINTR when removing the throw Blocking_canceled() in ipc.cc.

Thanks, please add 6bf908e0c1043df33cb50046a573bd21000cd196 and 85283f2dd991d63bba1b24ed858479af9387fc89 to staging. It first commit fixes the Blockade case where several times wakeup() are called, before the to be woken up thread actually manages to check for the outdated _owner member.

chelmuth, 2020-07-15:
@alex-ab please also address the arm_vmm build issue.

alex-ab, 2020-07-15:
> @alex-ab please also address the arm_vmm build issue.

Thanks, please add 7c2264121dd5ad657009ed4c984e676b499693b0 and cfbf6ba8b3e5460cfe5cd99870226d62073abb7d

chelmuth, 2020-07-15:
Merged all four commits to staging.

commit 7871d5f1 (2020-07-14): base: remove Cancelable_lock

- base/cancelable_lock.h becomes base/lock.h
- all members become private within base/lock.h
- solely Mutex and Blockade are friends to use base/lock.h

Fixes #3819
commit 747fbfed (2020-07-20): base: remove Cancelable_lock

- base/cancelable_lock.h becomes base/lock.h
- all members become private within base/lock.h
- solely Mutex and Blockade are friends to use base/lock.h

Fixes #3819
commit 41380ff7 (2020-07-30): base: remove Cancelable_lock

- base/cancelable_lock.h becomes base/lock.h
- all members become private within base/lock.h
- solely Mutex and Blockade are friends to use base/lock.h

Fixes #3819
closed 2020-08-25 by nfeske