genodelabs/genode #3820: sculpt_manager: add graphical CPU affinity configuration option

alex-ab, 2020-07-14:


alex-ab, 2020-07-14:
I added a commit with which one can configure graphical the CPU affinity during interactive deployment. If the system has more than one CPU, an optional CPU affinity option is shown. By default it is already defined with all available CPUs selected, as it is the case currently. With the commit now one can choose the location xpos/ypos and the dimension (width/height) graphical. I tested it with the  #3635 on 20.02 in various combinations, e.g  width/height of 2x2, 4x1, 4x2, 8x2 and unorthodox combination like 3x3. With #3635 the height represent the hyperthreads of one and the same core on base-nova. @nfeske: may you please have a look and suggest changes, I tried to adhere to the sculpt_manager flow/style hopefully.

alex-ab, 2020-07-14:
[image]
[image]
[image]
[image]

alex-ab, 2020-07-14:
The first 3 pictures are for the height case > 1, the last picture is the case height == 1 (so no hyperthreads enabled). In the latter case I dropped the terms core and thread and instead just use CPU.

nfeske, 2020-07-15:
This is extremely cool to see! Thanks for posting screen shots even.

> @nfeske: may you please have a look and suggest changes [...]

The UI and implementation looks already very nice. My suggestions are as follows:

* The CPU configuration dialog should better not be presented as a radio button but as a proper sub menu with a triangle pointing to the right to go inside, which turns into a left-pointing triangle once the dialog becomes visible. It should work similar to how to enter and leave a directory in the file browser.
* The CPUs should become squares (checkboxes) instead of circles (radio buttons). Maybe, I can even come up with a dedicated CPU icon later.
* Labels ("thread" or "core") should start with an upper-case letter.
* We may give the sub menu a more general name than "CPU affinity" so that the dialog can later host other settings like CPU priorities and RAM quota. Maybe "Resource assignment" would be a fitting name?
* Code-wise, I'd like to encourage you to split the affinity dialog functionality into a dedicated `Dialog` object that is aggregated by the popup menu. This way, we can keep the topic nicely separated from the surrounding code. To see the direction I'd like to go, please have a look at the nested aggregation of `Storage_dialog`, `Storage_device_dialog`, `Partition_dialog`, and `Fs_dialog`. Thanks to this pattern, the different concerns of the various dialog levels become nicely separated, and dialogs become very easy to reuse. E.g., think of adding an affinity settings sub dialog into the graph nodes later on. The same code can be reused.

I like the pragmatism of the UI model with selecting/deselecting nodes, staying within the limited feature set of the menu view. Later, once the menu view becomes more complete, we could revise it to become more intuitive (e.g., using a rubber box) but for now, your approach strikes a great balance.

alex-ab, 2020-07-15:
Thanks, I will rework it.

alex-ab, 2020-07-23:
I reworked the commit with 21a14d50318f72be58bf7b3078683fdb26e55f0a and tried to address your comments.

nfeske, 2020-08-14:
The code looks excellent! Thanks for the careful rework.

nfeske, 2020-08-14:
I have just tried out the new feature and it works flawlessly.

nfeske, 2020-08-14:
Merged to staging.

With the fixup https://github.com/genodelabs/genode/commit/12bb95e9e6aaa5f7cbec19e1caf3b642cda08361 the CPU matrix gets centered horizontally, which results in a more balanced look.

commit a5c12270 (2020-08-14): sculpt_manager: add graphical CPU affinity config

Fixes #3820
closed 2020-08-25 by nfeske