genodelabs/genode #3830: timeout framework: multi-threaded timeout object construction/destruction problem

cproc, 2020-07-22:
The test case in [1] creates and destroys a 'Timer::One_shot_timeout' object in a loop in a dedicated thread and schedules a timeout in between. A delay loop before the destruction of the timeout object is (currently manually) tuned so that the timeout handler is executed for around 50% of the number of loop iterations and for the other 50% the timeout object gets destroyed before the handler could be called. The observed effect is that usually the test gets stuck sooner or later with the following backtraces:

```
entrypoint thread:

#0  pseudo_end () at /.../repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
#1  0x000000005004eeda in lx_futex (val=0, op=0, uaddr=<optimized out>) at /.../repos/base-linux/src/lib/syscall/linux_syscalls.h:441
#2  thread_stop_myself (myself=<optimized out>) at /.../repos/base-linux/src/include/base/internal/lock_helper.h:62
#3  Genode::Lock::lock (this=this@entry=0x101e9f8 <Component::construct(Genode::Env&)::main+15128>, myself=...) at /.../repos/base/src/lib/base/lock.cc:128
#4  0x000000005004f3ac in Genode::Mutex::acquire (this=this@entry=0x101e9f8 <Component::construct(Genode::Env&)::main+15128>) at /.../repos/base/src/lib/base/mutex.cc:25
#5  0x00000000500ab4a0 in Genode::Mutex::Guard::Guard (mutex=..., this=<synthetic pointer>) at /.../repos/base/include/base/mutex.h:62
#6  Genode::Alarm_timeout_scheduler::_alarm_get_pending_alarm() () at /.../repos/base/src/lib/timeout/timeout.cc:237
#7  0x00000000500ab606 in Genode::Alarm_timeout_scheduler::_alarm_handle(unsigned long long) () at /.../repos/base/src/lib/timeout/timeout.cc:285
#8  0x00000000500abdf5 in Genode::Alarm_timeout_scheduler::handle_timeout (this=0x101e9e0 <Component::construct(Genode::Env&)::main+15104>, duration=...) at /.../repos/base/src/lib/timeout/timeout.cc:88
#9  0x00000000500af6f3 in Timer::Connection::_handle_timeout (this=0x101e638 <Component::construct(Genode::Env&)::main+14168>) at /.../repos/base/src/lib/timeout/timer_connection.cc:95
#10 0x0000000050045c53 in Genode::Entrypoint::Signal_proxy_component::signal() () at /.../repos/base/src/lib/base/entrypoint.cc:60
#11 0x00000000500475e0 in Genode::Meta::call_member<Genode::Meta::Empty, Genode::Entrypoint::Signal_proxy_component, Genode::Meta::Empty> (func=<optimized out>, server=...) at /.../repos/base/include/util/meta.h:462
#12 Genode::Entrypoint::Signal_proxy::Rpc_signal::serve<Genode::Entrypoint::Signal_proxy_component, Genode::Meta::Empty> (args=<synthetic pointer>..., server=...) at /.../repos/base/include/base/entrypoint.h:56
#13 Genode::Rpc_dispatcher<Genode::Entrypoint::Signal_proxy, Genode::Entrypoint::Signal_proxy_component>::_do_serve<Genode::Entrypoint::Signal_proxy::Rpc_signal> (args=<synthetic pointer>..., this=0x500fc9d0 <Genode::bootstrap_component()::startup+3472>) at /.../repos/base/include/base/rpc_server.h:146
#14 Genode::Rpc_dispatcher<Genode::Entrypoint::Signal_proxy, Genode::Entrypoint::Signal_proxy_component>::_do_dispatch<Genode::Meta::Type_list<Genode::Entrypoint::Signal_proxy::Rpc_signal> > (out=..., in=..., opcode=..., this=0x500fc9d0 <Genode::bootstrap_component()::startup+3472>) at /.../repos/base/include/base/rpc_server.h:180
#15 Genode::Rpc_dispatcher<Genode::Entrypoint::Signal_proxy, Genode::Entrypoint::Signal_proxy_component>::dispatch (out=..., in=..., opcode=..., this=0x500fc9d0 <Genode::bootstrap_component()::startup+3472>) at /.../repos/base/include/base/rpc_server.h:235
#16 Genode::Rpc_object<Genode::Entrypoint::Signal_proxy, Genode::Entrypoint::Signal_proxy_component>::dispatch (this=0x500fc948 <Genode::bootstrap_component()::startup+3336>, opcode=..., in=..., out=...) at /.../repos/base/include/base/rpc_server.h:272
#17 0x0000000050067d47 in Genode::Rpc_entrypoint::<lambda(Genode::Rpc_object_base*)>::operator() (obj=0x500fc948 <Genode::bootstrap_component()::startup+3336>, __closure=<synthetic pointer>) at /.../repos/base/include/base/mutex.h:64
#18 Genode::Object_pool<Genode::Rpc_object_base>::apply<Genode::Rpc_entrypoint::entry()::<lambda(Genode::Rpc_object_base*)> > (func=..., capid=<optimized out>, this=0x500fbe50 <Genode::bootstrap_component()::startup+528>) at /.../repos/base/include/base/object_pool.h:147
#19 Genode::Rpc_entrypoint::entry() () at /.../repos/base/src/lib/base/rpc_dispatch_loop.cc:77
#20 0x0000000050095705 in Genode::Thread::_thread_start() () at /.../repos/base/include/base/output.h:54
#21 0x00000000500aad92 in thread_start () at /.../repos/base-linux/src/lib/syscall/spec/x86_64/lx_clone.S:62
```

```
test thread:

#0  pseudo_end () at /.../repos/base-linux/src/lib/syscall/spec/x86_64/lx_syscall.S:29
#1  0x000000005004eeda in lx_futex (val=0, op=0, uaddr=<optimized out>) at /.../repos/base-linux/src/lib/syscall/linux_syscalls.h:441
#2  thread_stop_myself (myself=<optimized out>) at /.../repos/base-linux/src/include/base/internal/lock_helper.h:62
#3  Genode::Lock::lock (this=this@entry=0x403ffd60, myself=...) at /.../repos/base/src/lib/base/lock.cc:128
#4  0x000000005004f3ac in Genode::Mutex::acquire (this=this@entry=0x403ffd60) at /.../repos/base/src/lib/base/mutex.cc:25
#5  0x00000000500abb27 in Genode::Mutex::Guard::Guard (mutex=..., this=<synthetic pointer>) at /.../repos/base/include/base/mutex.h:62
#6  Genode::Alarm_timeout_scheduler::_alarm_discard(Genode::Timeout::Alarm*) () at /.../repos/base/src/lib/timeout/timeout.cc:405
#7  0x00000000500abce2 in Genode::Timeout::discard (this=this@entry=0x403ffd58) at /.../repos/base/src/lib/timeout/timeout.cc:43
#8  0x0000000001000e46 in Genode::Timeout::~Timeout (this=0x403ffd58, __in_chrg=<optimized out>) at /.../repos/base/include/timer/timeout.h:214
#9  Timer::One_shot_timeout<Thread_test::Test_thread>::~One_shot_timeout (this=0x403ffd50, __in_chrg=<optimized out>) at /.../repos/base/include/timer_session/connection.h:86
#10 Thread_test::Test_thread::entry (this=0x101ea50 <Component::construct(Genode::Env&)::main+15216>) at /.../repos/os/src/test/timeout/main.cc:811
#11 0x0000000050095705 in Genode::Thread::_thread_start() () at /.../repos/base/include/base/output.h:54
#12 0x00000000500aad92 in thread_start () at /.../repos/base-linux/src/lib/syscall/spec/x86_64/lx_clone.S:62
```

When building with -O0, sometimes a page fault occurs with the following backtraces:

```
entrypoint thread:

#0  0x000000005011e4a7 in vtable for __cxxabiv1::__si_class_type_info () from ld.lib.so
#1  0x00000000500e2a63 in Genode::Timeout::Alarm::_on_alarm (this=0x403ffdf8) at /.../repos/base/src/lib/timeout/timeout.cc:58
#2  0x00000000500e3b8b in Genode::Alarm_timeout_scheduler::_alarm_handle (this=0x102c240 <Component::construct(Genode::Env&)::main+15104>, curr_time=117145883) at /.../repos/base/src/lib/timeout/timeout.cc:310
#3  0x00000000500e2c3c in Genode::Alarm_timeout_scheduler::handle_timeout (this=0x102c240 <Component::construct(Genode::Env&)::main+15104>, duration=...) at /.../repos/base/src/lib/timeout/timeout.cc:88
#4  0x00000000500e5526 in Timer::Connection::_handle_timeout (this=0x102be98 <Component::construct(Genode::Env&)::main+14168>) at /.../repos/base/src/lib/timeout/timer_connection.cc:95
#5  0x00000000500e8d1e in Genode::Signal_handler<Timer::Connection, Genode::Entrypoint>::dispatch (this=0x102c110 <Component::construct(Genode::Env&)::main+14800>) at /.../repos/base/include/base/signal.h:486
#6  0x000000005006bf7a in Genode::Entrypoint::_dispatch_signal (this=0x50151ba0 <Genode::bootstrap_component()::startup+352>, sig=...) at /.../repos/base/src/lib/base/entrypoint.cc:81
#7  0x000000005006bdb1 in Genode::Entrypoint::Signal_proxy_component::signal (this=0x50152748 <Genode::bootstrap_component()::startup+3336>) at /.../repos/base/src/lib/base/entrypoint.cc:60
#8  0x00000000500729b3 in Genode::Meta::call_member<Genode::Meta::Empty, Genode::Entrypoint::Signal_proxy_component, Genode::Meta::Empty> (server=..., func=(void (Genode::Entrypoint::Signal_proxy_component::*)(Genode::Entrypoint::Signal_proxy_component * const)) 0x5006bce8 <Genode::Entrypoint::Signal_proxy_component::signal()>) at /.../repos/base/include/util/meta.h:464
#9  0x00000000500728bf in Genode::Entrypoint::Signal_proxy::Rpc_signal::serve<Genode::Entrypoint::Signal_proxy_component, Genode::Meta::Empty> (server=..., args=...) at /.../repos/base/include/base/entrypoint.h:56
#10 0x0000000050072762 in Genode::Rpc_dispatcher<Genode::Entrypoint::Signal_proxy, Genode::Entrypoint::Signal_proxy_component>::_do_serve<Genode::Entrypoint::Signal_proxy::Rpc_signal> (this=0x501527d0 <Genode::bootstrap_compone [... cut ...]
The page fault might be caused by an invalid handler, which might have got destroyed while a timeout was scheduled. At least I didn't see the page fault occur anymore when I added a '~One_shot_timeout()' destructor which discards the timeout first, but that did not solve the 'test getting stuck' problem.

The problem originally occurred in context of the 'Pthread_blockade', which creates and destroys 'One_shot_timeout' objects (wrapped by 'Libc::Timeout') dynamically, too.

[1] https://github.com/cproc/genode/commit/0c175a371b211ddd2f1f7c484722c5b3586c4f4f

alex-ab, 2020-07-22:
It looks like the very same issue @chelmuth reported some weeks ago already. What about avoiding using the two locks in different order, which is the root cause imo.
In '_alarm_handle (+ _alarm_get_pending_alarm())' we get the order of
``` 
L1.acquire L2.acquire L2.release L1.release
```
and in _alarm_discard the code does:
```
L2.acquire L1.acquire L1.release L2.release
```
being L1: alarm->_dispatch_mutex and L2: _mutex.

Does the commit af4e93bb1b712517ad3f4c6024494394b0183801 solve the issue, but keeps still the semantic as envisioned, ping @m-stein ping.

alex-ab, 2020-07-22:
> Does the commit [af4e93b](https://github.com/genodelabs/genode/commit/af4e93bb1b712517ad3f4c6024494394b0183801) solve the issue, but keeps still the semantic as envisioned, ping @m-stein ping ?

I fear not, 8ee5eb2471937b575c516675d6d9a024da80a0e3 is imo also required. The locking in the timeout framework should be reworked definitely.

chelmuth, 2020-07-23:
> It looks like the very same issue @chelmuth reported some weeks ago already. What about avoiding using the two locks in different order, which is the root cause imo.

Seems your memory is more retentive than mine as I can't remember this incident. Nevertheless and from the background of my current efforts to render networking with libc more robust and deadlock-free, I'd like to declare a fix for this issue as an urgent matter. So @m-stein, what's your opinion on how to address the timeout-framework locking aches?

alex-ab, 2020-07-23:
> Seems your memory is more retentive than mine as I can't remember this incident.

My memory told me just, there was something, and then I found and looked into your email to staff.

m-stein, 2020-07-23:
I'm at it...

chelmuth, 2020-07-24:
I merged 04552ebd0358173c2477b7d67cd29823cd830fb9 as `TEMPORARY` for the time being as it quick-fixes @cproc's scenario but we achieved a consensus offline that a proper fix for the timeout framework is needed. @m-stein volunteered to record the plan in a new issue.

commit 04552ebd (2020-07-24): TEMPORARY base: avoid deadlock and page fault

Fixes #3830
commit 96a7b2c8 (2020-07-24): fixup "TEMPORARY base: avoid deadlock and page fault"

Fixes #3830
commit 0f4d1c3b (2020-07-30): TEMPORARY base: avoid deadlock and page fault

Fixes #3830
commit 60106ac2 (2020-08-25): base: avoid deadlock and page fault

Fixes #3830
closed 2020-08-25 by nfeske