genodelabs/genode #3831: tool/seccomp: allow 'restart_syscall' on x86_64

cproc, 2020-07-23:
This syscall occurred while debugging with GDB. According to a related mozilla bug tracker issue [1], allowing this syscall should be safe.

[1] https://bugzilla.mozilla.org/show_bug.cgi?id=907002

chelmuth, 2020-07-24:
I merged your commit. The open question is still: Is the system call also needed on x86_32 and ARM? Would you mind to gather the required info?

cproc, 2020-08-25:
Reopening for the open x86_32 and ARM question.

nfeske, 2021-02-26:
@cproc What should happen with this issue?

cproc, 2021-02-26:
@nfeske: the syscall occurred sporadically while debugging on x86_64. To find out if it can also occur on x86_32 and ARM, one would probably need to investigate the Linux kernel source. If it doesn't have a high priority, we could close the issue for now and reopen it if the same problem is observed on x86_32 or ARM in the future.

nfeske, 2021-02-26:
Thank you for the clarification. So I'll close it for now.

commit b4d9d9aa (2020-07-24): tool/seccomp: allow 'restart_syscall' on x86_64

Fixes #3831
commit 9b6cc75f (2020-07-30): tool/seccomp: allow 'restart_syscall' on x86_64

Fixes #3831
closed 2021-02-26 by nfeske