genodelabs/genode #3834: base: statically check for reasonable Tslab block size

chelmuth, 2020-07-28:
The `Tslab<T, BLOCKSIZE>` utility is a slab allocator with entries of type `T` and the given slab-block size. The implementation should check if at least one (better eight as in the original slab paper) object of type `T` fits into the allocator.

chelmuth, 2020-07-28:
In the commit above I tested most of our platforms (kernel + board). I had to reduce the check to 2 entries because of `Cpu_thread_allocator` which uses page-sized slab blocks regardless of the actual `Cpu_thread` object size.

chelmuth, 2020-10-09:
I'd like to merge the commit above but was waiting for a note about the 2-object slab blocks. Is there any improvement possible?

nfeske, 2020-11-10:
I share your sentiment that a slab allocator with two entries per block defeats the idea of using a slab allocator in the first place. Intuitively, I'd draw the line at 8 entries or higher.

chelmuth, 2021-02-26:
@skalk could you please share your thoughts about the CPU thread allocator slab?

skalk, 2021-02-26:
@chelmuth actually a corner in base-hw, I never actively investigated before. As far as I can see, it exists more-or-less unmodified since the introduction of base-hw. The Cpu_thread_components allocated by this allocator vary in size regarding the actual kernel/architecture. On Nova/x86_64 we've a size of about 1300 bytes, on hw/x86_64 it is more than 2K, because here the whole kernel-object (register set, fpu state etc.) is part of this object too.
If I just take a Tslab here, and bump it to 8 objects used, init's ELF loading fails (on hw) due to Out_of_ram exception. Maybe, we should not use a Tslab here in general? I mean, a lot of components do not use more than 2 threads. Mostly, thread allocation/freeing does not occur frequently?! May we just use a heap here?

chelmuth, 2021-03-03:
> Maybe, we should not use a Tslab here in general? I mean, a lot of components do not use more than 2 threads. Mostly, thread allocation/freeing does not occur frequently?! May we just use a heap here?

I agree that we don't want to raise the resource demands of simple components (with just 2 threads). The heap may help as it allocates memory chunks of growing size starting at 4K.

chelmuth, 2022-12-20:
This issue was marked [fixed] on Oct 9, 2020 as the static check works as intended. Also, with 493924a35e3bca10593baec4c81313c1fb436e89 the base-hw thread allocator no longer uses `Tslab`. Is it time to merge the static check?

chelmuth, 2022-12-20:
@skalk mentioned that base-hw does not (and did never?) block this issue. So I tested a build and got stuck in base-nova.

```
    COMPILE  cpu_session_component.o                                             
In file included from .../genode.git/repos/base-nova/src/core/include/rpc_cap_factory.h:21,
                 from .../genode.git/repos/base-nova/src/core/include/pager.h:30,
                 from .../genode.git/repos/base/src/core/include/cpu_session_component.h:26,
                 from .../genode.git/repos/base/src/core/cpu_session_component.cc:21:
.../genode.git/repos/base/include/base/tslab.h: In instantiation of ‘struct Genode::Tslab<Genode::Cpu_thread_component, 4072>’:
.../genode.git/repos/base/src/core/include/cpu_session_component.h:48:30:   required from here
.../genode.git/repos/base/include/base/tslab.h:26:28: error: static assertion failed
   26 |  static_assert(sizeof(T)*8 <= BLOCK_SIZE);                               
      |                ~~~~~~~~~~~~^~~~~~~~~~~~~                                 
make[2]: *** [.../genode.git/repos/base/mk/generic.mk:62: cpu_session_component.o] Error 1
make[1]: *** [var/libdeps:77: core-nova.lib] Error 2                             
make: *** [Makefile:336: gen_deps_and_build_targets] Error 2 
```

alex-ab, 2023-01-03:
> @skalk mentioned that base-hw does not (and did never?) block this issue. So I tested a build and got stuck in base-nova.

It seems to be an _all other kernels beside hw_ problem. All kernels beside hw fail with this message. The reason is that base-hw uses a separate/own Cpu_thread allocator implementation, whereby the other uses the default allocator in base which is based on Tslab. (see [base-hw](https://github.com/genodelabs/genode/blob/master/repos/base-hw/src/core/cpu_thread_allocator.h) vs [generic](https://github.com/genodelabs/genode/blob/master/repos/base/src/core/include/cpu_thread_allocator.h)).

alex-ab, 2023-01-03:
I created a list of the various sizes of the Cpu_thread object to get an idea whether a Tslab is still useful or not, yes, depends:
```
sizeof() of Cpu_thread_component and Platform_thread

-- 32bit --

nova      x86_32:   820   72
okl4      x86_32:   916   52
pistachio x86_32:   920   56 
fiasco    x86_32:   940   76 
fiasco.oc x86_32:  1016  112
sel4      x86_32:  1084  216

hw        arm_v7a: 1568  848

-- 64bit --

nova      x86_64:  1360   96
fiasco.oc x86_64:  1576  168
sel4      x86_64:  1600  248
hw        x86_64:  2776 1610

hw        arm_v8a: 2872 1712
```

alex-ab, 2023-12-04:
No objections from my side

commit 0046edf7 (2023-12-13): base: statically check for reasonable Tslab block size

Per default Tslab checks that 8 slabs incl. overhead fit into one block.
If this is not desired the template parameter 'MIN_SLABS_PER_BLOCK' can
be used to control the minimum number of blocks.

Fixes #3834