genodelabs/genode #3837: sculpt_manager: support CPU service announcement and routing

alex-ab, 2020-07-30:


alex-ab, 2020-07-30:
6dd5e73c4e3cdcc2487a45cc63772de0ba6a16bd adds the feature to depot_deploy and sculpt_manager. 7dfeb32b4481289376cd8d046b1f7097d8e19f9c is an option, in case we want not to bother the user with selecting a CPU service if there is actually no choice (if solely the parent provide it).

nfeske, 2020-09-10:
I'm planning to eventually merge the PD service with the CPU service (once we manage to eliminate the use of multiple CPU sessions by VirtualBox). Since only the PD service will prevail in the long run, I would not introduce the notion of a CPU session to the Sculpt manager. We may instead consider adding the option to route the PD session. If specified, the routing would be applied to both the CPU session and PD session.

nfeske, 2021-02-25:
Commit https://github.com/nfeske/genode/commit/2b3c0f5b2e87fc06ba53f6ec4c371f82d1936248 implements the proxy idea that we discussed. I named the proxy component "shim".

Commit https://github.com/nfeske/genode/commit/5e360035f64b6d13ae9931e92a574b570331af0d adjusts the cpu_balancer.run script.

With the two commits, I don't see the error anymore but I'm not 100% sure if the output is the expected one. When enabling the verbose mode, something useful seems to happen ("useful" as far as burning CPU cycles may get). The run script does not finish on Qemu/NOVA/x86_64 though. So something may still be missing.

alex-ab, 2021-03-01:
With commits 3c9e73db39adda23c7e4a5c75e6deeb08d3666f2, c83b92285fc7f328d090a200f7e044967e05b888, 56a879b65fbce5076522cc3d521da84d6e03616 a component now can provide pd and cpu service, but solely the pd service is shown in the sculpt manager configuration. Additionally, with the shim commits of #4029 and the (unfinished) prototype a7c254db9d897983e66776120860b4f21c06af85 to integrate the shim in sculpt_manager, the target of the issue becomes usable.

nfeske, 2021-03-01:
Commit https://github.com/genodelabs/genode/commit/3c9e73db39adda23c7e4a5c75e6deeb08d3666f2 looks very good to me.

Regarding https://github.com/genodelabs/genode/commit/a7c254db9d897983e66776120860b4f21c06af85, could you try implementing the insertion of the shim at the depot-deploy level (when generating the runtime config from the deploy config)? If this works out, the solution would make the manual editing of the deploy config safer and easier (the user won't need to know the technicalities about the shim). It would also make the feature available to users of the standalone depot-deploy component.

alex-ab, 2021-03-02:
With commit 0d730d1937ebd69b34654a5676c9a90eb4a2d050 the shim is solely inserted at the depot-deploy level. I  required strip the leading "child ->" from session requests of the shim client (2de2b14d87ff7aacd3f2e303b945c2c21b8d4dc0) for the already configured routes by the sculpt_manager. We may live without 2de2b14d87ff7aacd3f2e303b945c2c21b8d4dc0, but then I would have to rewrite all routes and prefixing with "child -> " in depot_deploy. What do you think ?

alex-ab, 2021-03-02:
issue_3837 contains now all changes. b6c09179e60ee86e8c7ea684a738a88198e55c3c contains the version without using 'child' as label of the shim, and accordingly #4029 commits are 4b0d4d66166099c566a60aa546db951a3005f1ba and d3f87d3393c21d0f6c775deb32f19ac3f6971267

nfeske, 2021-03-02:
Good work! I'm glad that it worked out so well.

I left a couple of remarks on the depot-deploy patch. These are little things. Over time while further maintaining the code, those little things make a difference though, so they are important to me. I hope you don't feel bothered by the nitpicking.

alex-ab, 2021-03-03:
All fine, I adjusted the commits on issue_3837 accordingly

nfeske, 2021-03-03:
With the latest adjustments, the patch looks excellent!

Can you recommend me a way of how I can try out the new PD-routing mechanism in Sculpt? I'd love to give the new dialog a try.

alex-ab, 2021-03-03:
You may remove the "count_pd_services" check in sculpt_manager/view/pd_route_dialog.cc in the beginning of the file, then the dialog shows up also if you have no service provider. Beside that, you would need to deploy a package providing PD/CPU service. I can do so with the cpu_balancer in the meantime for the 21.03 index.

alex-ab, 2021-03-03:
I rebased everything to staging, incorporated your comments, squashed the fixup together and published two packages under 21.03 for test driving the dialog - branch issue_3837_sculpt.

nfeske, 2021-03-03:
I just gave the feature a try on Sculpt. Apparently, the route for the shim binary was not yet generated. I added the fixup https://github.com/nfeske/genode/commit/4bc072a8d61481daf0170cd457fe7bbba3ab485b for it. The fixup also improves the formatting of the generated /config/managed/runtime XML. I also renamed the flag to `route_binary_to_shim` to draw the connection to the shim more closely.

With the fixup, I could successfully start the top view with the PD routed to the CPU balancer. The ref-account warnings by top_view appear in the log, which is logical. Despite the warnings, it seems to work perfectly, which surprises me.

I merged the commit series to staging.

As one remaining detail, the PD selection dialog does not open at the first mouse click. One has to click twice on the "Protection domain" item. This is always reproducible. @alex-ab Could you have a look?

alex-ab, 2021-03-04:
> As one remaining detail, the PD selection dialog does not open at the first mouse click. One has to click twice on the "Protection domain" item. This is always reproducible. @alex-ab Could you have a look?

Commit 3e02faaa465a5bf41db18b8f18964a275f836be5 solves the twice click issue.

nfeske, 2021-03-05:
Works perfect now. I merged your fixup to staging.

nfeske, 2021-03-12:
Fixed in master.

closed 2021-03-12 by nfeske