genodelabs/genode #3842: nova: add support to migrate global threads

alex-ab, 2020-07-31:


alex-ab, 2020-09-22:
@nfeske, @chelmuth: please consider for review

nfeske, 2020-09-22:
The changes look excellent! It is great to see also the collateral code improvements (like swapping out pointers for references). I have no reservation against merging the changes is as.

As one minor question, since the commit "base: add thread migration test" touches core, would it make sense to split it into two commits, one for the core-related changes, and one for the actual test?

alex-ab, 2020-09-23:
@nfeske: thx, I rebased + adjusted the commits to your suggestions.

alex-ab, 2020-09-23:
@nfeske: The factored out migration test into a standalone test is ready.

nfeske, 2020-09-23:
@alex-ab Superb! I merged your commits to staging.

commit 74627736 (2020-09-23): nova: add migration support for global threads

Fixes #3842
commit 99fa2036 (2020-10-09): nova: add migration support for global threads

Fixes #3842
closed 2020-10-09 by chelmuth