genodelabs/genode #3843: A dynamic thread scheduler

alex-ab, 2020-08-07:
The idea is to have an optional CPU service, let it name cpu_scheduler atm, which can be used to migrate threads dynamically, depending on some policy. For components, where this dynamic migration is desired, the CPU session creation requests are routed to the cpu_scheduler service. The cpu_scheduler may now monitor the utilization of the clients and their threads, idle utilization of all CPUs and depending on some configured policy, migrate threads over CPUs. This issue is currently WIP.

alex-ab, 2020-11-12:
@nfeske: the issue_3843 branch contains the last state and I renamed everything to cpu_balancer. The commits 1c739e37f793401ad2064cb32dde4e79a8557433 and 6d027cbdf5a27adab221fa5527ff8a1fbd87bec7 are the essential commits. The remaining two are for a graphical frontend of the cpu_balancer, so they are optional but handy on Sculpt.

nfeske, 2020-11-16:
Thanks @alex-ab for pointing me to the right commits. The feature and its implementation looks really cool! I merged both commits to staging. May you nevertheless consider the following points as fixups?

* A few file headers are missing (config.h) or incomplete (lacking a brief description)
* I wonder, is it possible to replace the uses of `List` with `Registry`? E.g., Sleeper_list, Child_list, ThreadX list. This would potentially eliminate the use of pointers altogether.
* Would you consider removing the commented-out (`#if 0`) code parts?
* The name `ThreadX` does not follow the coding style. It would be nice to give it a name with an intuitive meaning.

At a few places, the code constructs/destructs a `Constructible` depending on a boolean value. Since this is a common pattern, there exists the `Reconstructible::conditional` method. Maybe it could be useful in some of these cases?

alex-ab, 2020-11-17:
Thanks @nfeske, I will try to address your comments asap.

alex-ab, 2020-11-19:
@nfeske: b3f2d5e6304055dc767f2309d530133a5272ffd7 contains your suggestions

nfeske, 2020-11-19:
Thanks @alex-ab, I am delighted! :-)

nfeske, 2020-11-27:
Given that the CPU balancer is part of Genode 20.11, should we close this issue?

alex-ab, 2020-11-27:
Yes. The additional graphical GUI frontend I will continue separately.

closed 2020-11-27 by alex-ab