genodelabs/genode #3847: zircon PS/2 driver: deadlock ahead

nfeske, 2020-08-13:
During my course of changing Genode's input drivers to become event clients (#3827), I noticed that the zx_pc_ps2 driver no longer works:

```
[init -> ps2_drv] Error: deadlock ahead, mutex=0x10bcb48, return ip=0x1032c92
[init -> test-input] --- Input test ---
[init -> ps2_drv] ZIRCON: i8042 device (unknown)
[init -> platform_drv] PS2 uses IRQ, vector 0xc
page fault, pd='init -> ps2_drv' thread='i8042-init' cpu=0 ip=0xd4a52 address=0x403fdff8 stack pointer=0x403fe000 qualifiers=0x6 irUWp reason=1
```

As this driver is not in regular use, I presume that this regression remained unnoticed for a long time.

How should we go about it? I'm particularly interested in @jklmnn's opinion about the future plans for the dde_zircon repository. There has not been much activity in this repository after the initial merge two years ago.

jklmnn, 2020-08-13:
Unfortunately I currently lack the time to fix this properly. Since the Zircon sources this driver uses are also outdated I'd suggest to remove it. Updating it to the latest Zircon (I didn't follow the development, but there have been major changes in their code base) and updating it to the latest Genode would probably require effort similar to a complete rewrite. So if someone actually needs this they'd have to rewrite it anyway.

nfeske, 2020-08-13:
Thank you @jklmnn for clarifying the situation so quickly. I followed your advice of removing dde_zircon. Should we decide to revive it later on, the code is still preserved as part of our history.

commit cd583811 (2020-08-13): Remove dde_zircon

Fixes #3847
commit 472bad8d (2020-08-14): Remove dde_zircon

Fixes #3847
closed 2020-08-25 by nfeske