genodelabs/genode #3848: Ada/SPARK: Exp_Int runtime support and strict warnings

m-stein, 2020-08-14:
The CBE project has raised the need for conventions regarding compiler warnings and style checks and the need for runtime support for the Exp_Int package (exponential function on integers).

nfeske, 2020-08-14:
Merged to staging.

cnuke, 2020-08-17:
@nfeske please merge f34a0be, 9a71936 and 3484613 to staging.

Rather than disabling the style checks I opted for fixing the code. After all, those tests are probably looked at first and should reflect the style that is enforced by the flags.

m-stein, 2020-08-17:
@cnuke Thanks a lot for doing this tedious work! I totally agree with your motivation.

nfeske, 2020-08-17:
That's very much appreciated. Merged to staging.

cnuke, 2020-08-18:
@nfeske please also merge e413a98 - apparently I missed adding this change yesterday.

nfeske, 2020-08-18:
No problem. I merged your fixup just now.

commit 6544d08b (2020-08-14): ada-runtime: add s-expint.ad*

* switch to fork of the Ada runtime provided by m-stein
* switch to a Ada-runtime state that provides the Exp_Int package (exponential
  function on integers)
* adapt spark lib, symbols, and recipes to incorporate the Exp_Int package

Fixes #3848
closed 2020-08-25 by nfeske