genodelabs/genode #3850: aes_cbc_4k: api recipe and import file

m-stein, 2020-08-14:
In the context of the CBE project, an API recipe for the aes_cbc_4k library became necessary. Also, dependent targets and libraries should import include directories of aes_cbc_4k.

nfeske, 2020-08-14:
In contrast to port hashes, recipe hashes should never checked in. They are updated (semi-)automatically when updating the master branch.

I'll remove the hash parts of your commits manually.

nfeske, 2020-08-14:
Please forget the comment above. ;-) The hashes are new. So they must be part of the commit.

nfeske, 2020-08-14:
Merged to staging.

m-stein, 2020-08-14:
@nfeske Thanks for reviewing and merging ;)

m-stein, 2020-08-14:
@nfeske With that, the CBE can now be built on clean genode/staging, the bash problems in the CBE tests have also disappeared :)

nfeske, 2020-08-14:
Thanks for the good news!

commit d9177b08 (2020-08-14): aes_cbc_4k: api recipe

Fixes #3850
closed 2020-08-25 by nfeske