genodelabs/genode #3866: Base-hw: W

zeehha, 2020-08-24:
Hi,

regarding the Armv8 virtualization on Base-hw, I think a misbehaving guest might be able to tie up the system. 

As far as I can see: 
When the world switcher[1] switches to the VM, it relies on an external event or an IRQ (timer interrupt) targeting the VM (as HRC_EL2.IMO is set, IRQs are routed to EL2) to be invoked again. However, a guest which does not generate a timer interrupt might be never preempted, since there is no EL2 timer set.

Sorry, in case I miss something.

Best,
Chris

[1] repos/base-hw/src/core/spec/arm_v8/virtualization

skalk, 2020-08-25:
@zeehha: you're right a VM exit is dependent on an exception, either an abort, hypercall, or interrupt. However, the base-hw kernel uses the EL1 physical timer to do its scheduling. Whenever, the time for the VM has elapsed, it produces an interrupt. The VM is using the EL1 virtual timer, and should not be able to alter the EL1 physical timer state. Therefore, the CNTHCTL_EL2 is set to 0b100 whenever we switch to a VM, and reset to 0b111 afterwards. Have a look at: repos/base-hw/src/core/spec/arm_v8/virtualization/exception_vector.s for more details.

Did you observed a system freeze, or do you have any conrete example that validates your assumption?

zeehha, 2020-08-25:
@skalk thank you for the answer. this clarifies my misunderstanding. I was missing, that Linux uses the virtual timer..

Best,
Chris

closed 2020-08-25 by zeehha