genodelabs/genode #3869: Genode::Connection::session called on an unitialized object.

tworaz, 2020-08-25:
I spotted this when compiling Genode code with clang:

genode/repos/base/include/pd_session/connection.h:35:31: error: base class 'Genode::Connection<Genode::Pd_session>' is uninitialized when used here to access 'Genode::Connection<Genode::Pd_session>::session' [-Werror,-Wuninitialized]
                Connection<Pd_session>(env, session(env.parent(),

I think this is a valid concern. A non static Genode::Connection::session method (which does actually touch "this") is called before the constructor of the Genode::Connection has been called.

nfeske, 2020-08-28:
You encountered a old wart that is not clean but - in the name of careful API evolution - actually intended. ;-)

Let me explain the background story. With the transition of the parent interface to work asynchronously, we had to solve the problem of storing the session arguments somewhere and settled on https://github.com/genodelabs/genode/blob/staging/repos/base/include/base/connection.h#L114. The session call has the side effect of populating this array. Since the session call happens before the `Connection` constructor is executed, this operation is indeed performed on an unconstructed object but on a part that we designated for this purpose.

This - of course - calls for a cleanup. But this ultimately effects all connection classes. While changing these, it would make sense to take the chance of also updating the way of how session-arguments are marshalled at once, so that we do not confront Genode users with multiple subsequent API changes at the same corner of the framework. Regarding the session arguments, I want to get rid of the use of the `Arg_string` syntax and the use of format strings but just use XML. So we will retain only a single parser in Genode. But this change is even more far reaching and not completely carved out. So we haven't tackled it yet.

 It is definitely on my list to clean up.

tworaz, 2020-08-28:
Thanks for the explanation. I tinkered with this for a while and I also don't see a way to fix this without changing API a bit. For now it does not seem to be a big issue for me I can just disable the problematic warning and move forward with my work.

commit 1ed864e7 (2023-03-06): base: remove format strings from API

This patch ultimatedly removes format strings from Genode's base API.
Users of the former base/snprintf.h and base/console.h headers may
use the free-standing 'format' library hosted in the ports repository.

Fixes #2064
Fixes #3869
commit c26b6ae2 (2023-03-07): base: remove format strings from API

This patch ultimatedly removes format strings from Genode's base API.
Users of the former base/snprintf.h and base/console.h headers may
use the free-standing 'format' library hosted in the ports repository.

Fixes #2064
Fixes #3869
closed 2023-03-13 by chelmuth