genodelabs/genode #3871: Fix performance loss with Enhanced SpeedStep Technology on Intel i7-8665U

m-stein, 2020-08-26:
One month ago, I received my new Lenovo Thinkpad T490 with an Intel i7-8665U CPU. After having installed Sculpt I noticed that the performance was very bad, almost worse than on my 8 years old T430. So, I started to play around with BIOS settings and indeed I could improve the situation through this means. Especially disabling the "Intel Enhanced SpeedStep" turned out to be very effective.

Therefore, I started a more detailed performance test. On a Xubuntu 20.04, I measured how long it takes to compile Genodes Virtualbox port with different BIOS/Kernel settings. I did this once with a native Xubuntu installation (base line) and once with Xubuntu as guest system in Virtualbox in a self-built Sculpt (genode/master at that time). These were the results:
```
Miti = Linux Mitigations für Spectre/Lockdown etc.
SpSt = Intel Enhanced SpeedStep Technology
HyTh = Hyperthreading

T490 (i7-8665U)
  Sculpt VBox Guest Xubuntu 20.04
    Miti 1, SpSt 1, HyTh 0      --	704 sec 100% 218% 376%  (CPU fan never noticable)
    Miti 1, SpSt 0, HyTh 0	--	375 sec  53% 116% 200%  (CPU fan always on high frequency)
    Miti 0, SpSt 0, HyTh 0	--	376 sec  53% 116% 201%  (CPU fan always on high frequency)
    Miti 0, SpSt 0, HyTh 1	--	322 sec  45% 100% 172%  (CPU fan always on high frequency)
  Native Xubuntu 20.04
    Miti 1, SpSt 1, HyTh 0	--	255 sec  36%  79% 136%  (CPU fan only on demand)
    Miti 1, SpSt 1, HyTh 1	--	213 sec  30%  66% 113%  (CPU fan only on demand)
    Miti 0, SpSt 1, HyTh 1	--	187 sec  26%  58% 100%  (CPU fan only on demand)
```
Note that switching on/off SpeedStep doesn't make a difference on native Xubuntu for reasons I haven't investigated yet. However, these results indicate to me that Intel Enhanced SpeedStep Technology requires OS support and that, if the OS lacks this support, the CPU always runs in high performance mode (of which I don't know if it is meant to be enabled all the time). On the other hand, if SpeedStep is disabled, performance drops tremendously.

That said, I would try to add SpeedStep support to Genode as soon as possible, so, that Sculpt becomes enjoyable again on my new machine.

chelmuth, 2020-08-26:
Also Windows 10 users seem to have problems with fan noise if the CPU temperature exceeds 50°C on T490 https://forums.lenovo.com/t5/ThinkPad-T400-T500-and-newer-T/T490-BIOS-Update-1-49-results-in-100-fan-speed/td-p/4485779

nfeske, 2020-08-27:
Thanks @m-stein for reporting. I'm very curious.

m-stein, 2020-08-31:
@chelmuth Thanks for the hint! For now I would assume that I don't have any issues with the hardware/bios as Xubuntu 20.04 works fast without fan noises.

m-stein, 2020-08-31:
I read a bit [here](https://www.kernel.org/doc/html/v4.12/admin-guide/pm/cpufreq.html) and [here](https://01.org/linuxgraphics/gfx-docs/drm/admin-guide/pm/intel_pstate.html) and found the following:

With my T490 Xubuntu 20.04, the intel_pstate driver gets loaded as scaling driver for cpufreq and for all CPUs. The intel_pstate driver ignores the generic cpufreq pstate governors and has its own specific governors "power_saving" and "performance" where "power_saving" is more like the generic "schedutil" or "ondemand" governor than the generic "power_saving" governor.

The intel_pstate driver is in active mode and enables hardware pstate management (HWP):
```
$ sudo less /var/log/syslog | grep -e "intel_pstate" -e "cpufreq"
Aug 31 11:17:51 squid set-cpufreq[719]: Setting powersave scheduler for all CPUs
Aug 31 14:33:49 squid systemd[1]: Starting LSB: Load kernel modules needed to enable cpufreq scaling...
Aug 31 14:33:49 squid loadcpufreq[21864]:  * Loading cpufreq kernel modules...
Aug 31 14:33:49 squid loadcpufreq[21864]:    ...done.
Aug 31 14:33:49 squid systemd[1]: Started LSB: Load kernel modules needed to enable cpufreq scaling.
Aug 31 14:33:49 squid cpufrequtils[21987]:  * CPUFreq Utilities: Setting ondemand CPUFreq governor...
Aug 31 14:33:49 squid cpufrequtils[21987]:  * disabled, governor not available...
Aug 31 14:33:49 squid cpufrequtils[21987]:    ...done.
Aug 31 14:35:57 squid systemd[1]: Starting LSB: Load kernel modules needed to enable cpufreq scaling...
Aug 31 14:35:57 squid kernel: [    0.933704] intel_pstate: Intel P-state driver initializing
Aug 31 14:35:57 squid kernel: [    0.934531] intel_pstate: HWP enabled
Aug 31 14:35:57 squid loadcpufreq[694]:  * Loading cpufreq kernel modules...
Aug 31 14:35:57 squid loadcpufreq[694]:    ...done.
Aug 31 14:35:57 squid systemd[1]: Started LSB: Load kernel modules needed to enable cpufreq scaling.
Aug 31 14:35:57 squid cpufrequtils[889]:  * CPUFreq Utilities: Setting ondemand CPUFreq governor...
Aug 31 14:35:57 squid cpufrequtils[889]:  * disabled, governor not available...
Aug 31 14:35:57 squid cpufrequtils[889]:    ...done.
Aug 31 14:36:02 squid set-cpufreq[697]: Setting powersave scheduler for all CPUs
```
This means, the intel_pstate driver, only once at startup, "will set the processor's Energy-Performance Preference (EPP) knob (if supported) or its Energy-Performance Bias (EPB) knob (otherwise) to whatever value it was previously set to via sysfs (or whatever default value it was set to by the platform firmware)." From there on, the processor selects P-states by itself. The driver can, however, still give hints to the processor's internal P-state selection logic.

So, I assume this is what I should do in Genode on startup to receive the same result.

m-stein, 2020-08-31:
I forgot that the manual also says, that the driver has to enable HWP before setting EPP and EPB.

m-stein, 2020-08-31:
According to 'arch/x86/include/asm/cpufeatures.h' in the Linux code, the presence of HWP and its features is denoted in CPUID level 0x6 (EAX), word 14, bits 7..11:
```
#define X86_FEATURE_HWP			(14*32+ 7) /* Intel Hardware P-states */
#define X86_FEATURE_HWP_NOTIFY		(14*32+ 8) /* HWP Notification */
#define X86_FEATURE_HWP_ACT_WINDOW	(14*32+ 9) /* HWP Activity Window */
#define X86_FEATURE_HWP_EPP		(14*32+10) /* HWP Energy Perf. Preference */
#define X86_FEATURE_HWP_PKG_REQ		(14*32+11) /* HWP Package Level Request */
```

m-stein, 2020-09-03:
Unfortunately, the CPU fan stays active all the time on Genode (I tested 30 sec idle periods) similar to when I simply disabled SpeedStep in BIOS.

m-stein, 2020-09-03:
I did the same test again several times. This time I had equally aligned test functions (4096 byte) and I also measured how long the fan keeps blowing once the test load is through (userland only). Furthermore, I found that the EE bit isn't set initially anyways, so I removed it from the tests and I needed a longer test load to get stable results with the fan so I calculated more prime numbers:
```
dumb calculation of prime numbers <= 400000:
   xubuntu 20.04:        ~19.300 ms, fan ~35 +- 5 sec
   hw kernel (hwp=1):    ~39'500 ms
   hw userland (hwp=1):  ~15'550 ms, fan ~35 +- 2 sec
```
The different alignments obviously were not the cause of the different results in kernel and userland.

m-stein, 2020-09-04:
I started another test in which I changed the EPP and EPB values (hints for the HWP which performance profile to use). Linux only sets one of these two according to the CPUID value.
```
epp = energy performance preference (msr 0x774, bits 24-31)
   0 = performance
   8 = balanced
   15 = power saving
epb = energy performance bias (msr 0x1b0, bits 0-7)
   0 = performance
   128 = balanced
   255 = power saving

dumb calculation of prime numbers <= 400000:
   hw userland (hwp=1, ee=0, epp=0,   epb=0):  ~15'560 ms, fan ~32 sec
   hw userland (hwp=1, ee=0, epp=?,   epb=15): ~15'622 ms, fan ~50 sec
   hw userland (hwp=1, ee=0, epp=255, epb=?):  ~15'790 ms, fan ~50 sec
   hw userland (hwp=1, ee=0, epp=255, epb=15): ~15'785 ms, fan ~58 sec
```
I don't know why the fan blows longer with power-saving settings but I think, for now, I will stay with always setting epp=0, epb=0.

m-stein, 2020-09-04:
I plan to move the settings discussed above to a core-specific code path of Bender (the MSRs have to be set at each core). I think I'll hard-code the settings and trigger them only when CPUID yields the value that my T490 yields.

m-stein, 2020-09-07:
This is a first version for NOVA that triggers HWP only for the one CPUID of my T490:
https://github.com/m-stein/NOVA/commit/f61175eeb34aede65043f801d80f6835a0e1912b Setup Intel HWP for i7-8665U cores

I'm working on a version that enables HWP for all cores that support it.

m-stein, 2020-09-07:
Ok, m-stein/NOVA@f61175e is obviously bogus as I have used CPUID without knowing what I'm doing.

alex-ab, 2020-10-28:
I reworked the bender/microcode tools as discussed offline. The microcode feature is now an option of bender, the same as @m-stein HWP work. @m-stein, please check whether this still works for you (bd772f69de87b42b0422922428c00c24763bbe23 and f92da229ba4a756182a9839e3ca18bf11abcb652, https://github.com/alex-ab/morbo/commits/intel_hwp)

m-stein, 2020-10-28:
@alex-ab This looks great, thanks for your efforts! (I'd merely suggest to rename the hwp file/plugin/option intel_hwp to make it more clear). I'll test it as soon as I'm back in the office next week.

alex-ab, 2020-11-06:
According to @m-stein the adjusted bender works for him. I adjusted the commit to rename everything from hwp to intel_hwp with 008086ff665a7e4cbe96314049f893f2aba76271 and set this issue as fixed.

nfeske, 2020-11-10:
Thanks for the green light, @alex-ab. Merged to staging.

chelmuth, 2020-11-11:
For some reason I can't pin yet 5019334348065c9978bd89a5a821c9d369af160b breaks OKL4 on Qemu x86_32. @alex-ab @m-stein any ideas?

alex-ab, 2020-11-11:
I will investigate. Feel free to take the commit from staging for the moment.

commit 50193343 (2020-11-10): Update to bender with optional microcode and HWP

plugin support.

Fixes #3871
closed 2020-11-27 by chelmuth