genodelabs/genode #3872: hw: Nvme::Controller::Initialization_failed

ssmore98, 2020-08-29:
Hi:

I am new to Genode. When I tried to use the nvme_drv driver, I am having problem initializing the controller. It results in the nvme_drv exiting. The qemu command is:

`qemu-system-x86_64 -enable-kvm -m 8G -serial mon:stdio -machine pc-q35-1.4 -acpitable file=/usr/share/qemu-kvm/acpi-dsdt.aml -device virtio-scsi-pci,id=scsi0 -drive file=/genode/tsss/genode/build/x86_64/var/run/tsss.img,if=none,format=raw,discard=unmap,aio=native,cache=none,id=scsi0,media=disk -device scsi-hd,drive=scsi0,bus=scsi0.0 -vga std -drive id=nvme0,file=/genode/tsss/genode/build/x86_64/bin/nvme0.raw,format=raw,if=none -device nvme,drive=nvme0,serial=nvme0,id=nvme0n1`

The output is:

```
Bender: Hello World.

kernel initialized
ROM modules:
 ROM: [0000000001dcd000,0000000001dce742) LICENSE.bitstream-vera
 ROM: [0000000003a78000,0000000003a8818c) Vera.ttf
 ROM: [0000000003996000,00000000039a2048) VeraMono.ttf
 ROM: [0000000001d14000,0000000001d3bc00) acpi_drv
 ROM: [0000000002e7b000,000000000362f800) bash.tar
 ROM: [00000000039a6000,0000000003a3c8b8) bootstrap-hw-pc.o
 ROM: [0000000001e62000,0000000001e64acb) config
 ROM: [00000000004f0000,00000000008d1ad0) core-hw-pc.o
 ROM: [0000000000030000,0000000000031000) core_log
 ROM: [0000000002085000,0000000002c8d000) coreutils.tar
 ROM: [0000000002c8d000,0000000002c8e268) drivers.config
 ROM: [00000000039a3000,00000000039a57df) en_us.chargen
 ROM: [0000000001d3c000,0000000001d3c033) fb_drv.config
 ROM: [0000000003a89000,0000000003a8902d) focus
 ROM: [0000000000954000,0000000000974b50) fs_rom
 ROM: [0000000001da6000,0000000001dccd00) gui_fb
 ROM: [00000000003e5000,0000000000444800) init
 ROM: [0000000001e60000,0000000001e61cb6) init.xsd
 ROM: [0000000001e74000,0000000001ea8070) input_filter
 ROM: [0000000002e0f000,0000000002e0f300) input_filter.config
 ROM: [0000000003630000,0000000003725520) ld.lib.so
 ROM: [0000000001eaa000,0000000002084c38) libc.lib.so
 ROM: [0000000003a3d000,0000000003a77d40) libm.lib.so
 ROM: [0000000001dcf000,0000000001dd71e0) libuuid.lib.so
 ROM: [00000000008f9000,00000000009535e8) ncurses.lib.so
 ROM: [0000000000480000,00000000004ef9f0) nitpicker
 ROM: [0000000001dda000,0000000001e10310) nvme_drv
 ROM: [0000000001d4e000,0000000001da5a18) platform_drv
 ROM: [000000000002e000,000000000002f000) platform_info
 ROM: [00000000008d2000,00000000008f8c38) pointer
 ROM: [0000000001e65000,0000000001e73fe0) popt.lib.so
 ROM: [0000000001d3d000,0000000001d4df98) posix.lib.so
 ROM: [00000000003b6000,00000000003e4678) ps2_drv
 ROM: [0000000002dee000,0000000002e0e520) report_rom
 ROM: [0000000001dd8000,0000000001dd8542) special.chargen
 ROM: [000000000375b000,0000000003905ab0) stdcxx.lib.so
 ROM: [0000000001e11000,0000000001e4ed98) terminal
 ROM: [0000000001ea9000,0000000001ea90bd) terminal.config
 ROM: [0000000002e66000,0000000002e7af08) timer
 ROM: [0000000002c8f000,0000000002ded4f0) usb_drv
 ROM: [0000000001dd9000,0000000001dd9007) usb_policy
 ROM: [0000000002e10000,0000000002e65510) vesa_fb_drv
 ROM: [0000000003726000,000000000375a5f0) vfs
 ROM: [0000000003906000,00000000039952a8) vfs.lib.so
 ROM: [0000000001e50000,0000000001e5fcc0) vfs_pipe.lib.so
 ROM: [0000000000445000,000000000047fc50) vfs_ttf.lib.so
 ROM: [0000000000975000,0000000001d13800) vim.tar
 ROM: [0000000001e4f000,0000000001e4f059) vimrc

Genode 20.05
8128 MiB RAM and 64533 caps assigned to init
[init] parent provides
[init]   service "ROM"
[init]   service "LOG"
[init]   service "RM"
[init]   service "CPU"
[init]   service "PD"
[... 168 lines cut ...]
[init -> drivers -> fb_drv] OEM: SeaBIOS VBE(C) 2011
[init -> drivers -> fb_drv] Found: physical frame buffer at 0xfd000000 size: 16777216
[init -> drivers -> fb_drv] fb mapped to 0xb000000
[init -> drivers -> fb_drv] using VESA mode 640x480@32
[init -> vfs] tar archive 'bash.tar' local at 0xc39f000, size is 8081408
[init -> drivers] child "input_filter" announces service "Input"
[init] child "vfs" announces service "File_system"
[init] child "vfs_rom" announces service "ROM"
[init -> /bin/bash]   0x40000000 .. 0x4fffffff: stack area
[init -> /bin/bash]   0x30000 .. 0x149fff: ld.lib.so
[init -> /bin/bash]   0xae0d000 .. 0xaffffff: libc.lib.so
[init -> /bin/bash]   0xad7d000 .. 0xae0cfff: vfs.lib.so
[init -> /bin/bash]   0xad43000 .. 0xad7cfff: libm.lib.so
[init -> /bin/bash]   0xad33000 .. 0xad42fff: posix.lib.so
[init -> drivers -> nvme_drv] Error: Csts::Rdy(1) failed
[init -> drivers -> nvme_drv] Error: fatal controller status
[init -> drivers -> nvme_drv] Error: Uncaught exception of type 'Nvme::Controller::Initialization_failed'
[init -> drivers -> nvme_drv] Warning: abort called - thread: ep
[init -> drivers] child "nvme_drv" exited with exit value 1
```

As far as I can tell, the problem is occuring in the driver initialization code (nvme/main.cc). The code is waiting for the controller ready bit to get set, and times out. After that it checks the controller fatal status bit and finds it set and determines that the controller initialization has failed.

thanks,
Sachin

alex-ab, 2020-08-29:
May you please try nova or fic as kernel? The hw kernel does not support MSI

ssmore98, 2020-08-29:
Same issue running on nova:

```
Bender: Hello World.

NOVA Microhypervisor v8-de795b1 (x86_64): Aug 28 2020 21:19:53 [gcc 8.3.0] [MBI2]

[ 0] TSC:2265030 kHz BUS:1001803 kHz (measured)
[ 0] CORE:0:0:0 6:2:3:0 [1] QEMU Virtual CPU version 2.0.0

Hypervisor NOVA (API v8)
core     image  [0000000000100000,0000000003512000)
binaries region [00000000002a4000,0000000003512000) free for reuse
detected physical memory: 0x0000000000000000 - size: 0x000000000009fc00
use      physical memory: 0x0000000000000000 - size: 0x000000000009f000
detected physical memory: 0x0000000000100000 - size: 0x00000000afefc000
use      physical memory: 0x0000000000100000 - size: 0x00000000afefc000
detected physical memory: 0x0000000100000000 - size: 0x0000000150000000
use      physical memory: 0x0000000100000000 - size: 0x0000000150000000
reserved memory: 0x7cbec000 - size: 0x34132f0 type=-2
reserved memory: 0x9fc00 - size: 0x400 type=2
reserved memory: 0xf0000 - size: 0x10000 type=2
reserved memory: 0xafffc000 - size: 0x10004000 type=2
reserved memory: 0xfed1c000 - size: 0x4000 type=2
reserved memory: 0xfeffc000 - size: 0x4000 type=2
reserved memory: 0xfffc0000 - size: 0x40000 type=2
reserved memory: 0xfd000000 - size: 0x12c000 type=-5
reserved memory: 0x1000000 - size: 0x800000 type=-1
reserved memory: 0x1800000 - size: 0x5000000 type=-1
reserved memory: 0xafffe3c0 - size: 0x0 type=-3
Hypervisor reports 1x1 CPU 
Warning: CPU has no invariant TSC.
mapping: affinity space -> kernel cpu id - package:core:thread
 remap (0x0) -> 0 - 0:0:0) boot cpu
ROM modules:
 ROM: [000000007e3c6000,000000007e3c7742) LICENSE.bitstream-vera
 ROM: [000000007ffed000,000000007fffd18c) Vera.ttf
 ROM: [000000007ffa2000,000000007ffae048) VeraMono.ttf
 ROM: [000000007e30d000,000000007e334c00) acpi_drv
 ROM: [000000007f47e000,000000007fc32800) bash.tar
 ROM: [000000007e45b000,000000007e45dacb) config
 ROM: [000000000002e000,0000000000032000) core_log
 ROM: [000000007e67e000,000000007f286000) coreutils.tar
 ROM: [000000007f286000,000000007f287268) drivers.config
 ROM: [000000007ffaf000,000000007ffb17df) en_us.chargen
 ROM: [000000007e335000,000000007e335033) fb_drv.config
 ROM: [000000007fffe000,000000007fffe02d) focus
 ROM: [000000007cf4d000,000000007cf6db50) fs_rom
 ROM: [000000007e39f000,000000007e3c5d00) gui_fb
 ROM: [000000007cdc0000,000000007ce1f800) init
 ROM: [000000007e459000,000000007e45acb6) init.xsd
 ROM: [000000007e46d000,000000007e4a1070) input_filter
 ROM: [000000007f408000,000000007f408300) input_filter.config
 ROM: [000000007fc33000,000000007fd315a8) ld.lib.so
 ROM: [000000007e4a3000,000000007e67dc38) libc.lib.so
 ROM: [000000007ffb2000,000000007ffecd40) libm.lib.so
 ROM: [000000007e3c8000,000000007e3d01e0) libuuid.lib.so
 ROM: [000000007cef2000,000000007cf4c5e8) ncurses.lib.so
 ROM: [000000007ce5b000,000000007ceca9f0) nitpicker
 ROM: [000000007e3d3000,000000007e409310) nvme_drv
 ROM: [000000007e347000,000000007e39ea18) platform_drv
 ROM: [000000000002c000,000000000002d000) platform_info
 ROM: [000000007cecb000,000000007cef1c38) pointer
 ROM: [000000007e45e000,000000007e46cfe0) popt.lib.so
 ROM: [000000007e336000,000000007e346f98) posix.lib.so
 ROM: [000000007cd91000,000000007cdbf678) ps2_drv
 ROM: [000000007f3e7000,000000007f407520) report_rom
 ROM: [000000007e3d1000,000000007e3d1542) special.chargen
 ROM: [000000007fd67000,000000007ff11ab0) stdcxx.lib.so
 ROM: [000000007e40a000,000000007e447d98) terminal
[... 183 lines cut ...]
[init -> drivers -> fb_drv] Found PCI VGA at 00:01.0
[init -> drivers -> fb_drv] fb mapped to 0x5000
[init -> drivers -> platform_drv] 00:04.0 uses MSI-X vector 0x9f, address 0xfee00000
[init -> drivers -> nvme_drv] NVMe PCIe controller found (0x8086:0x5845)
[init -> drivers -> usb_drv] No <nic> config node found - not starting the USB Nic (Network) service
[init -> drivers -> usb_drv] No <raw> config node found - not starting external USB service
[init -> drivers -> usb_drv] Enabled UHCI (USB 1.0/1.1) support
[init -> drivers -> usb_drv] Enabled EHCI (USB 2.0) support
[init -> drivers -> usb_drv] Enabled XHCI (USB 3.0) support
[init -> drivers -> platform_drv] PS2 uses IRQ, vector 0x1
[init -> drivers -> platform_drv] PS2 uses IRQ, vector 0xc
[init -> drivers] child "ps2_drv" announces service "Input"
[init -> drivers] child "usb_drv" announces service "Input"
[init -> drivers] child "input_filter" announces service "Input"
[init -> drivers -> fb_drv] Found: VESA BIOS version 3.0
[init -> drivers -> fb_drv] OEM: SeaBIOS VBE(C) 2011
[init -> drivers -> fb_drv] Found: physical frame buffer at 0xfd000000 size: 16777216
[init -> drivers -> fb_drv] fb mapped to 0xb000000
[init -> drivers -> fb_drv] using VESA mode 640x480@32
[init -> drivers -> nvme_drv] Error: Csts::Rdy(1) failed
[init -> drivers -> nvme_drv] Error: fatal controller status
[init -> drivers -> nvme_drv] Error: Uncaught exception of type 'Nvme::Controller::Initialization_failed'
[init -> drivers -> nvme_drv] Warning: abort called - thread: ep
[init -> drivers] child "nvme_drv" exited with exit value 1
```

alex-ab, 2020-08-29:
Ok, you may check https://github.com/genodelabs/genode/blob/master/repos/os/run/nvme.run and the qemu args we use there. The test runs nightly and is our simplest test with the nvme driver in Qemu. However, we don't use nor test with qemu-kvm as you do.

ssmore98, 2020-08-29:
With the qemu options you suggested:

`qemu-system-x86_64 -no-kvm -display sdl -cpu core2duo -nographic -m 512 -drive id=nvme0,file=genode/build/x86_64/bin/nvme.raw,format=raw,if=none -device nvme,drive=nvme0,serial=fnord,id=nvme0n1 -serial mon:stdio -drive format=raw,file=genode/build/x86_64/var/run/tsss.img -machine q35`

 the VM does not boot:

```
SeaBIOS (version 1.11.0-2.el7)

iPXE (http://ipxe.org) 00:02.0 C980 PCI2.10 PnP PMM+1FF94800+1FED4800 C980
                                                                               

Booting from Hard Disk...
Boot failed: not a bootable disk

Booting from Floppy...
Boot failed: could not read the boot disk

Booting from ROM...
iPXE (PCI 00:02.0) starting execution...ok
iPXE initialising devices...ok

iPXE 1.0.0+ (133f4c) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP VLAN AoE ELF MBOOT PXE bzImage Menu PXEXT

net0: 52:54:00:12:34:56 using 82540em on 0000:00:02.0 (open)
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
Configuring (net0 52:54:00:12:34:56).............. ok
net0: 10.0.2.15/255.255.255.0 gw 10.0.2.2
Nothing to boot: No such file or directory (http://ipxe.org/2d03e13b)
No more network devices

```

alex-ab, 2020-08-29:
Which qemu version do you use ? Does the simple invocation of the make ... run/nvme work for, just to be sure that it could work in principle on your side.

ssmore98, 2020-08-29:
```
$ qemu-system-x86_64 -version
QEMU emulator version 2.0.0, Copyright (c) 2003-2008 Fabrice Bellard
```
I ran run/nvme in a fresh installation and it does not boot either. It will boot only if I replace machine q35 with pc.

I am compiling on `Ubuntu 20.04.1 LTS` docker container and running qemu on `CentOS Linux release 7.8.2003 (Core)`.

alex-ab, 2020-08-30:
I fear the used qemu version of you is too old. We use 3.1 or newer (current is 5.2).

ssmore98, 2020-08-30:
Thank you for your help! It may be non-trivial to upgrade the qemu on centos 7. I am going to close this issue at this time.

closed 2020-08-30 by ssmore98