genodelabs/genode #3874: libc: consistently use monitor mechanism for internal control flow

nfeske, 2020-09-03:
Up to now, the libc-internal control flow is a conglomerate of several mechanisms (kernel routine, suspend interface, monitor) and a scary complicated loop in `Kernel::run`. The monitor mechanism can in principle cover all use cases. Hence, the libc should be consolidated to the sole use of the monitor.

nfeske, 2020-09-04:
The patch https://github.com/chelmuth/genode/commit/9424d77e23b7e49b268b7e0485d73f522180b610 is not complete. The patch works fine with POSIX programs like the bash.run scenario. But when testing Sculpt with the patch, I noticed that the menu_view instances do not come up. Those are libc components, not POSIX programs.

nfeske, 2020-09-07:
> ... I noticed that the menu_view instances do not come up.

I have to correct myself. Menu view and Sculpt work fine with the patch.

skalk, 2020-09-29:
The test is broken under hw/qemu-x86_64 as well.

chelmuth, 2020-09-29:
I can't reproduce the issue on NOVA. So, we have to look deeper into base-hw.

chelmuth, 2020-10-09:
@skalk commit f09b0dc2243c24f8df9000237a72af02b76a026b synchronizes the test program execution with the HTTP server startup and fixes the failure.

nfeske, 2023-06-16:
The transition to the monitor mechanism is complete. Further libc-internal improvements deserve dedicated issues.

closed 2023-06-16 by nfeske