genodelabs/genode #3882: bash: prevent overriding the `getenv` symbol of the libc

nfeske, 2020-09-09:
@cnuke reported that the Genodians.org scenario failed when updated to the current version of Genode. The problem is a page fault in the generate step.

It turned out that the fault was triggered by tclsh (gosh) on the attempt to call getenv during its initialization. In this particular situation, tclsh was spawned by bash. During the execution of bash, the libc called `getenv`, which is normally a libc-provided function. However, as it turns out, bash also provides the same symbol, which prompts the dynamic linker to resolve calls of this function from the libc to the bash binary. After execve-ing tclsh, the bash binary is no longer present but the resolved jump slot of the libc still points to it. The next attempt to call `getenv` by the libc (tclsh calls `__xuname`, which calls `getenv`), ends up in the middle of nowhere.

Fortunately, it is possible to disable the bash-provided getenv implementation in bash's configure step, which solves this problem at hand. However, we are not safe from similar aliasing problems like this.

nfeske, 2020-09-09:
With the commit https://github.com/nfeske/genode/commit/46f16a0abbe9a85ef2fef44db4515e986dc37922, the genodians.org scenario runs again.

chelmuth, 2020-09-25:
Fix entered master branch.

closed 2020-09-25 by chelmuth