genodelabs/genode #3884: timeout: stress-test and streamline multi-threading support

m-stein, 2020-09-10:
Issues like #3881 and #3830 deal with the following problem: The Timeout framework was initially written having only single threaded contexts in mind and the SMP supports seems to stem mainly from the legacy Alarm framework on which Timeout is built. However, at least in libc/posix components the framework is meanwhile used by multiple threads even from different CPU cores, although this use case isn't tested well.

In order to resolve this discrepancy, I'd like to add several steps to the Timeout test that put a severe stress on multi-threaded usage patterns. In the context of this work, I'd also like to streamline the synchronization mechanism in the framework.

m-stein, 2020-09-14:
I just noticed that the symbols files for ld and libc were out of sync and that they were not sorted.
The corresponding fixes might already enter staging:
94cdacbac2 base/libc: sort symbol files
8705467f73 libc: sync symbols file with ld symbols file

m-stein, 2020-09-14:
@chelmuth Sorry for the confusion, the sorting wasn't necessary. My system just sorts differently with ``LC_COLLATE=C sort`` compared to a vim-internal ``:'<,'>sort`` (which yields the same order the one on staging).

m-stein, 2020-09-22:
I have a commit series that would fix this issue:
e3a613c4da timeout: rework locking scheme and clean up
52e1c442d1 timeout: remove alarm abstraction and clean up
010b59a3c5 base: get rid of alarm lib
c8b8e3e61a dde_rump: get rid of public headers
60378bfee3 dde_rump: remove unused rump_cgd
f6db7fdf26 run/rump_fat: raise ram quota
68c5a16be1 trace_logger: warn on multiple buffer wraps
64e26af31d timeout: test smp support

chelmuth, 2020-09-23:
I just merged the preparation commits until e55308b0d1 fixup "base: remove alarm library from base" (including small fixups for depot recipes). Now, I'll review the two top commits.

m-stein, 2020-09-28:
@chelmuth This should fix the issue:
23c84ea9a2 timeout: rework timeout framework
ede21cbb49 timeout: test smp support

chelmuth, 2020-09-30:
I merged both commits to staging. Thanks for tackling this! Nevertheless, would you mind to replace occurrences of `^\+[^I]+\{$` by the correct coding style?

m-stein, 2020-09-30:
@chelmuth I tried to match the pattern you gave in the affected files, but in vane. Could you please give an example what you mean?

chelmuth, 2020-10-01:
I referred to the following code places (maybe some not touched by the commit because I just grep'd).

https://github.com/genodelabs/genode/blob/staging/repos/base/src/lib/timeout/timeout.cc#L120
https://github.com/genodelabs/genode/blob/staging/repos/base/src/lib/timeout/timer_connection_time.cc#L42
https://github.com/genodelabs/genode/blob/staging/repos/base/src/lib/timeout/timer_connection_time.cc#L160

chelmuth, 2020-10-01:
I fixed issues with the symbols file in 555005d0ff7573fc3806e3efada0e0bbcd32999b. @m-stein please look into compiler errors on `make -C build/x86_64 timer/pit KERNEL=sel4`.

m-stein, 2020-10-01:
@chlmuth I've pushed fixups according to your suggestions and tested ``make timer`` with all kernels.

m-stein, 2020-10-05:
@chelmuth I've added a fixup that should prevent the problems with 32 bit platforms:

7520c3ae7b fixup "timeout: test smp support" (raise stack size)

m-stein, 2020-10-05:
@chelmuth Added the fixup you suggested. Tested it with run/wifi on x86_64/nova:
4eb985ebd7 fixup "timeout: rework timeout framework" (fix bug in application of rate limit)

Thanks for your investigation!

m-stein, 2020-10-05:
I have further fixups that should fix the timeout test on all platforms:
34c279ce51 fixup "timeout: rework timeout framework" (generalize scheduling of timeouts)
3c646d773d fixup "timeout: rework timeout framework" (fix drift in periodic timeouts)

I have another fixup that reduces the overhead of the framework by removing the strategy of scheduling 0 us for scheduler updates but I haven't tested it enough yet.

chelmuth, 2020-10-06:
Merged both commits to staging.

m-stein, 2020-10-06:
@chelmuth These fixups should finalize this issue. I've tested them with run/timeout on all kernels on x86 and arm:
4485f6fab6 fixup "timeout: rework timeout framework" (make setting the time source timeout generic)
2995199732 fixup "timeout: rework timeout framework" (do not schedule time source timeouts of 0 us)

m-stein, 2020-10-08:
I've pushed two further commits:
a01d7ae9e2 test/entrypoint: raise test duration
af4cc9b3d8 test/timer_rate: determine lowest accurate period

Regarding the timer_rate test:
It is a simple bisect approach for finding the lowest period value for Timer_session::trigger_periodic that still yields timeouts that fulfill a configured maximum error (percentage). The measurement duration resp. starting period and the termination criterion can be configured as well. I'm not sure if and how this should be integrated into CI. It is not really a test but a benchmark.

Regarding the other commit:
Although the test does what it should as well with the new duration, this seems like a workaround to me. Shouldn't we limit the timer handling rate further on pistachio in order to prevent such "DoS" ?

chelmuth, 2020-10-09:
Thanks, I'm currently investigating the capabilities of *timer_rate* but already merged both commits to staging.

closed 2020-10-09 by chelmuth

nfeske, 2021-04-01:
Commit https://github.com/genodelabs/genode/commit/46aaec735e7c30ba2e6a2edd9058e6e461c4c2df fixes a subtle performance regression I observed in Sculpt. Caused by the changed semantics of `trigger_periodic(0)`, the nitpicker instance of the leitzentrale was confronted with signals at the maximum rate after presumably disabling periodic timeouts.

The commit restores (and documents) the original semantics. @m-stein do you approve?

The effect is especially well measurable when running Sculpt on Qemu. With the commit, the boot time is reduced from 68 seconds to 60 seconds without the constant load put on the leitzentrale nitpicker instance.

m-stein, 2021-04-01:
@nfeske I must admit that I don't like this kind of special treatment of certain input values. I would rather vote for using a Periodic_timeout object and destruct it in order to discard the timeout. The silent clamping to 1 us at the other hand is merely a protection against misuse which I added as it seemed complicated to me to ensure that 0 can't be used as argument.

m-stein, 2021-04-01:
@nfeske Apart from that, it's cool that you could fix this regression, and, technically speaking I have no objections, but I would vote for fixing the user and keeping the interface simple in this case.

nfeske, 2021-04-01:
Please let me explain my chain of thoughts that led to the commit.

You are right that in most cases, the `Periodic_timeout` is the way to go. But there are nevertheless valid use cases for the direct use of the timer's RPC interface: In particular, when a precise and cheap time source is not of interest but time-triggered execution is needed. In those cases, the samping / interpolation of timestamps remains unused but it increases overhead, jitter, and complexity (periodic interaction with the timer session). Think of init and nitpicker.

Technically, passing 0 for disabling the periodic trigger is an overloading of the argument, which I generally don't favour either. I see your point when you speak of taste. But in the concrete case, there is no sensible handling of the value 0 at the RPC interface anyway. For this reason, it had a special semantics in place since the very first version of the interface. By grep'ing the source tree, one can see that nitpicker is not the only component that relied on it. At face value - disregarding taste concerns - I cannot remember a single case where an argument of 0 was passed with the expectation of an infinitely high rate.

Silently changing input values is not right - at best it masks errors and muddies the water. This is illustrated by my today's debugging session to uncover this quite severe performance regression. If I had seen a message, I would have been able to address the problem months ago. Today, after my suspicion was raised, It took me quite a while to get to the rook cause because I was not aware of the semantic change of `trigger_periodic` in the first place. (this change was not documented after all) So when skimming over my code, the call of `trigger_periodic(0)` looked obvious and innocent.

I agree that the unholy argument 0 should result in a diagnostic message when passed to a `Periodic_timeout`. Here the value really does not make sense because the right opproach of destructing a `Periodic_timeout` object is the obvious solution. However, printing a diagnisic message by the timeout framework while passing the value unfiltered from the timer client to the timer driver would open opportunities for abuse. The message would be printed by the high-priority timer driver after all. A client should never be able to deliberately spam the log in this way.

Hence, I argue that handling the value 0 at the RPC interface of the timer driver is the most sensible solution. Handling it by keeping the traditional semantics in tact does not harm. If a 0 is wrongly passed as argument to the `Periodic_timeout`, a diagnistic message is printed. But this comes from component that has the bug, not the timer driver.

m-stein, 2021-04-01:
> Silently changing input values is not right

You're right and I'd like to apologize for having changed this semantic without explicitly pointing to it and also for having unnecessarily consumed your time this way! I must admit that in the long course of this issue, my discipline in documentation couldn't catch up at some points. I'll try to keep it in mind.

For the rest, I sense that I currently don't have the time to argument in such a profound fashion about the topic. I just wanted to give an opinion as you asked but I'm not intending to block the process.

commit 0494b620 (2020-09-30): timeout: rework timeout framework

* get rid of alarm abstraction
* get rid of Timeout::Time type
* get rid of pointer arguments
* get rid of _discard_timeout indirection
* get rid of 65th bit in stored time values
* get rid of Timeout_scheduler interface
* get rid of uninitialized deadlines
* get rid of default arguments
* get rid of Timeout::_periodic
* get rid of Timeout::Raw
* use list abstraction
* only one interface for timeout handlers
* rework locking scheme to be smp safe
* move all method definitions to CC file
* name mutexes more accurate
* fix when & how to set time-source timeout
 [... cut]