genodelabs/genode #3895: vbox: MP-BIOS bug: 8254 timer not connected to IO-APIC

cnuke, 2020-09-17:
Every now and than, especially when starting multiple VMs concurrently, I notice the following message when I start my Debian 9 (4.9.0-13-amd64) VMs:

```
[…] ..MP-BIOS bug: 8254 timer not connected to IO-APIC
```
Sometimes it is merely a diagnostic message but more often it leads to a kernel panic:

```
```

If I start the 3 VMs concurrently, i.e. as part of the automated deployment on Sculpt, at least one of them if not all will abort booting with the above mentioned panic. Starting the VMs one after another manually (waiting until one shows the LUKS passphrase prompt) works fine.

While trying a different Linux distribution, in the case void-linux, which features a more recent kernel - 5.3.9 during installation and 5.8.5 afterwards I always end up with the kernel panic.

I tried disabling *IOAPIC* from `machine.vbox` but setting `<BIOS> <IOAPIC enabled="false"/> </BIOS>`.  This allows the Linux guest to boot but SMP does not work, e.g. the guest shows only 1 CPU. Specifying `noapic` in the kernel commandline allows the guest to boot but the I/O performance is severly degraded.

I guess that is related to the way we currently operate the `TSC` (see #1768), not trapping `rdtsc` but instructing VirtualBox nonetheless to use `TSC_EMULATED`, which in itself might lead to timing issues due to preemption.

That being said, @alex-ab thankfully busied himself with briefly looking into my problems and stumpled upon the kernel commandline option `no_timer_check` that instructs the kernel to omit the timer check and is apparently used when building *cloud* images [1]. This option is also set automatically by the kernel if it observes that it is running virtualized.

  [1] https://lists.fedoraproject.org/pipermail/cloud/2014-June/003975.html

Setting this option allows my VMs to boot concurrently as well as to use the more recent kernels. Besides that I cannot observe any other resulting issue with using the option.

nfeske, 2021-02-26:
For VBox6, we are leveraging the KVM clock mechanism, which will hopefully fix such issues. I doubt that we will go to the bottom of it for VBox5.

@cnuke What do you think, shall we close this issue?

cnuke, 2021-03-01:
@nfeske fine with, I'll close the issue.

closed 2021-03-01 by cnuke

giellamo, 2022-05-26:
This still happens to me since when I enabled Hyper-V

Windows 11 - VBox 6.1.34 - Host Ubuntu Focal - IO Apic enabled

chelmuth, 2022-05-27:
> This still happens to me since when I enabled Hyper-V
> 
> Windows 11 - VBox 6.1.34 - Host Ubuntu Focal - IO Apic enabled

Please tell us why this setup is related to Genode.

giellamo, 2022-05-27:
Yes, sure.

In my mind this was a virtualbox repository. Please forgive me.