genodelabs/genode #3897: fs_rom: inconsistent behavior if the file is removed

rite, 2020-09-19:
If the file has been removed, a client with attached signal handler will read an empty dataspace. However, if no signal handler is attached, a client will read the content of the file before it was deleted.

rite, 2020-09-19:
It can be reproduced using `run/bash` and f1a6583 (based on 20.05):

```
echo -n "hello" > /tmp/test.txt
rm /tmp/test.txt
```

You'll notice the difference in the logged ROM content depending on whether [line 24 in repos/os/src/test/fs_rom_delete/main.cc](https://github.com/rite/genode/commit/f1a65832bef93b7af7963b820faa1ac2ba57c853#diff-5b4eb6226f4e25274236708173048432R24) is uncommented or not.

...sorry for the crude polling-loop...

rite, 2020-09-19:
2ee98bf is a draft that makes sure that `fs_rom` handles the "lookup failed" case the same way for notifications as well as reads. This means that the client doesn't need to attach a dummy signal handler in order to read an empty ROM in case the file has been deleted.

rite, 2020-09-19:
> It can be reproduced using `run/bash` and [f1a6583](https://github.com/genodelabs/genode/commit/f1a65832bef93b7af7963b820faa1ac2ba57c853) (based on 20.05):
> 
> ```
> echo -n "hello" > /tmp/test.txt
> rm /tmp/test.txt
> ```

Unfortunately, the current genodelabs/master (f6337a6) can't be used to show the issue because the `rm`-command leads to:

```
[init -> /bin/bash] Error: deadlock ahead, mutex=0x10ff99f0, return ip=0x500540f6
```
I suppose it's worth a dedicated issue but I'm unsure if it is already covered in one of the libc issues, for example. As always, feel free to proceed at your discretion ;)

chelmuth, 2020-09-21:
@rite I tested *rm* in *bash* with the following steps on master  but could not reproduce the deadlock. Is there anything I'm missing?

```
make -C build/x86_64 run/bash KERNEL=linux BOARD=linux

bash> for i in $(seq 100); do echo -n "hello" > /tmp/test.txt; rm /tmp/test.txt; done
```
```

```

rite, 2020-09-21:
> @rite I tested _rm_ in _bash_ with the following steps on master but could not reproduce the deadlock. Is there anything I'm missing?

Sorry, I mixed things up during the aggregation of this issue... The correct test application should look like the one in 
797cb6f.

`rm` may lead to `[init -> vfs] Error: deadlock ahead, mutex=0x7ffff5fff4f8, return ip=0xafa7664` or `[init -> vfs_rom] Error: bad packet seek position`. `rm` might fail on both 20.05 and current master yet my impression is that on current master it occurs more often.

nfeske, 2020-09-28:
@rite thank you for the detailed report. Your interim fix looks fine. After reviewing the fs_rom implementation, however, I came to the conclusion that the code is currently more complicated than it should be. E.g., the method called `_notify_client_about_new_version` does not merely do what it says but it also changes the payload. This hints at an underlying design issue. I guess it crept in while fixing symptoms of the prior vfs server implementation.

I'll take from this issue and from your recent mailing-list posting (about renamed files) that fs_rom will ultimately need to undergo a design review as a whole. Until we are able to get to it, however, I hope that you can live with your fix?

rite, 2020-09-28:
> Until we are able to get to it, however, I hope that you can live with your fix?

@nfeske Sure - for me there's no need to tackle it right now.

nfeske, 2024-11-18:
@rite I've merged the slightly adjusted https://github.com/genodelabs/genode/commit/808759501c2bbebd591c545a20f2dd52cecc462b to staging today.