genodelabs/genode #3900: i.MX8 support for NXP MX8_DSI_OLED1 display

ssumpf, 2020-09-23:
Add MIPI DSI support to our existing i.MX8M Quad HDMI framebuffer driver.

nfeske, 2020-09-24:
Is this commit ready for review/merging? I'm asking because the issue is not marked as fixed yet.

ssumpf, 2020-09-24:
@nfeske: Not yet, I am adding touch support at the moment and wanted to use this issue as well. Or is it better to open another issue for touch?

nfeske, 2020-09-24:
> Or is it better to open another issue for touch?

I'm fine with either way. ;-)

ssumpf, 2020-09-24:
@nfeske: The commits above add touch support for the display, change the gpio driver to take advantage of the new platform driver, and clean up the run scripts that used gpio. Next I gonna check for gpio usage within world.

ssumpf, 2020-09-30:
@nfeske: 945c72a incorporates your suggestions regarding `throw` and `unsigned`.

ssumpf, 2020-10-05:
@nfeske: I have tried to incorporate your suggestions into 27bc58d regarding GPIO. So the relevant commits are:   
76bf71a 
77b877d 
933dca7
3d43147 
e18b885 
e4a29e5
85639da 
27bc58d
from my dde_linux_imx8_issue3900_1 branch:
https://github.com/ssumpf/genode/tree/dde_linux_imx8_issue3900_1

Please consider for staging.

nfeske, 2020-10-05:
Thank you @ssumpf. I merged all commits except for the quota adjustment in the demo.run script (commit dd0450489b4ee25295554035b448179c52a01a8e) because this would introduce an inconsistency with all the other run scripts that are based on the drivers_interactive package. I wonder, may it be possible to shave off the RAM quotas of the components within the drivers subsystem so that we can retain the current 64MiB limit?

ssumpf, 2020-10-05:
@nfeske: Great. The RAM quota commit was not meant for staging anyway and does not appear in the list above. As said, I have checked the memory consumption of the driver regarding HDMI, which was about 32 MB, 2 * 8  MB for the framebuffer, 8 MB for the slab allocators - which should be changed anyway and is a relict from OMAP4's "90% percent" performance goal that turned out to be caused by the CPU frequency. This and the other 8 MB I will have a look into again because the 64MB limit should suffice.

nfeske, 2020-10-05:
Cool, that's all the better. Thanks for the feedback!

chelmuth, 2020-10-06:
The commit series seems to break `./tool/depot/create UPDATE_VERSIONS=1 FORCE=1 REBUILD= -j4 genodelabs/src/imx53_qsb_drivers`. Please have a look.

ssumpf, 2020-10-06:
@chelmuth: Ok.

ssumpf, 2020-10-06:
@chelmuth: a84bf44 fixes the package problem for imx53

chelmuth, 2020-10-06:
Merci, works like a charm now.

chelmuth, 2020-10-09:
All commits in this issue entered master today. Can we close the issue?

closed 2020-10-09 by ssumpf