genodelabs/genode #3914: nova: support avx

alex-ab, 2020-10-13:


nfeske, 2021-02-26:
What is the story behind this issue?

alex-ab, 2021-02-26:
The idea is to enable [AVX](https://en.wikipedia.org/wiki/Advanced_Vector_Extensions) so that the load in guests can be reduced, e.g. for crypto, like AES, during video streaming of secured connections. Enabling this feature in the kernel is the one part, additional effort in our vm interface is required since other instructions must be used and the size of the FPU state increases.

nfeske, 2021-02-26:
Thank you for the explanation. I'm admittedly not quite sure about the prioritization of this idea related to the many other topics at hand. Intuitively it feels a bit intangible to me, both regarding the effort needed and regarding the benefit.

Since you have opened this issue some months ago, how do you expect this issue to evolve?

alex-ab, 2021-02-26:
> Since you have opened this issue some months ago, how do you expect this issue to evolve?

It is a side track which may or may not evolve, as time permit. We can close it for now.

commit f5a9d5e6 (2024-08-06): nova: enable FPU AVX support

Add extended FPU state detection and handling (via xsave and friends) to the
kernel, which has to store/load more FPU state (~512 -> 2k++) during context
switching of threads. Additional the referenced nova branch contains various
optimization during VM destruction and cross core IPC resource caching.

The FPU work is based upon upstream NOVA kernel commits and Hedron sources.

Issue #5314
Fixes #3914
commit 6aeb5c03 (2024-08-12): nova: enable FPU AVX support

Add extended FPU state detection and handling (via xsave and friends) to the
kernel, which has to store/load more FPU state (~512 -> 2k++) during context
switching of threads. Additional the referenced nova branch contains various
optimization during VM destruction and cross core IPC resource caching.

This FPU work is based upon upstream NOVA kernel and Hedron commits.

Issue #5314
Fixes #3914
closed 2021-02-26 by alex-ab