genodelabs/genode #3918: sculpt-20-08 hangs on Dell OptiPlex 3050

dckc, 2020-10-17:
It boots as far as [figure 1][f1] but then it's unresponsive to mouse / keyboard input.

I'm struggling to figure out which of the [hardware requirements][hw] I am running into. It's strange because I'm pretty sure I managed to get an earlier version of sculpt woking fine on this machine.

For reference: [Probe #1e62d6029b of Dell 08NPPY A00 desktop PC (OptiPlex 3050)](https://linux-hardware.org/?probe=1e62d6029b)

`sculpt-20-08-b.img` hash
SHA256 d8d92e701e454766364b5e3233b75a5edabf48d761301b3b9912dbb69af85899

[hw]: https://genode.org/documentation/articles/sculpt-20-08#Hardware_requirements_and_preparations
[f1]: https://genode.org/documentation/articles/sculpt-20-08#sculpt_20_08_panel

alex-ab, 2020-10-26:
You may try to use a PS/2 mouse/keyboard, just in case the usb driver has an issue. According to https://i.dell.com/sites/csdocuments/Shared-Content_data-Sheets_Documents/en/OptiPlex-3050-Towers-Technical-Specifications.pdf, there could be such an connector. Also, according to the spec, there is a chance that you can get serial output which would help to isolate the issue.

alex-ab, 2020-11-17:
You may try following [image](https://depot.genode.org/alex-ab/images/sculpt-20-08-migrate-amd.img) which contains an updated USB driver.

dckc, 2020-11-19:
Thanks; that image works nicely on my HP Envy notebook.

I haven't found time to try it on the Dell OptiPlex yet - rebooting my main workstation involves a lot of context switching.

alex-ab, 2021-03-25:
@dckc: I'm closing the issue, since the new Sculpt 21.03 image contains all adjustments of the test image which worked for your HP Envy.

closed 2021-03-25 by alex-ab