genodelabs/genode #3923: base: support process-local signal submission

cproc, 2020-10-26:
If a signal receiver is known to reside in the same process as the signal submitter, for example when using libc monitors, process-local signal submission can noticeably reduce overhead time. The `Signal_receiver` class already supports this feature and this issue is about adding a `Signal_context::local_submit()` function, which can be called instead of using a `Signal_transmitter`.

cproc, 2020-10-26:
@chelmuth: temporary test commit for staging

nfeske, 2020-11-05:
The `Signal_context::local_submit` method should not have the `cnt` argument. The count is a legacy that is not (and should not be) used anywhere. In the implementation, when creating the `Data` object, just pass 1 as count value.

nfeske, 2020-11-05:
Thanks @cproc for implementing this optimization idea so swiftly and cleanly. With the minor change as requested above, I'm very happy with the commit.

nfeske, 2020-11-27:
Fixed in master.

closed 2020-11-27 by nfeske

chelmuth, 2022-12-05:
@cproc do you approve my analysis and patch above?