genodelabs/genode #3927: platform_drv(x86): ipxe_nic_drv not working with MSI-X for RTL8111/8168/8411 device

alex-ab, 2020-10-28:


chelmuth, 2020-10-29:
Do you have any notes (a reference in the spec maybe) you may add to the commit message why this is required?

alex-ab, 2020-10-30:
I looked through the spec and couldn't find a hint. In this case it is MMIO provided by a PCI BAR of the device. AFAIK the spec of the device is not public in order to check.

alex-ab, 2020-10-30:
@cnuke acutally found the datasheet for the realtek device, and there you see that you have to use 2x32bit accesses https://usermanual.wiki/Document/rtl8111cgrrealtekr11.25570219/view.

alex-ab, 2020-10-30:
Some more references by @cnuke: https://www.illumos.org/issues/4217, http://openbsd-archive.7691.n7.nabble.com/MSI-X-fix-td367604.html

commit 9d2403ea (2020-10-30): platform_drv(x86): use 32bit writes for MSI-X

Fixes #3927
commit 11e261ad (2020-11-23): platform_drv(x86): use 32bit writes for MSI-X

Fixes #3927
closed 2020-11-27 by chelmuth