genodelabs/genode #3931: nic_router: make propagation of link states at interfaces usable for clients

m-stein, 2020-10-30:
Currently, there exist at least three issues with the propagation of link states in the NIC router:

1. The logic of when to change the link state is at least misleading. For instance, re-configuring the "DNS server" attribute of a DHCP server should cause a link down/up sequence but the code doesn't reflect this causality (although the link down/up is done as a side-effect of other re-configuration code.

2. The link state returned by Nic_session::link_state() doesn't really change on artificial link down/up sequences used to inform  clients about changed DHCP information provided through a DHCP server of the NIC router. The NIC router simply optimizes away this state assuming that a client considers a link state to be changed when receiving the link state signal. This assumption doesn't hold.

3. When solving issue 2, the following question arises: How to ensure that clients are able to recognize each link state change? Or in other words: How to deal with the situation where a link state changes faster than clients can recognize it?

m-stein, 2020-10-30:
Commit 009f7b6508 is a prototype for a configurable minimum dwell time for link states and a real state change on link down/up sequences. The feature is showcased by the adapted run/ping_nic_router test. Please consider this work in progress!

m-stein, 2020-10-30:
@ssumpf Please have a try on the above mentioned commit ;)

ssumpf, 2020-10-30:
@m-stein: Ok, thanks.

m-stein, 2020-11-17:
This is a clean approach to the above mentioned problems 2 and 3:
4788079c87 nic_router: configurable min link-state dwell-time
8f78cc8198 nic_router: allow for config values of 0 seconds
aba750ce19 nic_router: clean up init of configuration members
ebd1919bda timeout_types.xsd: microseconds type

m-stein, 2020-11-17:
Fixed:
4dc1274461 nic_router: fix link state on dhcp server reconfig
f3158dd419 nic_router: configurable min link-state dwell-time
b5c4ac4c65 nic_router: allow for config values of 0 seconds
3b0b86d7bc nic_router: clean up init of configuration members
8b9f11bd0d timeout_types.xsd: microseconds type

m-stein, 2020-11-30:
@chelmuth @nfeske I just rebased and added a fixup that changes the semantic in the way that acknowledging a preserved link state isn't triggered through a timeout anymore but through the client reading the preserved link state (renames commit also):

589fc01c9b nic_router: fix link state on dhcp server reconfig
cada31b001 Fixup "nic_router: configurable min link-state dwell-time" (preserve link state until read)
de9131ace6 nic_router: configurable min link-state dwell-time
d7df9c99a6 nic_router: allow for config values of 0 seconds
90fd10b2ba nic_router: clean up init of configuration members
f7de38766d timeout_types.xsd: microseconds typ

chelmuth, 2020-12-01:
Merged the series except f7de387 timeout_types.xsd: microseconds type

m-stein, 2020-12-01:
@chelmuth Added a Squash commit:
74e1603fb4 Squash "nic_router: preserve session link state until read"

commit f8a231ca (2020-12-01): nic_router: fix link state on dhcp server reconfig

When reconfiguring the NIC router modifies the information that DHCP clients
received through a DHCP server of the NIC router, the link state of the
correspodning interface has to do a "down-up" sequence. This provides a signal
to the DHCP clients to redo DHCP, receive the new information and update
accordingly.

Fixes #3931
commit 7add4bfe (2020-12-01): nic_router: fix link state on dhcp server reconfig

When reconfiguring the NIC router modifies the information that DHCP clients
received through a DHCP server of the NIC router, the link state of the
correspodning interface has to do a "down-up" sequence. This provides a signal
to the DHCP clients to redo DHCP, receive the new information and update
accordingly.

Fixes #3931
commit 8d8edaea (2020-12-09): nic_router: fix link state on dhcp server reconfig

When reconfiguring the NIC router modifies the information that DHCP clients
received through a DHCP server of the NIC router, the link state of the
correspodning interface has to do a "down-up" sequence. This provides a signal
to the DHCP clients to redo DHCP, receive the new information and update
accordingly.

Fixes #3931
closed 2020-12-09 by nfeske