genodelabs/genode #3935: nic_router: support disabling ARP for a domain

m-stein, 2020-11-04:
The combination of LTE modems and the NIC router brings up a use case that isn't supported so far by the latter. A modem might not expect it's NIC session client using ARP for requesting MAC addresses. For instance, with a NIC router uplink in domain "uplink" connected directly to the modem driver and a ping client behind a downlink interface in the domain "default" (icmp routed to "uplink"), we try to ping IP 8.8.8.8 through the NIC router and the modem. We told domain "uplink" its ip (10.124.194.249) and gateway (10.124.194.250) that were determined by the modem. Domain "uplink" now tries to request the MAC for IP 10.124.194.250 via ARP. This gets ignored by the modem and the ping never gets through.

In Linux such interfaces are tagged as ``NOARP`` instead. Therefore, the above mentioned ARP request is spared out and the target IP 8.8.8.8 is directly addressed instead:

ICMP request from source:
```
Frame 28: 98 bytes on wire (784 bits), 98 bytes captured (784 bits) on interface wwan0, id 0
Ethernet II, Src: 52:90:80:c7:47:06 (52:90:80:c7:47:06), Dst: 52:90:80:c7:47:06 (52:90:80:c7:47:06)
    Destination: 52:90:80:c7:47:06 (52:90:80:c7:47:06)
    Source: 52:90:80:c7:47:06 (52:90:80:c7:47:06)
    Type: IPv4 (0x0800)
Internet Protocol Version 4, Src: 10.124.194.249, Dst: 8.8.8.8
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
    Total Length: 84
    Identification: 0xd7f4 (55284)
    Flags: 0x4000, Don't fragment
    ...0 0000 0000 0000 = Fragment offset: 0
    Time to live: 64
    Protocol: ICMP (1)
    Header checksum: 0x852f [validation disabled]
    [Header checksum status: Unverified]
    Source: 10.124.194.249
    Destination: 8.8.8.8
Internet Control Message Protocol
```
ICMP response from target:
```
Frame 29: 98 bytes on wire (784 bits), 98 bytes captured (784 bits) on interface wwan0, id 0
Ethernet II, Src: 00:00:00_00:00:00 (00:00:00:00:00:00), Dst: 52:90:80:c7:47:06 (52:90:80:c7:47:06)
    Destination: 52:90:80:c7:47:06 (52:90:80:c7:47:06)
    Source: 00:00:00_00:00:00 (00:00:00:00:00:00)
    Type: IPv4 (0x0800)
Internet Protocol Version 4, Src: 8.8.8.8, Dst: 10.124.194.249
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
    Total Length: 84
    Identification: 0x0000 (0)
    Flags: 0x0000
    ...0 0000 0000 0000 = Fragment offset: 0
    Time to live: 112
    Protocol: ICMP (1)
    Header checksum: 0x6d24 [validation disabled]
    [Header checksum status: Unverified]
    Source: 8.8.8.8
    Destination: 10.124.194.249
Internet Control Message Protocol
```

One solution to this issue would be to let the user mark NIC router domains as ``NOARP`` too (``<domain use_arp="no" ...>``). Of course, another option would be to have a proxy component between NIC router and modem but I think it's sensible to do it inside the NIC router.

m-stein, 2020-11-04:
@ssumpf I created a prototype that you should try out and merged it with an adapted version of the minimum dwell for link states: 
 2bbca5d648 WIP "nic_router: support and show-case disabling arp at a domain"
 bc24274866 WIP "nic_router: configurable minimum dwell time for link states"

ssumpf, 2020-11-05:
@m-stein: Cool, thanks :+1:

ssumpf, 2020-11-05:
@m-stein: 2bbca5d works like a charm :100:

chelmuth, 2020-12-01:
I'd like to merge this work. @m-stein could you please refer me to clean commits.

m-stein, 2020-12-02:
@chelmuth This is a working, rebased version of the WIP commit:
762b10766f WIP "nic_router: support and show-case disabling arp at a domain"
I'm currently working at the fix-ups.

m-stein, 2020-12-02:
@chelmuth I reased and added a clean-up squash commit:
5351572852 Squash "nic_router: support and show-case disabling arp at a domain"
53444d7fa1 WIP "nic_router: support and show-case disabling arp at a domain"

chelmuth, 2020-12-03:
Thanks. Squashed and merged both commits with no objections.

commit 9e1a14b8 (2020-12-03): nic_router: support and show-case disabling arp at a domain

* Introduce 'use_arp' configuration flag at the NIC router in order to disable
  requesting IP-to-MAC address resolutions via ARP for domains.
* Add automated run/nic_router_disable_arp test to the autopilot that tests
  the effect of the new flag
* Document the new flag and the new test in the NIC router README

Fixes #3935
commit a124f5b8 (2020-12-09): nic_router: support and show-case disabling arp at a domain

* Introduce 'use_arp' configuration flag at the NIC router in order to disable
  requesting IP-to-MAC address resolutions via ARP for domains.
* Add automated run/nic_router_disable_arp test to the autopilot that tests
  the effect of the new flag
* Document the new flag and the new test in the NIC router README

Fixes #3935
closed 2020-12-09 by nfeske