genodelabs/genode #3937: gems: import CBE Genode code

m-stein, 2020-11-05:
As described in https://github.com/cbe-project/cbe/issues/70, the CBE repository is stripped down to the generic core logic and all Genode-specific CBE code moves to the "gems" repository. Furthermore, the remaining CBE repo will become a Genode port.

m-stein, 2020-11-11:
Fixed:
ce2dd0a981 gems: import Genode-specific code of the CBE

nfeske, 2020-11-11:
The file `io_job.h` appears twice.

Could you re-check the include guards? I just has a glimpse at io_job.h that uses an unconventional camel-case form instead of uppercase letters.

The directory gems/include/util does not exist yet. Would you mind placing the corresponding files at gems/include/gems instead? The include/util location as present in base and os is meant to contain only fairly freestanding utilities, not classes that depend of higher-level framework infrastructure.

nfeske, 2020-11-11:
I wonder how the `download_coreplus` pkg relates to the CBE?

nfeske, 2020-11-11:
Just browsing through the commit as a whole, it entails an immensely impressive body of work. In particular the amount and care of the validation/testing-related code speaks volumes.

m-stein, 2020-11-11:
@nfeske Thank you for your kind words and your fast but thorough review!

Regarding the redundant ``io_job.h```headers: Unfortunately, the classes in these two headers are not interchangeable without further work. I'd love to get rid of one right away but I had to limit clean-up overhead to a reasonable amount.

@cnuke Can you as author of these headers give some insight on why there are two ``Io_job`` classes?

Regarding ``gems/include/utils``: I'd go further and move the contained headers to ``gems/include/cbe`` as they aren't related to any other code than the CBE. I also felt the desire to do the same for the headers in ``gems/include/cbe_check``, ``gems/include/cbe_init``, ``gems/include/cbe_dump``, and ``gems/include/cbe_crypto`` as they only bloat the include dir while being all specific to the CBE ecosystem.

Regarding the ``download_coreplus`` package: It's part of the CBE demo described in the corresponding [Genodians article](https://genodians.org/m-stein/2020-06-12-cbe-linux-vm). Therefore it's referenced by the abstract package ``cbe_demo`` that is meant to comprise all prerequisites for the demo. Of course, it would be nice to use a more generic approach for downloading the image. Can you give an advice on this?

m-stein, 2020-11-11:
@nfeske Besides, wouldn't ``libports`` be a better destination for the CBE code?

m-stein, 2020-11-11:
> To avoid spilling too many subdirectories at gems/include/, would it be an option to use the structure include/cbe/dump/, include/cbe/crypto and so forth? That is, instead of having a flat number of cbe_-prefixed directories, introducing a cbe/ hierarchy level.

That's exactly what I was heading for :) Nice that our ideas align!

> I think gems is the right place because all the glue (components, vfs plugins, test infrastructure) around the CBE core library is fairly substantial and goes far beyond a porting effort - we are speaking quite literally of "gems" here. ;-)

Ok, that makes sense. I just felt a bit strange as it's the first and only port in this repo. Thanks for the clarification!

m-stein, 2020-11-11:
I've updated the commit according to the suggestions of @nfeske:
fb5fadd8d0 gems: import Genode-specific code of the CBE

nfeske, 2020-11-12:
Wow, that was quick! Thank you. I just merged the new version to staging.

cnuke, 2020-11-12:
> @cnuke Can you as author of these headers give some insight on why there are two `Io_job` classes?

IIRC it started with the private `io_job.h` in the VFS-CBE plugin which I then re-used for the other plugins. While doing that its implementation changed slightly and I put it into a _global_ include directory. I planned to use it in the CBE plugin as well eventually but in the end lacked the time to do so.

chelmuth, 2020-11-13:
We had a small amount of hickups with the current version in the CI.
```
vfs_cbe_crypto_memcopy/2020-11-12/src/lib/vfs/cbe_crypto/vfs.cc:536:38: error: narrowing conversion of ‘(((Vfs::file_size)((Genode::addr_t)((Vfs_cbe_crypto::Keys_f
       .fileno = (Genode::addr_t)this | index,
                 ~~~~~~~~~~~~~~~~~~~~~^~~~~~~

Error: missing README in package recipe, expected:
       .../repos/gems/recipes/pkg/cbe_init_trust_anchor/README

/data/genode/contrib/cbe-7bef1d81eef5ca1e8b99305c7b32fe8302d4eead/src/lib/cbe_common/cbe-primitive.adb:188:23: run-time library configuration error
.../genode/contrib/cbe-7bef1d81eef5ca1e8b99305c7b32fe8302d4eead/src/lib/cbe_common/cbe-primitive.adb:188:23: file s-imenne.ads not found
.../genode/contrib/cbe-7bef1d81eef5ca1e8b99305c7b32fe8302d4eead/src/lib/cbe_common/cbe-primitive.adb:188:23: entity "System.Img_Enum_New.Image_Enumeration_8" not avai
compilation abandoned
make[3]: *** [.../genode/repos/base/mk/generic.mk:96: cbe-primitive.o] Error 1
.../genode/contrib/cbe-7bef1d81eef5ca1e8b99305c7b32fe8302d4eead/src/lib/cbe_common/cbe-debug.adb:265:18: run-time library configuration error
.../genode/contrib/cbe-7bef1d81eef5ca1e8b99305c7b32fe8302d4eead/src/lib/cbe_common/cbe-debug.adb:265:18: file s-imenne.ads not found
.../genode/contrib/cbe-7bef1d81eef5ca1e8b99305c7b32fe8302d4eead/src/lib/cbe_common/cbe-debug.adb:265:18: entity "System.Img_Enum_New.Image_Enumeration_8" not availabl
compilation abandoned
make[3]: *** [.../genode/repos/base/mk/generic.mk:96: cbe-debug.o] Error 1
    COMPILE  cbe.o
  Program app/log_core/log_core
.../genode/contrib/cbe-7bef1d81eef5ca1e8b99305c7b32fe8302d4eead/src/lib/cbe_common/cbe-request.adb:127:23: run-time library configuration error
.../genode/contrib/cbe-7bef1d81eef5ca1e8b99305c7b32fe8302d4eead/src/lib/cbe_common/cbe-request.adb:127:23: file s-imenne.ads not found
.../genode/contrib/cbe-7bef1d81eef5ca1e8b99305c7b32fe8302d4eead/src/lib/cbe_common/cbe-request.adb:127:23: entity "System.Img_Enum_New.Image_Enumeration_8" not availa
compilation abandoned
make[3]: *** [.../genode/repos/base/mk/generic.mk:97: cbe-request.o] Error 1
.../genode/contrib/cbe-7bef1d81eef5ca1e8b99305c7b32fe8302d4eead/src/lib/cbe_common/cbe.adb:122:15: run-time library configuration error
.../genode/contrib/cbe-7bef1d81eef5ca1e8b99305c7b32fe8302d4eead/src/lib/cbe_common/cbe.adb:122:15: file s-imenne.ads not found
.../genode/contrib/cbe-7bef1d81eef5ca1e8b99305c7b32fe8302d4eead/src/lib/cbe_common/cbe.adb:122:15: entity "System.Img_Enum_New.Image_Enumeration_8" not available
compilation abandoned
make[3]: *** [.../genode/repos/base/mk/generic.mk:97: cbe.o] Error 1
make[3]: Target 'all' not remade because of errors.
make[2]: *** [var/libdeps:659: cbe_common.lib] Error 2
```

m-stein, 2020-11-16:
@chelmuth Thank you for this helpful summary :)

Regarding the narrowing conversion: It seems this only happens on 32 bit platforms and the whole CBE was, until now, only meant for x86_64. That said, I'm uncertain how to react to this. Should we debug building the archives with 32 bit now, and if not, how do I prevent that archives are being built for architectures other than x86_64?

Regarding the missing README: This was an abandoned package and I removed it:
1041eef952 Fixup "gems: import Genode-specific code of the CBE" (remove unused pkg/cbe_init_trust_anchor)

Regarding the "entity ... not available" errors: I couldn't reproduce this with any architecture so I'd say that I have to know the compilation context - I'll look at it again as soon as I'm in the office today.

m-stein, 2020-11-16:
@cnuke @nfeske Regarding ``io_job.h``: When I'm done with more important things, I'd like to clean up the code in a way that only one implementation of the vfs-crypto and vfs-trust-anchor back-end modules remains that is used by CBE tester, CBE VFS, CBE init etc. . This will include getting rid of the redundant ``io_job.h`` .

cnuke, 2020-11-16:
@m-stein thanks, much appreciated.

nfeske, 2020-11-17:
@m-stein do you keep the build issues on non-x86_64 in mind?

nfeske, 2020-11-17:
BTW, there is also a built issue on x86_64:

```
cbe-primitive.adb:188:23: file s-imenne.ads not found
[2020-11-17 15:36:57] /data/genode/contrib/cbe-7bef1d81eef5ca1e8b99305c7b32fe8302d4eead/src/lib/cbe_common/cbe-primitive.adb:188:23: entity "System.Img_Enum_New.Image_Enumeration_8" not available
```

m-stein, 2020-11-18:
@nfeske Yes, I'm at it.

nfeske, 2020-11-20:
The REQUIRES mechanism is only meant for target.mk files, not for libraries. For the latter, the correct way to handle this situation is moving the library-description files to a spec/x86_64 directory. I added the fixup https://github.com/genodelabs/genode/commit/4cbb3b0aeab0565ff7272c01843628663093234f

m-stein, 2020-11-23:
@nfeske I'm sorry for that. Somehow, I messed up the idea with the libraries and even failed to commit my changes to the target.mk's O.o

m-stein, 2020-11-23:
Thanks for the fixups!

m-stein, 2020-11-23:
I have some more fixups that also implement the suggestions by @nfeske:
0cffca04bb fixup "gems: import Genode-specific code of the CBE" (get rid of cbe lib target.mk files in gems)
2923ee3f15 fixup "gems: import Genode-specific code of the CBE" (remove cbe packages that are not part of the demo)
bc7da27bd7 fixup "gems: import Genode-specific code of the CBE" (get rid of cbe raw archives)
8328a7f3b6 fixup "gems: import Genode-specific code of the CBE" (let recipes/src/cbe replace other cbe src/api archives)
3cf0feca7c fixup "gems: import Genode-specific code of the CBE" (adapt recipes to new spec subfolders)

nfeske, 2020-11-23:
Thanks @m-stein, this is much more comprehensive now.

m-stein, 2020-11-23:
@nfeske This commit breaks the cbe tests:
0cffca04bb fixup "gems: import Genode-specific code of the CBE" (get rid of cbe lib target.mk files in gems)

Removing the commit from staging again should be fine. I removed the target files as, normally, we do not have such files for libraries and, as they do not set the TARGET variable the following warnings occur in the src/cbe recipe:
```
overriding recipe for target '.prg'
ignoring old recipe for target '.prg'
```
However, I now suppose that vfs plugins are a special case, as, sometimes, they are interchangeable with the same application (e.g. cbe_crypto_memcopy and cbe_crypto_aes_cbc ) and therefore have to be selected at integration time and not within the target.

That said, I would suggest to set the TARGET variable in such targets:
64d90b0cdc fixup "gems: import Genode-specific code of the CBE" (set TARGET in cbe lib target files)

m-stein, 2020-11-23:
This should fix the missing RT support that our nightly tests complain about:
f9c90ce2e5 ada runtime: support image of enums

nfeske, 2020-11-23:
Thank you for the update. I merged both commits to staging.

> That said, I would suggest to set the TARGET variable in such targets:

Normally, the build system builds a library only as a side affect of building a target that depends on the library. VFS plugins are special because they are not linked to targets directly but loaded according to a runtime policy. The target.mk files are merely used to trick the build system to create the libraries. The name of the target does not matter because no target is produced. But specifying a name cannot hurt. When looking at the existing VFS plugins, this is not perfectly consistent as of now.

commit 2b5c5ba3 (2020-11-12): gems: import Genode-specific code of the CBE

The CBE repository contained a lot of Genode-specific code despite the fact
that the CBE core logic is not bound to Genode in any way. Therefore the
Genode-specific CBE code is moved to the 'gems' repository to form part of
Genode mainline. The remaining CBE code becomes a port in Genode instead of
being invoked as sub-repository.

The commit combines the following work steps:

* add all files removed from CBE repository
* add CBE port files
* make all CBE libs and targets build again
* make all CBE run scripts succeed again
* make all CBE recipes  [... cut]
commit 30b8f4ef (2020-11-27): gems: import Genode-specific code of the CBE

The CBE repository contained a lot of Genode-specific code despite the fact
that the CBE core logic is not bound to Genode in any way. Therefore the
Genode-specific CBE code is moved to the 'gems' repository to form part of
Genode mainline. The remaining CBE code becomes a port in Genode instead of
being invoked as sub-repository.

The commit combines the following work steps:

* add all files removed from CBE repository
* add CBE port files
* make all CBE libs and targets build again
* make all CBE run scripts succeed again
* make all CBE recipes  [... cut]
closed 2020-11-27 by chelmuth