genodelabs/genode #3939: dde_ipxe: realtek driver stops processing packets on larger downloads

alex-ab, 2020-11-05:


alex-ab, 2020-11-05:
With commit a99263f1178a61ec8aad754eab1596061feeb648 I could download the file successful on various runs with fetchurl_lwip. Sometimes, the isr (interrupt status register) is 0 but actually data to be processed is available in the tx and rx rings.

alex-ab, 2021-02-05:
With debug commit e2b2c2da7ce12fda7ec90d9c0ad653100c5d0839 the attached trace can be observed. The patch shows the interrupt status register (isr) values of the realtek device before we call the netdev_poll and the status after we have finished the processing in case that actually no rx data is available. Additionally, periodically a artificial timer checks the isr and show the value, in case that there is no progress anymore.

What one can see from the output is that sometimes on a interrupt, there is actual no rx data available. The isr = 0x4 indeed tells just that. In the case that the device/driver don't make progress anymore, isr becomes 0x1 which means that rx is done. It seems, since this isr value stays and is not cleared, no further interrupts are generated (line isr before=84 now=1).

I spend already some time to understand/debug the actual reason, but I can't pinpoint it by now. With commit 079186f5e4c4a0b1d69792526cd399dde17e2f09 the driver code is requested one time again if no rx data was processed on an interrupt. The effect is that rings are processed a second time and the isr = 0x1 is cleared. With this quirk the download works reliable on two different machines.

In contrast to the original quirk e389db56330cd3be8d4c81380546dec345570a35, 079186f5e4c4a0b1d69792526cd399dde17e2f09 does not require a periodic timer to let the driver work with realtek devices. (It is ugly nevertheless ...)

With 8205c4096ff7e3c4edf3cd895ea5b38950ad853f the throughput of the realtek device also becomes usable, e.g. increasing from few kB/s to several MB/s (saturating my DSL connection).

```
[init -> report_rom]   <progress>
[init -> report_rom]   	<fetch url="https://depot.genode.org/alex-ab/raw/vm-tc-firefox/2018-06-27.tar.xz" total="131577252.0" now="1048576.0"/>
[init -> report_rom]   </progress>
[init -> drivers -> nic_drv] no data processed on irq, isr before=84 now=0
[init -> drivers -> nic_drv] 
[init -> drivers -> nic_drv] no data processed on irq, isr before=0 now=0
[init -> drivers -> nic_drv] 
[init -> drivers -> nic_drv] no data processed on irq, isr before=84 now=0
[init -> drivers -> nic_drv] 
[init -> drivers -> nic_drv] no data processed on irq, isr before=0 now=0
[init -> drivers -> nic_drv] 
[init -> drivers -> nic_drv] no data processed on irq, isr before=84 now=0
[init -> drivers -> nic_drv] 
[init -> drivers -> nic_drv] no data processed on irq, isr before=84 now=0
[init -> report_rom] report 'fetchurl -> progress'
[init -> drivers -> nic_drv] 
[init -> report_rom]   <progress>
[init -> report_rom]   	<fetch url="https://depot.genode.org/alex-ab/raw/vm-tc-firefox/2018-06-27.tar.xz" total="131577252.0" now="1130496.0"/>
[init -> report_rom]   </progress>
[init -> drivers -> nic_drv] no data processed on irq, isr before=0 now=0
[init -> drivers -> nic_drv] 
[init -> drivers -> nic_drv] no data processed on irq, isr before=84 now=0
[init -> drivers -> nic_drv] 
[init -> drivers -> nic_drv] no data processed on irq, isr before=0 now=0
[init -> drivers -> nic_drv] 
[init -> drivers -> nic_drv] no data processed on irq, isr before=84 now=0
[init -> drivers -> nic_drv] 
[init -> drivers -> nic_drv] no data processed on irq, isr before=84 now=0
[init -> drivers -> nic_drv] 
[init -> drivers -> nic_drv] no data processed on irq, isr before=84 now=1
[init -> drivers -> nic_drv] 
[init -> report_rom] report 'fetchurl -> progress'
[init -> report_rom]   <progress>
[init -> report_rom]   	<fetch url="https://depot.genode.org/alex-ab/raw/vm-tc-firefox/2018-06-27.tar.xz" total="131577252.0" now="1196032.0"/>
[init -> report_rom]   </progress>
[init -> drivers -> nic_drv] ping
[init -> drivers -> nic_drv] ping: 0 isr=51
[init -> drivers -> nic_drv] 
[init -> report_rom] report 'fetchurl -> progress'
[init -> report_rom]   <progress>
[init -> report_rom]   	<fetch url="https://depot.genode.org/alex-ab/raw/vm-tc-firefox/2018-06-27.tar.xz" total="131577252.0" now="1196032.0"/>
[init -> report_rom]   </progress>
[init -> drivers -> nic_drv] ping
[init -> drivers -> nic_drv] ping: 0 isr=51
[init -> drivers -> nic_drv] 

... (multiple times same message - no progress)

[init -> report_rom] report 'fetchurl -> progress'
[init -> report_rom]   <progress>
[init -> report_rom]   	<fetch url="https://depot.genode.org/alex-ab/raw/vm-tc-firefox/2018-06-27.tar.xz" total="131577252.0" now="1196032.0"/>
[init -> report_rom]   </progress>
[init -> drivers -> nic_drv] ping
[init -> drivers -> nic_drv] ping: 0 isr=51
[init -> drivers -> nic_drv] 
[init -> fetchurl] Error: Operation was aborted by an application callback, failed to fetch https://depot.genode.org/alex-ab/raw/vm-tc-firefox/2018-06-27.tar.xz
[init -> report_rom] report 'fetchurl -> progress'
[init -> report_rom]   <progress>
[init -> report_rom]   	<fetch url="https://depot.genode.org/alex-ab/raw/vm-tc-firefox/2018-06-27.tar.xz" total="131577252.0" now="1196032.0" timeout="true"/>
[init -> report_rom]   </progress>
[init -> fetchurl] fetch https://depot.genode.org/alex-ab/raw/vm-tc-firefox/2018-06-27.tar.xz
```

chelmuth, 2021-02-05:
Great, I merged both commits to staging and flagged the issue **`fixed`**.

commit df6fbc12 (2021-02-05): dde_ipxe: poll once again on irq without rx packets

This quirk is required for Realtek devices.

Fixes #3939
closed 2021-02-23 by nfeske