genodelabs/genode #3942: cached_fs_rom: resolve symlinks

ehmry, 2020-11-06:
Support for traversing symlinks was left out of the initial implementation of the cached_fs_rom for the sake of simplicity. I think that there are compelling uses for symlinks that justify adding support. Symlinks allow multiple file-system paths to refer to the same cached dataspace and if links are uncached then they can be used to update ROM content for new clients without disturbing existing clients.

ehmry, 2020-11-06:
Implemented at 35a8c1f2a3d69d3ded71efa7268082a37f3100ba.

I choose to follow what I started in dc8b4eeb40550e5db33320219066b112d42a75c7 and used the `diag` attribute attached to sessions requests to trace what the server is doing.

nfeske, 2020-11-13:
After pondering some time over your proposal, I don't want to follow you towards this direction because I'm afraid of the big can of worms for security that could arise from it. In traditional OSes, symlinks remain a major attack vector. Without even becoming creative, I can immediately spot two possible attack:

* Someone may deliver symlinks as part of raw/binary depot archives, e.g., shipping a symlink pointing outside the archive. Why would we want to open that door? That would be such a mess. Depot binary archives should better remain just a flat list of files.

* Infinite symlink cycles. I'm a bit puzzled. When you implemented the hardlink handling in the VFS TAR plugin, you paid attention to this problem and solved it in rather cool way (https://github.com/genodelabs/genode/blob/master/repos/os/src/lib/vfs/tar_file_system.h#L528). But in your proposed patch you did not see the problem coming? This leaves the impression that you have not put much thoughts into the possible risks, reinforcing by doubts.

Given the central role the cached_fs_rom server for Sculpt or similar scenarios, there is too much at stake.

This primary blocker aside, let me share additional remarks/concerns:

* The issue text fails to illustrate the usefulness of the proposal. The statement "I think that there are compelling uses for symlinks that justify adding support" is not a supportive argument because any two people may have different notions of "compelling".

* Generally, we cannot assume that symlinks at the file-system session level refer to the same file-system session. Their interpretation is up to the POSIX/libc level, which works on the VFS, not a single file-system session. By starting to attach more semantics than "a symlink is an arbitrary string" at the file-system-session level, we suggest a meaning where none should be.

* By extending the feature set of cached_fs_rom, the component's behavior diverges from the fs_rom component, which would be unexpected by users. So ultimately, I see the (reasonably) call to bring those components on par. The consequence would be letting the additional complexity slowly creep into components like this. This contradicts our goal to leave such fundamental components as simple as possible.

* Because your patch https://github.com/ehmry/genode/commit/35a8c1f2a3d69d3ded71efa7268082a37f3100ba moves various bits and pieces of code around, it becomes impossible to follow the semantic changes without reviewing the code as a whole, which puts a huge burden on the reviewer. Even if I agreed on the direction, I would be unable to assess the implications of the patch without putting a huge amount of effort in it. This friction can be avoided by splitting restructuring/cleanup work from semantic changes into separate commits.

ehmry, 2020-11-13:
Meh.

closed 2020-11-13 by ehmry