genodelabs/genode #3945: base: deadlock ahead message in signal framework

alex-ab, 2020-11-10:
```
[init -> usb_drv] dev_info: USB disconnect, device number 2
[init -> usb_hid_drv] Error: deadlock ahead, mutex=0x18948, return ip=0xabca7
[init -> usb_drv] dev_info: new low-speed USB device number 2 using xhci_hcd
```

```
genode.git/repos/base/src/lib/base/signal_common.cc:215
   abc8f:       4c 89 e7                mov    %r12,%rdi
   abc92:       48 b8 10 19 fb ff ff    movabs $0xfffffffffffb1910,%rax
   abc99:       ff ff ff 
   abc9c:       48 01 d8                add    %rbx,%rax
   abc9f:       ff d0                   callq  *%rax
genode.git/repos/base/include/base/mutex.h:62
   abca1:       48 89 ef                mov    %rbp,%rdi
   abca4:       41 ff d5                callq  *%r13
genode.git/repos/base/src/lib/base/signal_common.cc:218
   abca7:       48 83 c4 18             add    $0x18,%rsp
genode.git/repos/base/include/base/mutex.h:64
   abcab:       48 89 ef                mov    %rbp,%rdi
   abcae:       4c 89 f0                mov    %r14,%rax
genode.git/repos/base/src/lib/base/signal_common.cc:218
```

```
215 .   _platform_finish_dissolve(context);                                         
216                                                                                 
217 .   Mutex::Guard context_destroy_guard(context->_destroy_mutex);                
218 }    
```

The lock holder dead-locks on the destroy mutex, which was taken in Signal()
```
78 Signal::Signal(Signal::Data data) : _data(data)
 79 {
 80 .   if (_data.context) {
 81 .   .   _data.context->_ref_cnt++;
 82
 83 .   .   /*
 84 .   .    * Defer the destruction of the context until the handling of the
 85 .   .    * 'Signal' has completed.
 86 .   .    *
 87 .   .    * Normally, the context can only have one 'Signal' in flight, which is
 88 .   .    * destroyed before 'pending_signal' is called the next time. However,
 89 .   .    * one exception is a signal handler that unexpectedly calls
 90 .   .    * 'pending_signal' itself (i.e., via 'wait_and_dispatch_one_io_signal').
 91 .   .    * As this is dangerous programming pattern (that should be fixed), we
 92 .   .    * print a warning.
 93 .   .    *
 94 .   .    * In this situation, the context-destroy lock is already taken by the
 95 .   .    * outer scope. To avoid a deadlock during the attempt to create
 96 .   .    * a second 'Signal' object for the same context, we take the lock
 97 .   .    * only in the outer scope (where the context's reference counter
 98 .   .    * is in its clear state).
 99 .   .    */
100 .   .   if (_data.context->_ref_cnt == 1) {
101 .   .   .   _data.context->_destroy_mutex.acquire();
102 .   .   } else { 
``

alex-ab, 2020-11-10:
Thanks @chelmuth for the information, that such a message in the context of the signal framework hints that a Signal_handler is destructed during the handler is in use. With 3e6afd2c1e4203ee550789bf3a076f729804abd7 this situation is detected and the destruction deferred.

alex-ab, 2020-11-10:
Sorry, c59cfe1f9452e9d6fe4010cb50e12342301b05fd contains all files

nfeske, 2020-11-10:
Merged to staging.

commit 37844224 (2020-11-10): usb_hid: avoid destruction of used signal handler

Fixes #3945
commit 8185743d (2020-11-17): usb_hid: avoid destruction of used signal handler

Fixes #3945
closed 2020-11-27 by chelmuth