genodelabs/genode #3946: x86 platform driver does not support multiple drivers for a given class

ehmry, 2020-11-10:
The x86 platform driver does not support handing out multiple devices within a PCI class without policies that explicitly map sessions to BDF ids. This is awkward in situations where make and number of PCI devices is known but the bus layout is not.

ehmry, 2020-11-10:
I have a patch at 1a687e4b6d3544bf369a01aeb1d8bdffc27139b1 that allows platform driver policies to select devices within a class using a simple index.

A configuration of such polices would be like this:
```xml
<config>
  <policy label_prefix="eth0.driver">
    <pci class="ETHERNET" index="0"/>
  </policy>
  <policy label_prefix="eth1.driver">
    <pci class="ETHERNET" index="1"/>
  </policy>
</config>
```

[Tested](https://asciinema.org/a/371875) with QEMU and two Virtio network devices.

chelmuth, 2020-11-10:
I understand that the current situation is inconvenient to use. On the other hand, your proposal reads to me like `eth0` and `eth1` from the pre-systemd Linux times, whose where replaced by port/slot naming for valid reasons. I doubt that policy rules like those above could make sense beyond a personal use case. Therefore, I think it's better to transfer the BDF information from the PCI devices ROM into the policy (manually).

ehmry, 2020-11-12:
The context here is that I have qemu machines that are connected through VDE. I'm assuming that regardless of the qemu version the virtio network devices will be enumerated in the same order, but I'm also assuming that the bus id of the devices could change between releases or that virtio-1.0 could be replaced with virtio-1.1. For now I'm just running tests but I would want to be able to reuse the test generator for building images that could be dropped into something like Proxmox without having to know the virtual PCI topology in advance.

closed 2020-11-13 by ehmry

chelmuth, 2020-11-16:
Your use case sounds reasonable and yields a fair requirement for the system as a whole. I do not know why you closed this issue already. But, if you identified and implemented a solution that does not change the internals of the platform driver it may be worthwhile to share it with others. I'm still certain the requirement can be fulfilled in a system-policy component like Sculpt's driver manager.