genodelabs/genode #3952: nic_router: support multiple DHCP option 6 entries

m-stein, 2020-11-17:
This issue consists of two, technically independent, but semantically connected aspects:

The DHCP client functionality of the NIC router is extended in a way that it reads and stores all DHCP option 6 entries of incoming DHCP offer packets instead of only the first one, as done currently. If configured, these entries are then reported by the NIC router.

The DHCP server functionality of the NIC router is extended in a way that it reads out all <dns-server ip="..."> sub-tags of the corresponding <dhcp-server> tag and communicates each through a DHCP option 6 within its DHCP offer packets. The new <dns-server ip="..."> tag replaces the former <dhcp-server dns_server="..."> attribute.

m-stein, 2020-11-17:
This is an implementation of the second aspect (not meant for merging):
2b04e54d4e nic_router:  \<dns-server> subtag for \<dhcp-server>
0bc44b5f12 os: run/nic_router_multiple_dns_servers

I got a bit into trouble with managing the ``Dns_server`` objects. With ``Genode::Fifo``, the order of entries remains as configured. However, on reconfiguration, I have to compare the DNS-server entries of the old DHCP server with those of the new one in order to detect changes and this is not easy with the current ``Genode::Fifo`` interface. The ``Genode::List`` class, at the other hand, is more comfortable in this regard but, by default, results in a reversed order of entries. I solved this problem by adding an ``insert_as_tail`` method to the NIC-router-local wrapper of ``Genode::List``. This method presumably scales bad but for now, we can assume that reconfiguration is a rare event and the number of DNS-server entries is usually not too big.

m-stein, 2020-11-18:
These commits solve both aspects of the issue (only for review, doc missing):
614187a9d6 nic_router: support multiple DHCP option 6 entries
4cec0d8ece os: run/nic_router_multiple_dns_servers
2b1ff58b84 net/dhcp.h: more specific include guards

The ``run/nic_router_multiple_dns_servers test`` demonstrates the forwarding of multiple DNS server addresses from a router-external DHCP server to the routers DHCP client, then to a router-internal DHCP server of another domain, and then to a NIC client of this other domain.

m-stein, 2020-11-23:
@chelmuth Fixed, documented and I've added a corresponding automated test to the autopilot:
795ed9f547 nic_router: support multiple DHCP option 6 entries
f911f4961b remove remnants of 'virtualbox_nic_router' test
d9f973d4a4 net/dhcp.h: more specific include guards

m-stein, 2020-11-26:
@chelmuth I had to add some mandatory fixups ...
d18aec61e4 Fixup "nic_router: support multiple DHCP option 6 entries" (report DNS servers as sub-tags)
abf0034a17 Fixup "nic_router: support multiple DHCP option 6 entries" (bug in Dns_server::equal_to)

... and also added an optional fixup that introduces another flavor of the NIC-router-DHCP test replacing the router-interal 'dns_server_from' propagation o DNS server addresses with a customizable manager component ...
56d425e822 Fixup "nic_router: support multiple DHCP option 6 entries" (managed DHCP test)

What do you think?

m-stein, 2020-11-26:
I think the new 'nic_router_dhcp_managed' test would be, in general, a good test/demonstration on feedback-driven managers for the NIC router. AFAIK, this is not done in any of our scenarios so far.

m-stein, 2020-11-26:
@chelmuth I've added to more fixups that do not change the test but document it and clean it up:
edd6ec069c Fixup "nic_router: support multiple DHCP option 6 entries" (README for run script)
76ace08e80 Fixup "nic_router: support multiple DHCP option 6 entries" (clean up run script)

m-stein, 2020-12-01:
@chelmuth My current staging contains an updated version:
3338734ffc nic_router: support multiple DHCP option 6 entries

Thanks for the review!

chelmuth, 2020-12-01:
Merged to staging.

commit 76e63591 (2020-12-01): nic_router: support multiple DHCP option 6 entries

* The NIC router now considers, memorizes, and, if configured, reports
  multiple DHCP option 6 entries from DHCP replies that it received as DHCP
  client
* A DHCP server at the NIC router can now be configured statically with
  multiple DNS server addresses to propagate
* The 'dns_server_from' attribute of the DHCP server of the NIC router now
  supports the forwarding of multiple DNS server addresses
* The automated run/nic_router_dhcp test tests all the above mentioned new
  functionality and reconfiguring it at runtime. The test was adde [... cut]
commit bad8caee (2020-12-09): nic_router: support multiple DHCP option 6 entries

* The NIC router now considers, memorizes, and, if configured, reports
  multiple DHCP option 6 entries from DHCP replies that it received as DHCP
  client
* A DHCP server at the NIC router can now be configured statically with
  multiple DNS server addresses to propagate
* The 'dns_server_from' attribute of the DHCP server of the NIC router now
  supports the forwarding of multiple DNS server addresses
* The automated run/nic_router_dhcp test tests all the above mentioned new
  functionality and reconfiguring it at runtime. The test was adde [... cut]
closed 2020-12-09 by nfeske