genodelabs/genode #3953: usb: pagefault on too large memory allocations by usb device drivers

alex-ab, 2020-11-20:
```
[init -> usb_hid_drv] dev_info: input: USB HID v1.10 Keyboard [HID 04d9:a067] on usb-usbbus-/input0
[init -> usb_hid_drv] Error: slab too large 262144 requested 245892 cached 2
attempted write at read-only memory (WRITE pf_addr=0x3c073 pf_ip=0x103c140 from pager_object: pd='init -> usb_hid_drv' thread='ep') 
page fault, pd='init -> usb_hid_drv' thread='ep' cpu=0 ip=0x103c140 address=0x3c073 stack pointer=0x407feb40 qualifiers=0x7 irUWP reason=2
[init -> usb_drv] dev_info: USB disconnect, device number 5
``

alex-ab, 2020-11-20:
After some discussion regarding the allocator for usb_hid with  @ssumpf and @cnuke, I came up with some commits which removes the red error messages and fulfill the allocation requests by the USB HID devices which cause them.

alex-ab, 2020-11-24:
@nfeske, @chelmuth : please consider 8a302b47c720bf3087c88bb0c9e927fb011a9a13 for staging, which prevents a page fault. The other commits can be also added later (after release), because they just would improve the supported devices.

nfeske, 2020-11-24:
Thanks for the hint.

nfeske, 2020-11-27:
Fixed in master.

alex-ab, 2020-11-27:
There are several follow up commits on this issue still not merged, how to handle them ?

nfeske, 2020-11-27:
Sorry, I was too eager to close the issue. Let us reopen it then.

commit d576ef16 (2020-12-01): usb_hid: use heap instead of dma/mem allocator

The usb_hid driver does not need to distinguish between
normal memory and DMA capable memory, since all requests are routed via the
USB raw session to the usb host driver. The default Malloc implementation
implements this distinction, however exposes restrictions on the size of
allocations. As seen now by several USB HID devices, the size of device
driver allocations depend on read out hardware features and can be
larger than we support with our specialized default Malloc implementation.

Since we don't need this functionality, switching to an we [... cut]
commit b097e598 (2020-12-09): usb_hid: use heap instead of dma/mem allocator

The usb_hid driver does not need to distinguish between
normal memory and DMA capable memory, since all requests are routed via the
USB raw session to the usb host driver. The default Malloc implementation
implements this distinction, however exposes restrictions on the size of
allocations. As seen now by several USB HID devices, the size of device
driver allocations depend on read out hardware features and can be
larger than we support with our specialized default Malloc implementation.

Since we don't need this functionality, switching to an we [... cut]
closed 2020-12-09 by nfeske