genodelabs/genode #3954: base: remove Thread_deprecated

alex-ab, 2020-11-23:


nfeske, 2020-11-23:
Thanks @alex-ab, this is very much appreciated.

Would you consider the removal of the deprecated `Thread` constructor also? Right now, your patch falls back to the constructor that takes no `Env &` argument at some places (look out for `deprecated` in thread.h), which we ultimately want to remove. For using the non-deprecated `Thread` constructor, we need to adjust the surrounding code to propagate the `Env &`.

nfeske, 2020-11-23:
@alex-ab I wonder a little bit about the removal of the lwip loopback test. Is this test no longer used / needed by anyone? @skalk, do you have an opinion being the author of the test? I guess it is obsolete (since we are universally using the libc's socket API nowadays) but I want to double check. Maybe it would be sensible to leave a comment about it as part of the commit message?

alex-ab, 2020-11-23:
> Would you consider the removal of the deprecated `Thread` constructor also? Right now, your patch falls back to the constructor that takes no `Env &` argument at some places (look out for [...]

This is in the current state not possible as far as I understand. A Env implementation is not available in core (core/main.cc). I can look into that, but this would actually end up in Env_deprecated re-work which I was told not to address.

alex-ab, 2020-11-23:
> @alex-ab I wonder a little bit about the removal of the lwip loopback test. Is this test no longer used / needed by anyone? @skalk, do you have an opinion being the author of the test? I guess it is [...]

I just saw that this test is not used anywhere and the test itself suggest in the header better to look in another component. So @skalk, do we need it still ?

nfeske, 2020-11-23:
> A Env implementation is not available in core (core/main.cc). I can look into that, but this would actually end up in Env_deprecated re-work which I was told not to address.

Indeed. Thanks for clarifying.

commit 4b0f2ac1 (2020-11-23): base: remove Thread_deprecated

Fixes #3954
commit 774b1f42 (2020-11-27): base: remove Thread_deprecated

Fixes #3954
closed 2020-11-27 by chelmuth