genodelabs/genode #3955: libc: argv[0] should always be non-NULL

ehmry, 2020-11-23:
The argv and envp parser in the posix library returns a null argv when no arguments or environmental variables are set. This is always be a valid array, and for compliance with POSIX (as a rule to account for missing checks) the value at `argv[0]` should not be a null pointer. GNU Hello explicity checks that this is the case and refuses to run otherwise.

ehmry, 2020-11-23:
Fix at c1974408aed9111a6046b6a0d1ffdda1c926409c

chelmuth, 2020-11-25:
Am I wrong or does your remark contradict to https://en.cppreference.com/w/c/language/main_function?

> argv  -  Pointer to the first element of an array of argc + 1  pointers, of which the last one is null and the previous ones, if any,  point to strings that represent the arguments passed to the program from  the host environment. If argv[0] is not a null pointer (or, equivalently, if argc  > 0), it points to a string that represents the program name, which  is empty if the program name is not available from the host environment.

ehmry, 2020-11-25:
Yes I am contradicting that, but reading the [POSIX exec](https://pubs.opengroup.org/onlinepubs/9699919799/functions/exec.html) documentation again I get the impressioni that POSIX standard is avoiding standardizing anything here, and my statement that POSIX requires argv[0] be non-null is not right (but not even wrong?).

As far as I can tell there isn't an efffective standard on if an zero argc is allowed, but there is a requirement that in any case that argv be a pointer to an array of at least one element. Anyway, this only an issue when starting POSIX apps directly and not from a shell and no <arg> nodes are in <config/>.

chelmuth, 2020-11-26:
I think can agree with the following.

- The argv and environ arrays are each terminated by a null pointer. The null pointer terminating the argv array is not counted in argc. (from POSIX linked above)
- For robustness we ensure argv is a valid pointer to an array of at least one member (a nullptr).

Did I get this right?

ehmry, 2020-11-26:
On second thought I think its better to leave argv as `{ NULL }` when argc is zero, this is most consistent with the specs.

ehmry, 2020-11-26:
Correct, new patch at 01bac4b0add9da8bbb4fcbe339605533ab7e7be5.

ehmry, 2020-11-28:
Now just the most conservative patch to return an empty arrays if there are no `<arg>` or `<env>` nodes configured.

commit 24230509 (2020-12-01): libc: always set argv and envp to valid arrays

The arrays passed to main(…) must always be valid null-terminated arrays.

Fix #3955
commit 306466fc (2020-12-09): libc: always set argv and envp to valid arrays

The arrays passed to main(…) must always be valid null-terminated arrays.

Fix #3955
closed 2020-12-09 by nfeske