genodelabs/genode #3960: rump: use -fcommon for rump_tools

ssumpf, 2020-11-25:
Newer host gcc versions (10) set `-fno-common` as default, which leads to multiple definitions of otherwise common symbols. "The idea of COMMON symbols in Unix is to allow multiple external definitions of a same variable (in different compilation units) under a single common symbol under certain conditions." - which is used by `rump_tools`.

nfeske, 2020-11-25:
I'm curious, may you share a bit of the background story of this issue?

ssumpf, 2020-11-25:
@nfeske: https://binarydodo.wordpress.com/2016/05/09/investigating-linking-with-common-symbols-in-elf/ - otherwise a lot of multiple definitions of symbols will occur.

nfeske, 2020-11-25:
I was unable draw the connection to the practical problem you encountered. I was wondering about a possible runtime effect. Since you speak of gcc-10, I guess it is about a build problem with a more recent Linux distro.

commit e3ea3191 (2020-11-25): dde_rump: Set '-fcommon' for rump host tools

GCC 10 has `-fno-common` as default which leads to multiple definitions of
of symbols.

Fixes #3960
closed 2020-11-27 by chelmuth