genodelabs/genode #3961: Turn NIC drivers from NIC servers into Uplink clients

m-stein, 2020-11-30:
In order to be able to restart NIC drivers, they have to become clients to their service. The current plan is to achieve this goal as follows:
1. Introduce new "Uplink" session interface that works pretty similar to the NIC session interface but has no link state and no accessor for the MAC address. The link state is given through the lifetime of the session instead. A link is "up" as long as a session is alive and becomes "down" when closing the session. The MAC address is provided by the client as session argument.
2. Create a dummy client that requests the new Uplink session and use it as test bed for adding Uplink-server support to the NIC router (can reuse large parts of the NIC-server functionality of the router).
3. Add optional, configurable Uplink-client support to the Intel NIC-driver and test it with the extended NIC router in a simple network scenario
4. Adapt Sculpt to use the Intel NIC-driver with the Uplink session and test it
5. Adapt remaining NIC drivers and driver sub-systems in order to use the new Uplink session
6. Clean up artifacts of the transition

m-stein, 2020-12-03:
My topic branch show-cases the new Uplink session with the iPXE NIC driver (configurable mode) in context of the run/ping_nic_router test:
2e63c74ce2 Turn NIC drivers from NIC servers into Uplink clients (make mode of ipxe nic drv configurable)
cd57a07406 Turn NIC drivers from NIC servers into Uplink clients (ipxe nic drv as uplink client)
1e46cb7a48 Turn NIC drivers from NIC servers into Uplink clients (adapt MAC address handling to Uplink semantic)
c3aa98e33d Turn NIC drivers from NIC servers into Uplink clients (misuse ping to test Uplink session)
1880a9ed54 Turn NIC drivers from NIC servers into Uplink clients (Uplink session and router support)

nfeske, 2020-12-05:
It is fantastic to see how well the plan takes shape! When looking at the first commit, I first wondered about the mac_address RPC function. All the happier I was when I saw it changed in a subsequent step. The result matches perfectly with what I had in mind.

I have two small remarks:

* Have you considered using the `Attached_ram_dataspace` as `Communication_buffer`?
* Instead of using the `Rpc_object` as base class for the session-component object, I advise using the more modern `Session_object` provided by base/session_object.h. It is readily equipped with the cap/ram accounting. So I would like to promote the use of that one for session interfaces in the future.

m-stein, 2020-12-07:
@nfeske Thanks a lot for the helpful feedback! I'll try to use ``Attached_ram_dataspace`` instead of ``Communication_buffer`` and ``Session_object`` instead of ``Rpc_object`` when cleaning up the code for staging.

m-stein, 2020-12-07:
@nfeske I was thinking about a generic interface between ``Uplink`` connection and driver. I think we should add the generic framework code directly to ``Uplink::Connection``, as we've done it with ``Timer::Connection`` too. I would even go as far as opening and closing the session inside ``Uplink::Connection`` according to the link state. Furthermore, I would introduce a virtual interface ``Uplink::Connection::Driver`` for calls from the connection to the driver. That said, the driver wouldn't have to inherit from the connection. What do you think?

m-stein, 2020-12-07:
PS: The driver would have to hand-over a reference of type ``Uplink::Connection::Driver &`` to the connection constructor.

nfeske, 2020-12-08:
>  I think we should add the generic framework code directly to Uplink::Connection, as we've done it with Timer::Connection too.

I agree. This pattern is also in place for the `Block::Connection`. So I am with you. :-) I'm just not sure if this goes too far, at the current stage of development.

> Furthermore, I would introduce a virtual interface Uplink::Connection::Driver for calls from the connection to the driver. That said, the driver wouldn't have to inherit from the connection. What do you think?

As general direction, the `Uplink::Connection` should be designed such that the driver controls the control flow, not the `Connection`. That is, the driver performs non-blocking calls to the `Uplink::Connection` but the `Uplink::Connection` never calls the driver. This way, it is always up to the driver to control where the calls are coming from. E.g., for DDE-Linux drivers, all code that interacts with Linux code must be executed by special threads that are associated with Linux context information. If the `Uplink::Connection` did the signal-handling on its own and called the `Driver` interface from a signal handler, we would violate the rule.

I encourage you to take the design cues from the `Block::Connection` and   `Block::Request_stream`. Those APIs are designed such that the API user is in the driving seat while safeguarding the API user from mistakes. In the `Block::Connection` case, you can see that the installation of a signal handler is solely up to the driver. So the driver is in full control of what happens when a signal occurs. The driver's interface is not passed as reference to the `Connection` constructor but rather as argument to the `update_jobs' method. This ensures that the driver is not closely coupled with the connection, which is good to avoid object lifetime issues (destructing the driver while the connection exists).

Granted, the notion of "jobs" does not make any sense in the context of a NIC driver. But maybe it is possible to model the situations that a NIC driver may encounter? I have not thought much about it yet, and I think this may need further discussion / exploration. But from the top of my head, I can imagine two scenarios:

1. The driver wants to process packets from the uplink, passing them to the hardware. The driver wishes to process batches of packets. The hardware may have limited capacity (e.g., constrained by DMA buffer size). To model this situation, the `Connection` could offer a method `process_outgoing_packets` that takes a lambda function `handler` as argument. For each outgoing packet, the connection calls the handler with the pointer to the packet data and size, and it returns an enum value `ACCEPTED`, `REJECTED` or `RETRY` (note the similarity to the block/request_stream.h). The `Connection` takes the return value into account for stopping the iterative processing. Once the `process_outgoing_packets` method returned, the driver can commit the batch of packets to the hardware.

2. The driver wants to deliver packets from the network device to the uplink server. There may be a batch of packets waiting to be delivered. The size of the batch deliverable to the client, however, may be smaller than the available batch (e.g., constrained by the packet-stream buffer). In this case, the driver may opt to either buffer or drop packets. To accommodate this situation, the `Connection` could provide a `try_deliver_incoming_packets` method. This method would follow the design of the `Block::Request_stream::try_acknowledge` method. That is, the `try_deliver_incoming_packets` method takes a lambda function as argument, which as repeatedly called with a reference to an interface `Packet` as argument. The `Packet` interface has one method called `deliver(data, size)`. The lambda function is repeatedly called as long as the lambda function actually calls `Packet::deliver` (as long as there are any remaining packets) and the packet-stream conditions are satisfied. Consequently, once `try_deliver_incoming_packets` returns, the largest-possible batch of packets got delivered. The method is prefixed with `try_` because it cannot guarantee that all pending packets are delivered. Hence, it raises awareness of the API user that precaution is needed here (implementing packet dropping or buffering of the remaining undelivered packets).

For waking up the uplink server, I'd use a dedicated `Connection::wakeup_if_needed` method, which triggers the signal at the uplink server side. This explicit wakeup gives the driver the control over the suitable point in time when a wakeup should happen. E.g., the driver may perform repeated calls to `process_outgoing_packets` and `deliver_incoming_packets` as long as progress can be made, and call the `wakup_if_needed` method before going idle. The method is called `wakeup_if_needed` instead of just `wakeup` to clarify that it can be safely called without the need for any potentially bug-prone optimization attempts like wrapping `wakeup` calls into some conditions. The driver should just call `wakeup_if_needed` and be done with it. Internally, the method will deliver signals only if needed. This is implicitly covered by the packet-stream's `wakeup` method. Again, the pattern follows the lines of the `Block::Request_stream`.

As I said above, these are my preliminary ideas. I'm not sure if now is the right time to try them out. But I don't want to stop you either.

Please note that the sketch above lacks a desirable feature, namely zero-copy networking. To would be nice to enable drivers to use the packet-stream buffer directly as DMA buffer. For the `Block::Request_stream`, I have addressed this via the `Request_stream::Payload`. Maybe, this could be taken as blue print for uplink scenario as well? I haven't thought this through. That said, I also don't see zero-copy networking as a strict requirement for now. I just wanted to raise this point.

m-stein, 2020-12-08:
@nfeske Thanks a lot for these very helpful suggestions that are way more detailed than I would have hoped for! Now that you mention it, I also see that keeping mandate over control flow outside the connection is a good idea. I also tried to get my design engine started by diving into the Block Request Stream but admittedly struggled applying the concept to the NIC context. Now, things become more clear :)

I'd love to continue the Uplink work directly with the new interface towards drivers. However, while contemplating about your feedback I sense more and more that it would be sensible to move this step to an extra commit in order to prevent time pressure regarding the general introduction of the Uplink session.

m-stein, 2020-12-08:
If there are no objections, for now I would go with a simple base class in a private header ``os/src/drivers/nic/include/drivers/nic/uplink_client_base.h`` for the generic stuff, that the drivers include as ``drivers/nic/uplink_client_base.h``. As soon as this issue is fixed, I would take this as base for transitioning to a public "Uplink Packet Stream" framework (e.g. ``os/include/uplink/packet_stream.h``).

nfeske, 2020-12-08:
@m-stein I agree with everything you just wrote. The intermediate step gives us more safety, which is good.

m-stein, 2020-12-10:
@nfeske I'm in doubt about a thing and wanted to ask before applying it to each NIC driver. I'm currently changing the drivers in a way that they allocate the correct network role (Uplink client or NIC server) from the heap via ``new`` according to the value of the ``mode`` configuration attribute. Beforehand, they normally had a class member of the NIC server (Root). I could as well use ``Constructible`` here, e.g.:
```
class Main {
   private:
      Constructible<Root> _root;
      Constructible<Uplink_client> _uplink_client;
}
```
... and construct only the one that is needed. But this always blocks RAM that is never used. At the other hand the duality of roles is temporary anyway. I tried to get an example from your issues #3813 and #3845, and found that you directly removed the old role without intermediate support for both roles.

nfeske, 2020-12-10:
I think that using the `Constructible` approach is fine. I like avoiding the dependence from a heap whenever possible. The object is not large anyway.

> and found that you directly removed the old role without intermediate support for both roles.

I took indeed an approach that was slightly different from the two roles but to the same effect. In 7bac1ec7e57241de3a6e360be40ac6730d9ef4a4 and 3863de95898db861baff7a43d9513083ca7b1fdf, I introduced adapter components that I used during the transition. To switch the role of a existing component, I put the corresponding adapter component in front of the old not-yet-transitioned component. Once the transition was finished, I removed the adapters.

During our discussion, I just remembered that I had to support both roles in the intermediate states of the transition, but I failed to remember this nifty detail. Do you think that this approach would make your life easier?

m-stein, 2020-12-10:
@nfeske I totally masked out the role of the adapter components, thanks for the clarification! This could indeed simplify things. I'll give it a try.

m-stein, 2020-12-10:
@nfeske Thanks for your support :)

m-stein, 2020-12-13:
There is a problem with the adapter. If it doesn't want to inspect/modify the packets, it has to know the MAC address of the Uplink client before the NIC client reads it out. But we don't know in which order these events take place. The adapter could of course have an extra thread for the handling of NIC client RPCs in order to block on access to the address but this seems awkward to me. Now, let's say we have two MAC addresses and translate between them. In this case the adapter would, AFAIK, become a bridge and I cannot say whether this would be more work than I'm willing to spent. Thus, I would say I continue with the two-Constructibles-approach in the drivers. Any opinions?

nfeske, 2020-12-13:
> Thus, I would say I continue with the two-Constructibles-approach in the drivers. Any opinions?

I agree. Let's not get too much consumed with the adapter idea. Both approaches the adapters and the roles feature are just temporary crutches anyway.

m-stein, 2020-12-15:
Just a feedback of the current state: On my branch ``3961_Turn_NIC_drivers_from_NIC_servers_into_Uplink_clients``, the following drivers are finished with optionially supporting the ``uplink_client`` mode:
* ipxe nic driver
* wifi driver
* linux nic driver
* fec nic driver
* usb driver

As far as I can see, the following drivers are missing:
* usb modem driver
* usb net driver
* zynq cadence driver
* lan9118 driver
* virtio nic driver

There is also the OpenVPN NIC server in world. I assume ignoring it for now is sensible.

nfeske, 2020-12-16:
Thank you @m-stein for the status update and the impressive accomplishments!

I think there is no way around doing the legwork of updating the remaining drivers. I would very much appreciate you taking these last steps. The finishing line is in sight.

The OpenVPN NIC server is stale and most certainly unused. Should the demand for OpenVPN resurface, we will need to have a fresh look at it anyway, most certainly changing the approach from a plain NIC interception component that plays client and server role to a mere client that uses a NIC and an uplink session. But that's not for now.

m-stein, 2020-12-16:
@nfeske Thanks for your feedback! I didn't want to give the impression that I'm tired of updating drivers :) I just wanted to enable others to get a picture of the timing of the project. My personal plan would be to finish with the remaining drivers till Monday.

m-stein, 2020-12-16:
The usb modem driver and usb net driver are off the list. The remaining drivers have some similarities. Maybe, with a good plan, I can adapt them all together till the weekend.

m-stein, 2021-01-06:
On my 3961_Turn_NIC_drivers_from_NIC_servers_into_Uplink_clients branch, I tested all NIC drivers with the new optional mode attribute set to their default value ``nic_server``, then I adapted all ``drivers_nic-*`` packages (dedicated commit) and some run scripts and tested all drivers in mode ``uplink_client``.

What's missing is testing the USB modem driver that I'd do together with @ssumpf as soon as he's back in office and testing the adapted ``drivers_nic-muen`` package that I'd leave to the build bot.

Now, I'd start adapting all remaining network scenarios to use the mode ``uplink_client``.

skalk, 2021-01-07:
@m-stein I've reviewed the FEC, LAN9118, Linux, and USB NIC driver changes as best as I was able to do. Of course, I could not proof everything semantically. You'll find very few remark inline in the commit.
Quite a good job, and nice to see how carefully you incorporated the role change into the existing code-base! Thank you.

m-stein, 2021-01-08:
@skalk Thanks a lot for the helpful and detailed review :) On my branch, there are now two fix-up commits ("skalk's review") that correspond to your objections. I've tested the affected drivers with the new state.

m-stein, 2021-01-08:
@nfeske I'd say these commits are ready for staging:
2a3eda7e95 nic drivers: provide optional Uplink-client mode
daa4232d20 nic_router: act as "Uplink" server
8e8cbc1d94 os: introduce new "Uplink" session
61b5ca74aa run/ping: support running manually on Linux

m-stein, 2021-01-08:
@nfeske The following commit should be safe to use for restartable NIC drivers in Sculpt but it doesn't adapt all run scripts / packages that use the altered ``drivers_nic-*`` archives, which is why it's not ready for staging yet:
4d1b91bbb7 Don't use the NIC server mode of NIC drivers

nfeske, 2021-01-11:
Thanks a lot @m-stein! I merged the commits you recommended to staging and will test the last one with Sculpt today.

m-stein, 2021-01-13:
@chelmuth Thanks for pointing me to the problems with the merged commits! Here are some fixes:
13acd2affc fixup "nic drivers: provide optional Uplink-client mode" (bug in wifi_drv)
06a722ce31 fixup "nic_router: act as "Uplink" server" (<provides> updates remaining repos)
b9bc0cfed9 fixup "nic_router: act as "Uplink" server" (<provides> updates dde_\*)
f93653da40 fixup "nic_router: act as "Uplink" server" (<provides> updates base\*, os)

... and here are some adaptions for ``world`` related to the recent NIC router changes:
https://github.com/m-stein/genode-world/commit/19be37de7b86c12179ac4644369be229ec23f318 nic_router: act as "Uplink" server
https://github.com/m-stein/genode-world/commit/406bbf54e0892380d04ae073f65641777639d56e nic_router: support multiple DHCP option 6 entries

m-stein, 2021-01-14:
@chelmuth Another fixup for the Sculpt manager:
4d1e121dcf fixup "nic_router: act as "Uplink" server" (<provides> updates sculpt_manager)

m-stein, 2021-01-14:
@chelmuth Sorry, I meant this one:
 ced29ff138 fixup "nic_router: act as "Uplink" server" (<provides> updates sculpt_manager)

nfeske, 2021-01-14:
Thanks. Merged to staging.

m-stein, 2021-01-27:
This should adapt all network scenarios in the genode repo to use the new "Uplink Client" mode of NIC drivers:
e7dad06607 Don't use the NIC server mode of NIC drivers

chelmuth, 2021-01-27:
Splendid! Merged the commit to staging.

m-stein, 2021-01-28:
@chelmuth I've copied the "no-qemu" message from the vbox_win script:
2d207e18ea fixup "Don't use the NIC server mode of NIC drivers" (fix "not supported on qemu" message)

chelmuth, 2021-01-28:
Okay, merged this to staging. Are you looking into the failed tests too?

chelmuth, 2021-01-29:
An incomplete summary of open issues with the last patch.

- [ ] seoul-auto: Error: target 'server/nic_router_' does not exist
- [ ] vm_stress_seoul-debian32: Error: target 'server/nic_router_' does not exist
- [ ] lwip: lynx: Can't access startfile http://10.0.3.2/
- [ ] nic_router_uplinks: var/run/nic_router_uplinks/genode/config:99: parser error
- [ ] nic_dump: timeout
- [ ] ping: timeout

m-stein, 2021-01-29:
@chelmuth Thanks for taking the time to create this list. I will look into it the next days.

m-stein, 2021-02-01:
@chelmuth There are a new commit and several fix-ups that address the list of problems you posted:
 70405d4796 fixup "Don't use the NIC server mode of NIC drivers" (fix run/ping.run)
f1a5c476df fixup "Don't use the NIC server mode of NIC drivers" (fix run/nic_dump.run)
d230408b9d fixup "Don't use the NIC server mode of NIC drivers" (fix run/nic_router_uplinks.run)
9e380416e0 fixup "Don't use the NIC server mode of NIC drivers" (fix run/lwip.run)
15fef79c3f fixup "Don't use the NIC server mode of NIC drivers" (fix run/seoul.inc)
f880683380 drivers/nic/mode: exception on malformed values

I'll have a look whether there are further tests failing due to this issue.

nfeske, 2021-02-01:
Thanks @m-stein! I merged your fixups to staging.

m-stein, 2021-02-02:
@chelmuth @nfeske Further fixups for the most recent failures:
1ea279e4c9 os: api/net archive
58b5a02b3b fixup "Don't use the NIC server mode of NIC drivers" (fix too early request in sntp_client.run)
f3a0dc7960 fixup "Don't use the NIC server mode of NIC drivers" (fix unexpected log order in lwip.run)
544bc6033f fixup "Don't use the NIC server mode of NIC drivers" (fix nic_dump on imx8)

Regarding the problems on imx6 I created issue #4010.

chelmuth, 2021-02-03:
Currently, the monolithic *usb_drv* net mode with `<config> <nic.../> </config>` is broken. Therefore, *netperf_l\*ip_usb30* fails. The dedicated *usb_net_drv* seems to work when running *run/usb_net*.

chelmuth, 2021-02-09:
@m-stein *netperf_\*ip_usb30* are still broken. Maybe because of missing routing configuration?
```
[init -> nic_drv] [31mError: Platform-session creation failed (ram_quota=16K, cap_quota=2)[0m[0m
[init -> nic_drv] [31mError: Uncaught exception of type 'Genode::Service_denied'[0m[0m
```

m-stein, 2021-02-09:
@chelmuth I'm at it.

m-stein, 2021-02-10:
I've added another 3 commits in order to calm the nightly tester:
8cac57c723 ports/recipes/src: usb_net_drv
9c90886a99 ports: rename *_usb30.run *_usb.run
0f9e1e0c83 fixup "Don't use the NIC server mode of NIC drivers" (fix netperf_*_usb30)

I'm also still at removing the monolithic USB approach for NIC from all scenarios.

m-stein, 2021-02-12:
Here's a fix for the failure of netperf_lxip_usb on x86_64/hw/pc last night (not directly related to but triggered through this issue):
eaec145f72 usb_net: allow calling netif_stop_queue dummy

chelmuth, 2021-02-12:
I merged the fix without the unused `backtrace()` function.

m-stein, 2021-02-12:
Oh, I'm sorry for that remnant and thanks for fixing it!

chelmuth, 2021-02-12:
The probing could be replaced by retrieving the kernel information from the *platform_info* ROM and requesting MSI on supported kernels only.

m-stein, 2021-02-12:
I had a look at the netperf_lwip test on x86_64/linux. The test just freezes at a certain point after having started the STREAM step without any suspicious output. When trying be a bit more verbose in the NIC router, the problem disappears. In general it seems that having slightly more load on the linux (in my case a youtube video or a  depot/create) makes the problem disappear.

When going back to the commit directly before the uplink session was introduced, the test succeeds. When minimally altering the test on staging to use the nic_server mode in the nic driver again, the test succeeds. That said, it seems that the implementation of the uplink client in the driver contains/triggers a timing-sensitive bug.

The log in the bad case (staging without other user programs running in linux):
```
Genode 20.11-174-g0198034c2a <local changes>
17592186044415 MiB RAM and 8997 caps assigned to init
[init] parent provides
[init]   service "ROM"
[init]   service "IRQ"
[init]   service "IO_MEM"
[init]   service "IO_PORT"
[init]   service "PD"
[init]   service "RM"
[init]   service "CPU"
[init]   service "LOG"
[init] child "timer"
[init]   RAM quota:  776K
[init]   cap quota:  66
[init]   ELF binary: timer
[init]   priority:   0
[init]   provides service Timer
[init] child "nic_drv"
[init]   RAM quota:  32520K
[init]   cap quota:  966
[init]   ELF binary: init
[init]   priority:   0
[init] child "nic_router"
[init]   RAM quota:  4872K
[init]   cap quota:  86
[init]   ELF binary: nic_router
[init]   priority:   0
[init]   provides service Nic
[init]   provides service Uplink
[init] child "netserver_genode"
[init]   RAM quota:  32520K
[init]   cap quota:  286
[init]   ELF binary: netserver
[init]   priority:   1
[init] child "timer" announces service "Timer"
[init -> nic_router] [uplink] static IP config: interface 10.0.2.55/24, gateway 10.0.2.1 P2P 0
[init -> nic_router] [uplink] NIC sessions: 0
[init -> nic_router] [server] static IP config: interface 10.0.3.1/24, gateway 0.0.0.0 P2P 0
[init -> nic_router] [server] NIC sessions: 0
[init -> nic_drv -> nic_drv] MAC address 02:00:00:00:00:01
[init -> nic_drv -> nic_drv] no config provided, using tap0
[init] child "nic_router" announces service "Nic"
[init] child "nic_router" announces service "Uplink"
[init -> nic_router] [uplink] NIC sessions: 1
Warning: invalid signal-context capability
[init -> nic_router] [server] NIC sessions: 1
[init -> netserver_genode] INFO: NET: Registered protocol family 16
[init -> netserver_genode] INFO: NET: Registered protocol family 2
[init -> netserver_genode] queue_delayed_work: queue_delayed_work called (from 0x108e177) return 0
[init -> netserver_genode] queue_delayed_work: queue_delayed_work called (from 0x10a88ea) return 0
[init -> netserver_genode] INFO: Hash tables configured (established 524288 bind 524288)
[init -> netserver_genode] Received mac: 02:02:02:02:02:01
[init -> netserver_genode] Using DHCP for interface configuration.
[init -> netserver_genode] queue_delayed_work: queue_delayed_work called (from 0x10a6c33) return 0
[init -> netserver_genode] NOTICE: Sending DHCP requests ., OK
[init -> netserver_genode] my address is 10.0.3.2
[init -> netserver_genode] queue_delayed_work: queue_delayed_work called (from 0x10a6c33) return 0
[init -> netserver_genode] queue_delayed_work: queue_delayed_work called (from 0x10a6c33) return 0
[init -> netserver_genode] INFO: IP-Config: Complete:
[init -> netserver_genode] INFO:      device=eth0, len=6 hwaddr=02:02:02:02:02:01, ipaddr=10.0.3.2, mask=255.255.255.0 , gw=10.0.3.1
[init -> netserver_genode] INFO:      host=111d204I4, domain=, nis-domain=
[init -> netserver_genode] INFO:      bootserver=10.0.3.1, rootserver=10.0.3.1, rootpath=
[init -> netserver_genode] INFO:      nameserver0=10.0.3.1
[init -> netserver_genode] Starting netserver with host 'IN(6)ADDR_ANY' port '12865' and family AF_INET
[... 113 lines cut ...]
[init -> netserver_genode] Warning: setitimer: setitimer not implemented
Recv   Send    Send                          Utilization       Service Demand
Socket Socket  Message  Elapsed              Recv     Send     Recv    Send
Size   Size    Size     Time     Throughput  local    remote   local   remote
bytes  bytes   bytes    secs.    10^6bits/s  % S      % U      us/KB   us/KB

    -1  16384      0    9.93         29.27   94.48    -1.00    2115.446  -2.799 

Alignment      Offset         Bytes    Bytes       Recvs   Bytes    Sends
Local  Remote  Local  Remote  Xfered   Per                 Per
Recv   Send    Recv   Send             Recv (avg)          Send (avg)
    8       8      0       0 36315136  1630.82     22268   4096.00   8866

Maximum
Segment
Size (bytes)
  1448

calculation: overall bytes / size per packet / time = packets per second
             36315136 Bytes / 1024 Bytes / 9.93 s =     3571 packets/s

! PERF: TCP_MAERTS              29.27 MBit/s ok
```

Note that I could get a successful case also with less load on the system with results of 120 Mbit/s but I wanted to have the other load to not use the nic.

m-stein, 2021-02-12:
The same seems to apply for the netperf_lxip test on x86_64/linux and for both tests on x86_32/linux.

I also observed cases now where the test runs successfully without additional load on linux and cases where it stops after having started the MAERTS step: 
```
Genode 20.11-174-g0198034c2a <local changes>
17592186044415 MiB RAM and 8997 caps assigned to init
[init] parent provides
[init]   service "ROM"
[init]   service "IRQ"
[init]   service "IO_MEM"
[init]   service "IO_PORT"
[init]   service "PD"
[init]   service "RM"
[init]   service "CPU"
[init]   service "LOG"
[init] child "timer"
[init]   RAM quota:  776K
[init]   cap quota:  66
[init]   ELF binary: timer
[init]   priority:   0
[init]   provides service Timer
[init] child "nic_drv"
[init]   RAM quota:  32520K
[init]   cap quota:  966
[init]   ELF binary: init
[init]   priority:   0
[init] child "nic_router"
[init]   RAM quota:  4872K
[init]   cap quota:  86
[init]   ELF binary: nic_router
[init]   priority:   0
[init]   provides service Nic
[init]   provides service Uplink
[init] child "netserver_genode"
[init]   RAM quota:  32520K
[init]   cap quota:  286
[init]   ELF binary: netserver
[init]   priority:   1
[init] child "timer" announces service "Timer"
[init -> nic_router] [uplink] static IP config: interface 10.0.2.55/24, gateway 10.0.2.1 P2P 0
[init -> nic_router] [uplink] NIC sessions: 0
[init -> nic_router] [server] static IP config: interface 10.0.3.1/24, gateway 0.0.0.0 P2P 0
[init -> nic_router] [server] NIC sessions: 0
[init -> nic_drv -> nic_drv] MAC address 02:00:00:00:00:01
[init -> nic_drv -> nic_drv] no config provided, using tap0
[init] child "nic_router" announces service "Nic"
[init] child "nic_router" announces service "Uplink"
[init -> nic_router] [uplink] NIC sessions: 1
Warning: invalid signal-context capability
[init -> nic_router] [server] NIC sessions: 1
[init -> netserver_genode] lwIP Nic interface down
[init -> netserver_genode] Warning: read blocked until lwIP interface is ready
[init -> netserver_genode] Warning: read blocked until lwIP interface is ready
[init -> netserver_genode] lwIP Nic interface up address=10.0.3.2 netmask=255.255.255.0 gateway=10.0.3.1
[init -> netserver_genode] Starting netserver with host 'IN(6)ADDR_ANY' port '12865' and family AF_INET

---------------------------- TCP_STREAM -----------------------
spawn netperf-2.6.0 -4 -H 10.0.2.55 -P 1 -v 2 -t TCP_STREAM -c -C -- 1024 -P 12864,12864
MIGRATED TCP STREAM TEST from 0.0.0.0 (0.0.0.0) port 12864 AF_INET to 10.0.2.55 () port 12864 AF_INET
[init -> netserver_genode] netperf: get_sock_buffer: getsockopt SO_SNDBUF: errno 42
[init -> netserver_genode] netperf: get_sock_buffer: getsockopt SO_RCVBUF: errno 42
[init -> netserver_genode] could not send timeout for send - test may not terminate
[init -> netserver_genode] could not send timeout for recv - test may not terminate
[init -> netserver_genode] Warning: setitimer: setitimer not implemented
Recv   Send    Send                          Utilization       Service Demand
Socket Socket  Message  Elapsed              Send     Recv     Send    Recv
Size   Size    Size     Time     Throughput  local    remote   local   remote
bytes  bytes   bytes    secs.    10^6bits/s  % S      % U      us/KB   us/KB

    -1  16384  16384    10.03        97.34   20.42    -1.00    137.481  -0.842 

Alignment      Offset         Bytes    Bytes       Sends   Bytes    Recvs
Local  Remote  Local  Remote  Xfered   Per                 Per
Send   Recv    Send   Recv             Send (avg)          Recv (avg)
    8       8      0       0 122011648  16384.00      7447   1503.33  81161

Maximum
Segment
Size (bytes)
  1448

calculation: overall bytes / size per packet / time = packets per second
             122011648 Bytes / 1024 Bytes / 10.03 s =    11880 packets/s

! PERF: TCP_STREAM              97.34 MBit/s ok

---------------------------- TCP_MAERTS -----------------------
spawn netperf-2.6.0 -4 -H 10.0.2.55 -P 1 -v 2 -t TCP_MAERTS -c -C -- 1024 -P 12864,12864
MIGRATED TCP MAERTS TEST from 0.0.0.0 (0.0.0.0) port 12864 AF_INET to 10.0.2.55 () port 12864 AF_INET
[init -> netserver_genode] netperf: get_sock_buffer: getsockopt SO_SNDBUF: errno 42
[init -> netserver_genode] netperf: get_sock_buffer: getsockopt SO_RCVBUF: errno 42
[init -> netserver_genode] could not send timeout for send - test may not terminate
[init -> netserver_genode] could not send timeout for recv - test may not terminate
[init -> netserver_genode] Warning: setitimer: setitimer not implemented
```

m-stein, 2021-02-15:
The netperf_lxip_bridge tests on okl4, pistachio and sel4 fail in a very similar way compared to netperf_lwip test on x86_64/linux. It's also not a stable issue. I had sporadic successful runs on okl4 at least.

m-stein, 2021-02-15:
I had a sporadic successful run now also on sel4.

m-stein, 2021-02-17:
There are 3 new commits regarding this issue:
679bef2062 run/netperf.inc: fix problem with reused ports
ee48579d69 lwip_lx.run: typo in doc
95e1e314d3 usb_net: allow calling netif_stop_queue dummy

m-stein, 2021-02-18:
Some commits to get rid of using old USB driver for NIC on RPI:
a9bc33578e drivers_nic-rpi: use usb_net and usb_host
f19958bab7 usb_host: compile errors with DEBUG_LINUX_PRINTK
10cea991a3 drivers/platform/rpi: remove dead code

This is required as the old USB driver is buggy with NIC since the Uplink-session introduction and we don't want to fix it.

nfeske, 2021-02-18:
I merged the three commits to staging.

chelmuth, 2022-04-26:
I was wondering what's missing to close this issue and found the following drivers require the removal of the server mode.

- [ ] usb_net
- [ ] nic/virtio
- [ ] lan9118

Thereafter os/xsd/nic_driver_types.xsd and os/src/drivers/nic/include/drivers/nic/mode.h could be cleaned up. Then the issue could also be closed.

skalk, 2022-04-26:
@chelmuth I've adapted the missing drivers and removed the nic/mode.h utility as well. As I've seen the only components that still use nic/root.h and nic/component.h are blackhole and nic_loopback. The latter component is not of much help anymore in my eyes. @nfeske do you think we could remove it? If you would like to preserve the nic session server role within the blackhole component, we could move the nowadays public nic/component.h and nic/root.h headers into the blackhole component's private directory. @nfeske @chelmuth do you agree?

nfeske, 2022-04-26:
> The latter component is not of much help anymore in my eyes. @nfeske do you think we could remove it?

The primary role of this component was to server as a blue print for implementing NIC servers, in particular the orderly interaction with the packet stream. I think we should preserve a minimal-complexity blue print for implementing NIC servers. Secretly, I dream of a revisit of the NIC session API - in the spirit of the `Block::Connection::Job` (client-side) and `Block::Request_stream` (server-side) APIs. I'd regard the simplistic NIC-loopback scenario as a good way to illustrate that.

In short, I'd keep nic/root.h and nic/component.h, and eventually modernize them. Since they are unrelated to the NIC drivers (by now), they do not stand in the way of closing this issue.

closed 2022-05-25 by chelmuth

m-stein, 2022-05-26:
@chelmuth Thanks a lot for doing all the clean-up work of this issue and thereby completing it!

m-stein, 2022-05-26:
@skalk @jschlatow Thanks to you as well! I should have re-read my last statement before sending it :)